lwfinger/rtw89 #1: Can't make

Pencil-Yao, 2021-06-03:
At `make` step, have error:

```
make -C /lib/modules/5.8.0-53-generic/build M=/home/ypf/ypf-app/rtw89 modules
make[1]: Entering directory '/usr/src/linux-headers-5.8.0-53-generic'
  CC [M]  /home/ypf/ypf-app/rtw89/core.o
/home/ypf/ypf-app/rtw89/core.c: In function ‘rtw89_core_txq_tasklet’:
/home/ypf/ypf-app/rtw89/core.c:1101:29: error: implicit declaration of function ‘from_tasklet’ [-Werror=implicit-function-declaration]
 1101 |  struct rtw89_dev *rtwdev = from_tasklet(rtwdev, t, txq_tasklet);
      |                             ^~~~~~~~~~~~
/home/ypf/ypf-app/rtw89/core.c:1101:53: error: ‘txq_tasklet’ undeclared (first use in this function)
 1101 |  struct rtw89_dev *rtwdev = from_tasklet(rtwdev, t, txq_tasklet);
      |                                                     ^~~~~~~~~~~
/home/ypf/ypf-app/rtw89/core.c:1101:53: note: each undeclared identifier is reported only once for each function it appears in
/home/ypf/ypf-app/rtw89/core.c: In function ‘rtw89_core_init’:
/home/ypf/ypf-app/rtw89/core.c:1548:2: error: implicit declaration of function ‘tasklet_setup’ [-Werror=implicit-function-declaration]
 1548 |  tasklet_setup(&rtwdev->txq_tasklet, rtw89_core_txq_tasklet);
      |  ^~~~~~~~~~~~~
cc1: some warnings being treated as errors
make[2]: *** [scripts/Makefile.build:286: /home/ypf/ypf-app/rtw89/core.o] Error 1
make[1]: *** [Makefile:1777: /home/ypf/ypf-app/rtw89] Error 2
make[1]: Leaving directory '/usr/src/linux-headers-5.8.0-53-generic'
make: *** [Makefile:49: all] Error 2
```

Pencil-Yao, 2021-06-03:
System info
```
udo lsb_release -a && uname -a
No LSB modules are available.
Distributor ID: Ubuntu
Description:    Ubuntu 20.10
Release:        20.10
Codename:       groovy
Linux ypf-YiEazy 5.8.0-53-generic #60-Ubuntu SMP Thu May 6 07:46:32 UTC 2021 x86_64 x86_64 x86_64 GNU/Linux
```

lwfinger, 2021-06-03:
I had not previously added any fixes for older kernels. That is now changed and the driver should build for kernel 5.8.

closed 2021-06-07 by Pencil-Yao