lwfinger/rtw89 #2: RTL8852AE on Ubuntu 20.04 (kernel 5.8.0)

Matth37, 2021-06-08:
Hello and thank you for your work, I have an issue when I want to use your drivers. I have a Lenovo Legion 5 Pro with a dual boot (Windows 10 / Ubuntu 20.04 for EFIs partitions), secure boot and fast boot (for Windows) disabled .

When I do steps with make / sudo make install, I have no problem (with VERBOSE=1). But when I try to enable the module with this command : `sudo modprobe rtw89pci`, nothing happens for the wifi except that :

```
sudo dmesg | grep rtw
[ 1357.728431] rtw89_pci 0000:04:00.0: Direct firmware load for rtw89/rtw8852a_fw.bin failed with error -2
[ 1357.728434] rtw89_pci 0000:04:00.0: failed to request firmware
[ 1357.730082] rtw89_pci 0000:04:00.0: MAC has already powered on
[ 1357.731612] rtw89_pci 0000:04:00.0: failed to wait firmware completion
[ 1357.731615] rtw89_pci 0000:04:00.0: failed to setup chip information
[ 1357.732117] rtw89_pci: probe of 0000:04:00.0 failed with error -22
```
And if I try to copy the unique rtw8852a_fw.bin in /lib/firmware/rtw89, it doesn't like this :
```
sudo modprobe rtw89pci 
Erreur de segmentation (core dumped) # segmentation fault
sudo modprobe -r rtw89pci 
modprobe: FATAL: Module rtw89pci is in use.

sudo dmesg | grep rtw
[    7.872266] rtw89core: loading out-of-tree module taints kernel.
[    7.872448] rtw89core: module verification failed: signature and/or required key missing - tainting kernel
[    7.873775] rtw89_pci 0000:04:00.0: enabling device (0000 -> 0003)
[    7.873934] rtw89_pci 0000:04:00.0: Direct firmware load for rtw89/rtw8852a_fw.bin failed with error -2
[    7.873936] rtw89_pci 0000:04:00.0: failed to request firmware
[    7.894681] rtw89_pci 0000:04:00.0: failed to wait firmware completion
[    7.894750] rtw89_pci 0000:04:00.0: failed to setup chip information
[    7.895129] rtw89_pci: probe of 0000:04:00.0 failed with error -22
[...]
[  155.313699] rtw89_pci 0000:04:00.0: loading /lib/firmware/rtw89/rtw8852a_fw.bin failed with error -20
[  155.313702] rtw89_pci 0000:04:00.0: Direct firmware load for rtw89/rtw8852a_fw.bin failed with error -20
[  155.313704] rtw89_pci 0000:04:00.0: failed to request firmware
[  155.315633] rtw89_pci 0000:04:00.0: MAC has already powered on
[  155.318235] rtw89_pci 0000:04:00.0: failed to wait firmware completion
[  155.318238] rtw89_pci 0000:04:00.0: failed to setup chip information
[  155.318552] rtw89_pci: probe of 0000:04:00.0 failed with error -22
[  517.524409] rtw89_pci 0000:04:00.0: MAC has already powered on
[  517.526061] RIP: 0010:rtw89_fw_download+0x8f/0xd0 [rtw89core]
[  517.526098]  ? rtw89_pci_probe+0x811/0xb40 [rtw89pci]
```
Have you an idea about the problem ? I did a topic on ubuntu forum : https://askubuntu.com/questions/1344184/realtek-rtl8852ae-issue-on-lenovo-legion-5-pro-16ach6-ubuntu-20-04

I tried this on multiple kernels. Here I stay on the kernel 5.8.0.

tituslesley89, 2021-06-11:
+1 having the exact same issue, look at the same [askubuntu forum](https://askubuntu.com/questions/1344184/realtek-rtl8852ae-issue-on-lenovo-legion-5-pro-16ach6-ubuntu-20-04) post for terminal information from me.

lwfinger, 2021-06-11:
The main problem on those forum articles are that the firmware is not installed. That is my fault as I failed to coopy it in the Makefile. The reason that sudo modprobe rtw89pci does not work is that there are two kernel modules, rtw89core.ko and  rtw89pci.ko. The latter one only drives the PCIe bus, whereas the first drives the wifi card. If you install the drivers, and do a 'sudo modprobe -v rtw89core', you will see that both are loaded.

Once you have both loaded, 'sudo modprobe -r rtw89pci ' will fail, but sudo modprobe -rv rtw89core' should show both being unloaded.

I cannot push the Makefile changes as the build broke under kernel 5.13. It is late here, thus that will not be fixed until tomorrow. You can do the same thing with a 'sudo cp *.bin /lib/modules/rtw89/."

nexustar, 2021-06-12:
having the exact same issue, is there a usable linux version now?

lwfinger, 2021-06-12:
I pushed the new Makefile, thus firmware will now be automatically installed.

I do not have the device, and I have no idea of whether it works.

tituslesley89, 2021-06-12:
Update from my end.

After 'cping' the binary & loading rtw89core, the wifi device still refuses to show up in 'ip link' or 'nmcli'. 

```
[root@lee-thinkpad ~]# lspci -v 
03:00.0 Network controller: Realtek Semiconductor Co., Ltd. Device 8852
        Subsystem: Lenovo Device 4852
        Physical Slot: 0
        Flags: bus master, fast devsel, latency 0, IRQ 87, IOMMU group 13
        I/O ports at 2000 [size=256]
        Memory at fd700000 (64-bit, non-prefetchable) [size=1M]
        Capabilities: [40] Power Management version 3
        Capabilities: [50] MSI: Enable- Count=1/1 Maskable- 64bit+
        Capabilities: [70] Express Endpoint, MSI 00
        Capabilities: [100] Advanced Error Reporting
        Capabilities: [148] Device Serial Number 00-e0-4c-ff-fe-88-52-01
        Capabilities: [158] Latency Tolerance Reporting
        Capabilities: [160] L1 PM Substates
        Kernel driver in use: rtw89_pci
        Kernel modules: rtw89pci

[root@lee-thinkpad ~]# lsmod | grep rtw
rtw89pci               40960  1
rtw89core             532480  1 rtw89pci
mac80211             1163264  2 rtw89pci,rtw89core
cfg80211             1036288  2 rtw89core,mac80211

[root@lee-thinkpad ~]# dmesg | grep firmware
[    0.105797] Spectre V2 : Enabling Restricted Speculation for firmware calls
[    1.473732] Bluetooth: hci0: RTL: assuming no firmware upload needed
[    1.560329] platform regulatory.0: Direct firmware load for regulatory.db failed with error -2
[    1.857750] psmouse serio1: elantech: assuming hardware version 4 (with firmware version 0x7f3001)
[    2.571471] [drm] Loading DMUB firmware via PSP: version=0x00000000
[    2.572219] [drm] Found VCN firmware Version ENC: 1.7 DEC: 4 VEP: 0 Revision: 17
[    2.572231] [drm] PSP loading VCN firmware
```

Note:
- the device is using rtw89pci as it's kernel module, not rtw89core.
- there is no mentioned on rtw89 in dmesg

What I have tried so far (for the past few days):
- blacklisting all above modules (as you cannot just `module -r` them), deleting the .ko files and installing them again.
- various secure boot / UEFI settings
- Finally, re-installed Arch & pulled the repo today (30 min ago from this post).

Try to figure out how to "force" that device to use a specific kernel module ie. force it to use rtw89core.
Since you don't have the device I am happy be a guinea pig.

lwfinger, 2021-06-12:
Your lsmod output shows that it is using rtw89core. The device ID is part of rtw89pci, which is why lspci thinks it is the driver; however, pci.c calls rtw89_core_init(), which is part of rtw89core. I do not know why ARCH fails to see the device. I think rtw89 has code similar to rtw88, and that driver shows up on my openSUSE systems. Please post your entire dmesg output. Run 'dmesg -t > dmesg.txt' and attach dmesg.txt to this issue.

Matth37, 2021-06-12:
I have exactly the same logs on Ubuntu 20.04.

tituslesley89, 2021-06-12:
Here you go [dmesg.txt](https://github.com/lwfinger/rtw89/files/6643031/dmesg.txt)

lwfinger, 2021-06-13:
What driver are you running? Your dmesg output shows a stack overflow in rtw89_pci_driver_init(); however, this repo has no such routine!

oraoto, 2021-06-13:
Have similar issue on Arch, so I add some printk to the module. It's segfaults when parsing the 74th section.
[dmesg.log](https://github.com/lwfinger/rtw89/files/6643571/dmesg.log)

The stacktrace shows rtw89_pci_driver_init too, but I did added printk to fw.c in this repo.

lwfinger, 2021-06-13:
I did find one place where the driver was putting a large struct on the stack, which may have caused your stack overflow.

I have pushed a commit that changes the stack item to a pointer and allocates the memory from the heap. I know that is a lot of jargon, but I hope that is the problem. Do a git pull and we will find out.

oraoto, 2021-06-13:
With the lastest commit
[dmesg.txt](https://github.com/lwfinger/rtw89/files/6644316/dmesg.txt)

oraoto, 2021-06-13:
It’s pretty dangerous, it corrupts my xfs filesystem.

Kawashima-Azumi, 2021-06-13:
Stack overflow cured, but the following error after the update ~2 hours ago:

[   55.076732] cfg80211: Loading compiled-in X.509 certificates for regulatory database
[   55.078680] cfg80211: Loaded X.509 cert 'sforshee: 00b28ddf47aef9cea7'
[   55.102886] rtw89core: loading out-of-tree module taints kernel.
[   55.105046] rtw89_pci 0000:03:00.0: enabling device (0000 -> 0003)
[   55.108124] rtw89_pci 0000:03:00.0: [ERR]fw bin size
[   55.108127] rtw89_pci 0000:03:00.0: parse fw header fail
[   55.108130] rtw89_pci 0000:03:00.0: [ERR]fwdl 0x1E0 = 0x1
[   55.108132] rtw89_pci 0000:03:00.0: [ERR]fwdl 0x83F2 = 0x0
[   55.108139] rtw89_pci 0000:03:00.0: [ERR]fw PC = 0xb89002ec
[   55.108153] rtw89_pci 0000:03:00.0: [ERR]fw PC = 0xb8900390
[   55.108166] rtw89_pci 0000:03:00.0: [ERR]fw PC = 0xb8900390
[   55.108180] rtw89_pci 0000:03:00.0: [ERR]fw PC = 0xb8900390
[   55.108193] rtw89_pci 0000:03:00.0: [ERR]fw PC = 0xb8900390
[   55.108207] rtw89_pci 0000:03:00.0: [ERR]fw PC = 0xb8900390
[   55.108220] rtw89_pci 0000:03:00.0: [ERR]fw PC = 0xb8900390
[   55.108234] rtw89_pci 0000:03:00.0: [ERR]fw PC = 0xb8900390
[   55.108247] rtw89_pci 0000:03:00.0: [ERR]fw PC = 0xb8900390
[   55.108260] rtw89_pci 0000:03:00.0: [ERR]fw PC = 0xb89003a0
[   55.108274] rtw89_pci 0000:03:00.0: [ERR]fw PC = 0xb8900390
[   55.108287] rtw89_pci 0000:03:00.0: [ERR]fw PC = 0xb8900398
[   55.108301] rtw89_pci 0000:03:00.0: [ERR]fw PC = 0xb8900390
[   55.108314] rtw89_pci 0000:03:00.0: [ERR]fw PC = 0xb8900390
[   55.108327] rtw89_pci 0000:03:00.0: [ERR]fw PC = 0xb8900390
[   55.108339] rtw89_pci 0000:03:00.0: failed to setup chip information
[   55.108774] rtw89_pci: probe of 0000:03:00.0 failed with error -22

lwfinger, 2021-06-13:
I have reverted the latest push as it seems defective. I have asked Realtek if they can supply me with a chip. Until they do, I will have to stop working on this driver.

oraoto, 2021-06-14:
[V4 patch from Ping-Ke Shih](https://patchwork.kernel.org/project/linux-wireless/list/?series=474957&state=*) combined with Makefile in this repository works find for me.

Kawashima-Azumi, 2021-06-14:
> [V4 patch from Ping-Ke Shih](https://patchwork.kernel.org/project/linux-wireless/list/?series=474957&state=*) combined with Makefile in this repository works find for me.

That original Makefile [there](https://patchwork.kernel.org/project/linux-wireless/patch/20210429080149.7068-2-pkshih@realtek.com/) works fine for me, but I had to use the "rtw8852a_table.[ch]" files from here. No idea how to generate that table & no time for today since it's late here.

lwfinger, 2021-06-14:
The only problem with the original Makefile was that it failed to copy the firmware. That table is generated by Realtek, not us. The repo is now updated with the new versions of the table.

Happy to hear that the RTL8852AE is working for you.

tituslesley89, 2021-06-15:
> [V4 patch from Ping-Ke Shih](https://patchwork.kernel.org/project/linux-wireless/list/?series=474957&state=*) combined with Makefile in this repository works find for me.

Sorry, newbie here. How do I exactly download and use this patch? Spent an hour googling around.
Installed pwclient & configured it, but then I am failing to figure out what the 'PATCH ID' is (is this even the right track?).

Apologize for re-use this thread on this topic.

dausruddin, 2021-06-15:
How do I remove this completely? Whenever I have this installed, my laptop woild stuck when reboot. I have tried this module on fresh installled Arch twice, so this moduke might really caused this issue.

I have tried `make uninstall` which ended with a successful message, but the system still stuck when rebooting.

<kbd>[image]</kbd>

Kawashima-Azumi, 2021-06-15:
> Sorry, newbie here. How do I exactly download and use this patch? Spent an hour googling around.
> Installed pwclient & configured it, but then I am failing to figure out what the 'PATCH ID' is (is [...]

Download the Realtek official v4 patch through the [series](https://patchwork.kernel.org/series/474957/mbox/) link on [this page](https://patchwork.kernel.org/project/linux-wireless/cover/20210429080149.7068-1-pkshih@realtek.com/), patch the kernel source with -p1 (man patch for details), and add the "rtw8852a_table.c" and ""rtw8852a_table.h" files from this repo to the official source collection, then enable the driver with "make menuconfig" or "make config" or by directly editing the kernel .config file, the "make" and "make modules_install" ...

Some constants may be missing if not using the latest kernel. Just add them to the beginning of the source and it will compile.

Kawashima-Azumi, 2021-06-15:
> How do I remove this completely? Whenever I have this installed, my laptop woild stuck when reboot. I have tried this module on fresh installled Arch twice, so this module might really caused this issue.
> 
> I have tried `make uninstall` which ended with a successful message, but the system still stuck when rebooting.
> 
> [image]

Uninstall only takes effect if you have already loaded the (buggy) driver. You can reboot anyway with the reset button, by power cycling your machine, or using magic-sysrq.

dausruddin, 2021-06-15:
> Uninstall only takes effect on the kernel after reboot

I rebooted multiple times already.

>  if you have already loaded the (buggy) driver

I can see the driver in dmesg, but wifi still doesn't work. So I believe `make uninstall` is the correct way to remove it?

> You can reboot anyway with the reset button, by power cycling your machine, or using magic-sysrq.

Now even after uninstalling, my system gonna freeze as in image above even after uninstalled the driver. Now, whenever I want to reboot or poweroff this laptop, I need to hold power button to force shutdown.

Kawashima-Azumi, 2021-06-15:
> I can see the driver in dmesg, but wifi still doesn't work. So I believe `make uninstall` is the correct way to remove it?

From what I see the "make uninstall" removes rtw89.ko, but the actual drivers are rtw89_pci.ko and rtw89_core.ko. This is however another issue.

tituslesley89, 2021-06-15:
> I rebooted multiple times already.
>
>
> I can see the driver in dmesg, but wifi still doesn't work. So I believe `make uninstall` is the correct way to remove it?
>
>
> Now even after uninstalling, my [...]

I blacklisted the drivers as mentioned in [this post](https://askubuntu.com/questions/110341/how-to-blacklist-kernel-modules) (modprobe -r was not removing the module cause it was in-use)

agj32mrgibbits, 2021-06-15:
I don't know if you guys are having the same problem I did but make sure your `/lib/firmware/rtw89/rtw8852a_fw.bin` file exists and that it's the right one by downloading it from this repository.

I was getting a stack overflow error. Turns out that for some reason mine was a different file size, can't remember what I might have done to cause that, I tried a lot of things over the past few days. Works like a charm now.

dausruddin, 2021-06-15:
> @Kawashima-Azumi From what I see the "make uninstall" removes rtw89.ko, but the actual drivers are rtw89pci.ko and rtw89core.ko. This is however another issue.

Thank you! I removed all leftover files in `/lib/modules/5.12.10-arch1-1/kernel/drivers/net/wireless/realtek/rtw89/` and I can reboot normally now. It is indeed caused by this driver.

lwfinger, 2021-06-15:
@tituslesley89: You do not need to apply the patch. I already did that. Allyou need to do is 'git pull', and you will have the latest source.

hansegucker, 2021-06-15:
For me, it's also failing. Looking into dmesg, I can see the following log output:

```
[  283.953322] rtw89_pci 0000:03:00.0: enabling device (0000 -> 0003)
[  283.953787] rtw89_pci 0000:03:00.0: firmware: direct-loading firmware rtw89/rtw8852a_fw.bin
[  283.957881] RIP: 0010:rtw89_fw_download+0x85/0xc0 [rtw89core]
[  283.957930] Modules linked in: rtw89pci(OE+) rtw89core(OE) mac80211 cfg80211 libarc4 xt_nat xt_tcpudp veth xt_conntrack nft_chain_nat xt_MASQUERADE nf_nat nf_conntrack_netlink nf_conntrack nf_defrag_ipv6 nf_defrag_ipv4 xfrm_user xfrm_algo nft_counter xt_addrtype nft_compat nf_tables libcrc32c nfnetlink br_netfilter bridge stp llc vboxnetadp(OE) vboxnetflt(OE) vboxdrv(OE) rfcomm cmac algif_hash algif_skcipher af_alg overlay bnep btusb btrtl btbcm binfmt_misc btintel snd_hda_codec_realtek snd_hda_codec_generic bluetooth snd_hda_codec_hdmi nls_ascii nls_cp437 snd_hda_intel vfat snd_intel_dspcfg soundwire_intel fat edac_mce_amd soundwire_generic_allocation snd_soc_core uvcvideo jitterentropy_rng videobuf2_vmalloc kvm_amd snd_compress videobuf2_memops drbg soundwire_cadence videobuf2_v4l2 ansi_cprng kvm snd_hda_codec videobuf2_common snd_hda_core snd_hwdep videodev soundwire_bus irqbypass joydev snd_pcm rapl mc sp5100_tco ecdh_generic pcspkr serio_raw efi_pstore wmi_bmof cdc_acm ecc snd_timer
[  284.126940] RIP: 0010:rtw89_fw_download+0x85/0xc0 [rtw89core]
```

My system is a Debian Sid using the kernel 5.10.0-7-amd64. `lspci` shows the following device information:

```
03:00.0 Network controller: Realtek Semiconductor Co., Ltd. Device 8852
```

lwfinger, 2021-06-15:
What firmware do you have? Run the command'sha256sum rtw8852a_fw.bin'. On my system, I get the following:
73aa70c4d415712dec7e17d732b81f37c180cc7fc9bda0f4ec7e8ddc6cf6b0d4  rtw8852a_fw.bin

If your system gets the same result, the next question is did you do a 'git pull' since Monday at 1700 UTC? That is when the firmware loading problem was last addressed. If you sha256sum is different, that too should be fixed with a 'git pull'.

hansegucker, 2021-06-15:
Hello @lwfinger, thanks for the quick answer. I get the same hash, `git pull` has been done the last time today morning.

lwfinger, 2021-06-15:
Good. Your dmesg tells me where the problem happened, but the "why" is cut off.

I just pushed some temporary printouts that may help in understanding what is happening. Please do another 'git pull', remake and install. Then after failure, run 'dmesg > dmesg.txt' and post dmesg.txt.

hansegucker, 2021-06-15:
Ok, here is my dmesg output: [dmesg.txt](https://github.com/lwfinger/rtw89/files/6658254/dmesg.txt)

tituslesley89, 2021-06-16:
Thanks for all the help so far. I'm quite stumpted here, not sure what I am doing wrong.

- `rtw8852a_fw.bin` hash equals the value you mentioned above, manually copied it over to `/lib/firmware/rtw89` .
- blacklisted the firmwares, and `modprobe -r` them and deleted the `.ko` files. Pulled the latest version of this repo and did another `make install`. Driver error is the exact same as my previous post (the weird `rtw89_pci_driver_init` error).
 - I do see some print statements from your latest commit, so it looks like your latest commit is being used.

[dmesg.txt](https://github.com/lwfinger/rtw89/files/6660580/dmesg.txt)

lwfinger, 2021-06-16:
Although I thought I had installed v4 of this driver, the author said I only had v3. I just  downloaded all the v4 code and fixed the kernel changes so that it will compile through kernel version 5.7. I'm working on 5.6 now, but I wanted to get the updated code to you.

hansegucker, 2021-06-16:
> Although I thought I had installed v4 of this driver, the author said I only had v3. I just downloaded all the v4 code and fixed the kernel changes so that it will compile through kernel version 5.7. I'm working on 5.6 now, but I wanted to get the updated code to you.

Oh, yeah, thank you very much. Now it works fine! 🎉

tituslesley89, 2021-06-16:
Works for me too. Thank you for all the work :)

tuket, 2021-06-18:
It works for me too! Thank you very much :)

lwfinger, 2021-06-18:
I am glad that it works for you. I am reopening this issue as I will have an announcement to make, probably early next week.

Matth37, 2021-06-20:
It's work for me too now (I tried for Ubuntu 21.04 with the 5.11 kernel). Thank you very much !

callmesusan, 2021-06-23:
works for me on fedora 34 after:

sudo dnf install dpkg g++ gcc make automake gcc-c++ kernel-devel

thanks a lot!

hazadam, 2021-06-25:
Thank you, I followed the installation instructions and it works for my dual boot Win 10 **Ubuntu 21.04 5.11** kernel.

lwfinger, 2021-06-25:
I think the driver is now working for everyone.

closed 2021-06-25 by lwfinger

jvbferrer, 2021-08-20:
Just installed Windows 10 in a laptop with RTL8852AE but can't find any driver to get Wi-Fi working. Everything I see is about Linux. Any word on this about Windows?

Thanks!

lwfinger, 2021-08-20:
I do not do Windows for anyone but my wife.\

Actually, the first entry in a DuckDuckGo query for "8852ae driver" was for Windows 10. In fact, 6 of the first 7 search results were for Windows - entry #4 was for this repo. I do not think you looked very hard!

jvbferrer, 2021-08-20:
> I do not do Windows for anyone but my wife.\
> 
> Actually, the first entry in a DuckDuckGo query for "8852ae driver" was for Windows 10. In fact, 6 of the first 7 search results were for Windows - entry #4 was for this repo. I do not think you looked very hard!

Yeah, I found that link of Lenovo's driver but wasn't sure if it would work with a HP laptop. I tried installing it and yes, worked perfectly. Thank you anyways!

lwfinger, 2021-10-19:
The error in dmesg.txt says that your kernel headers do not match the running kernel.

frankly-de, 2021-12-10:
I'm afraid I am also not able to get this to work. I get the following outputs:

> prediger@prediger-ThinkBook-15-G3-ACL:$ cd rtw89 && make && sudo make install
make -C /lib/modules/5.11.0-41-generic/build M=/home/prediger/rtw89 modules
make[1]: Verzeichnis „/usr/src/linux-headers-5.11.0-41-generic“ wird betreten
  CC [M]  /home/prediger/rtw89/core.o
  CC [M]  /home/prediger/rtw89/debug.o
  CC [M]  /home/prediger/rtw89/mac80211.o
  CC [M]  /home/prediger/rtw89/mac.o
  CC [M]  /home/prediger/rtw89/phy.o
  CC [M]  /home/prediger/rtw89/fw.o
  CC [M]  /home/prediger/rtw89/rtw8852a.o
  CC [M]  /home/prediger/rtw89/rtw8852a_table.o
  CC [M]  /home/prediger/rtw89/rtw8852a_rfk.o
  CC [M]  /home/prediger/rtw89/rtw8852a_rfk_table.o
  CC [M]  /home/prediger/rtw89/cam.o
  CC [M]  /home/prediger/rtw89/efuse.o
  CC [M]  /home/prediger/rtw89/regd.o
  CC [M]  /home/prediger/rtw89/coex.o
  CC [M]  /home/prediger/rtw89/ps.o
  CC [M]  /home/prediger/rtw89/sar.o
  CC [M]  /home/prediger/rtw89/ser.o
  CC [M]  /home/prediger/rtw89/util.o
  LD [M]  /home/prediger/rtw89/rtw89core.o
  CC [M]  /home/prediger/rtw89/pci.o
  LD [M]  /home/prediger/rtw89/rtw89pci.o
  MODPOST /home/prediger/rtw89/Module.symvers
  CC [M]  /home/prediger/rtw89/rtw89core.mod.o
  LD [M]  /home/prediger/rtw89/rtw89core.ko
  BTF [M] /home/prediger/rtw89/rtw89core.ko
Skipping BTF generation for /home/prediger/rtw89/rtw89core.ko due to unavailability of vmlinux
  CC [M]  /home/prediger/rtw89/rtw89pci.mod.o
  LD [M]  /home/prediger/rtw89/rtw89pci.ko
  BTF [M] /home/prediger/rtw89/rtw89pci.ko
Skipping BTF generation for /home/prediger/rtw89/rtw89pci.ko due to unavailability of vmlinux
make[1]: Verzeichnis „/usr/src/linux-headers-5.11.0-41-generic“ wird verlassen
make -C /lib/modules/5.11.0-41-generic/build M=/home/prediger/rtw89 modules
make[1]: Verzeichnis „/usr/src/linux-headers-5.11.0-41-generic“ wird betreten
make[1]: Verzeichnis „/usr/src/linux-headers-5.11.0-41-generic“ wird verlassen
cp: Aufruf von stat für 'rtl8852au*.bin' nicht möglich: Datei oder Verzeichnis nicht gefunden
make: *** [Makefile:61: install] Fehler 1
prediger@prediger-ThinkBook-15-G3-ACL:~/rtw89$ sudo modprobe rtw89pci
modprobe: ERROR: could not insert 'rtw89pci': Operation not permitted

lshw gives me this
> prediger@prediger-ThinkBook-15-G3-ACL:~$ sudo lshw -C network
[sudo] Passwort für prediger: 
/proc/cpuinfo
  *-network                 
       Beschreibung: Ethernet interface
       Produkt: RTL8111/8168/8411 PCI Express Gigabit Ethernet Controller
       Hersteller: Realtek Semiconductor Co., Ltd.
       Physische ID: 0
       Bus-Informationen: pci@0000:04:00.0
       Logischer Name: enp4s0
       Version: 15
       Seriennummer: 7c:8a:e1:88:cc:35
       Größe: 100Mbit/s
       Kapazität: 1Gbit/s
       Breite: 64 bits
       Takt: 33MHz
       Fähigkeiten: pm msi pciexpress msix bus_master cap_list ethernet physical tp mii 10bt 10bt-fd 100bt 100bt-fd 1000bt-fd autonegotiation
       Konfiguration: autonegotiation=on broadcast=yes driver=r8169 driverversion=5.11.0-41-generic duplex=full firmware=rtl8168h-2_0.0.2 02/26/15 ip=192.168.178.31 latency=0 link=yes multicast=yes port=twisted pair speed=100Mbit/s
       Ressourcen: irq:52 ioport:3000(Größe=256) memory:d0604000-d0604fff memory:d0600000-d0603fff
  *-network UNGEFORDERT
       Beschreibung: Network controller
       Produkt: Realtek Semiconductor Co., Ltd.
       Hersteller: Realtek Semiconductor Co., Ltd.
       Physische ID: 0
       Bus-Informationen: pci@0000:05:00.0
[... 22 lines cut ...]
00:14.3 ISA bridge: Advanced Micro Devices, Inc. [AMD] FCH LPC Bridge (rev 51)
00:18.0 Host bridge: Advanced Micro Devices, Inc. [AMD] Renoir Device 24: Function 0
00:18.1 Host bridge: Advanced Micro Devices, Inc. [AMD] Renoir Device 24: Function 1
00:18.2 Host bridge: Advanced Micro Devices, Inc. [AMD] Renoir Device 24: Function 2
00:18.3 Host bridge: Advanced Micro Devices, Inc. [AMD] Renoir Device 24: Function 3
00:18.4 Host bridge: Advanced Micro Devices, Inc. [AMD] Renoir Device 24: Function 4
00:18.5 Host bridge: Advanced Micro Devices, Inc. [AMD] Renoir Device 24: Function 5
00:18.6 Host bridge: Advanced Micro Devices, Inc. [AMD] Renoir Device 24: Function 6
00:18.7 Host bridge: Advanced Micro Devices, Inc. [AMD] Renoir Device 24: Function 7
01:00.0 SD Host controller: O2 Micro, Inc. SD/MMC Card Reader Controller (rev 01)
02:00.0 Non-Volatile memory controller: Samsung Electronics Co Ltd Device a809
03:00.0 Non-Volatile memory controller: Samsung Electronics Co Ltd Device a809
04:00.0 Ethernet controller: Realtek Semiconductor Co., Ltd. RTL8111/8168/8411 PCI Express Gigabit Ethernet Controller (rev 15)
05:00.0 Network controller: Realtek Semiconductor Co., Ltd. Device 8852
06:00.0 VGA compatible controller: Advanced Micro Devices, Inc. [AMD/ATI] Lucienne (rev c1)
06:00.1 Audio device: Advanced Micro Devices, Inc. [AMD/ATI] Device 1637
06:00.2 Encryption controller: Advanced Micro Devices, Inc. [AMD] Family 17h (Models 10h-1fh) Platform Security Processor
06:00.3 USB controller: Advanced Micro Devices, Inc. [AMD] Renoir USB 3.1
06:00.4 USB controller: Advanced Micro Devices, Inc. [AMD] Renoir USB 3.1
06:00.5 Multimedia controller: Advanced Micro Devices, Inc. [AMD] Raven/Raven2/FireFlight/Renoir Audio Processor (rev 01)
06:00.6 Audio device: Advanced Micro Devices, Inc. [AMD] Family 17h (Models 10h-1fh) HD Audio Controller
07:00.0 SATA controller: Advanced Micro Devices, Inc. [AMD] FCH SATA Controller [AHCI mode] (rev 81)
07:00.1 SATA controller: Advanced Micro Devices, Inc. [AMD] FCH SATA Controller [AHCI mode] (rev 81)

Help would be greatly appreciated. I am used to working with Linux (using Ubuntu for 14 years now), but usually simply use it. So I may need quite detailed instructions.

frankly-de, 2021-12-10:
Sorry to not being up to it. Quite frankly, this is the first time I've come accros dmesg. I'll attach the read-out here:
[2021-12-10 - dmesg.txt](https://github.com/lwfinger/rtw89/files/7694464/2021-12-10.-.dmesg.txt)
I tried to follow your advice, but got stuck with this:

> prediger@prediger-ThinkBook-15-G3-ACL:~/rtw89$ sudo make sign-install
make -C /lib/modules/5.11.0-41-generic/build M=/home/prediger/rtw89 modules
make[1]: Verzeichnis „/usr/src/linux-headers-5.11.0-41-generic“ wird betreten
make[1]: Verzeichnis „/usr/src/linux-headers-5.11.0-41-generic“ wird verlassen
mkdir -p ~/mok
pushd ~/mok
~/mok /home/prediger/rtw89
openssl req -new -x509 -newkey rsa:2048 -keyout MOK.priv -outform DER -out MOK.der -nodes -days 36500 -subj "/CN=Custom MOK/"
Generating a RSA private key
..........................+++++
.........................+++++
writing new private key to 'MOK.priv'
[-----]
mokutil --import MOK.der
input password: 
input password again: 
popd
/bin/bash: Zeile 1: popd: Der Verzeichnisstapel ist leer.
make: *** [Makefile:99: sign] Fehler 1