lwfinger/rtw89 #5: “unavailability of vmlinux” - missing dependency?

mogoh, 2021-06-23:
Hello,

I tried to compile everything under a fresh fedora 34, and I got this output:

```
$ make
make -C /lib/modules/5.12.11-300.fc34.x86_64/build M=/home/mogoh/src/github.com/lwfinger/rtw89 modules
make[1]: Verzeichnis „/usr/src/kernels/5.12.11-300.fc34.x86_64“ wird betreten
  CC [M]  /home/mogoh/src/github.com/lwfinger/rtw89/core.o
  CC [M]  /home/mogoh/src/github.com/lwfinger/rtw89/debug.o
  CC [M]  /home/mogoh/src/github.com/lwfinger/rtw89/mac80211.o
  CC [M]  /home/mogoh/src/github.com/lwfinger/rtw89/mac.o
  CC [M]  /home/mogoh/src/github.com/lwfinger/rtw89/phy.o
  CC [M]  /home/mogoh/src/github.com/lwfinger/rtw89/fw.o
  CC [M]  /home/mogoh/src/github.com/lwfinger/rtw89/rtw8852a.o
  CC [M]  /home/mogoh/src/github.com/lwfinger/rtw89/rtw8852a_table.o
  CC [M]  /home/mogoh/src/github.com/lwfinger/rtw89/rtw8852a_rfk.o
  CC [M]  /home/mogoh/src/github.com/lwfinger/rtw89/rtw8852a_rfk_table.o
  CC [M]  /home/mogoh/src/github.com/lwfinger/rtw89/cam.o
  CC [M]  /home/mogoh/src/github.com/lwfinger/rtw89/efuse.o
  CC [M]  /home/mogoh/src/github.com/lwfinger/rtw89/regd.o
  CC [M]  /home/mogoh/src/github.com/lwfinger/rtw89/coex.o
  CC [M]  /home/mogoh/src/github.com/lwfinger/rtw89/ps.o
  CC [M]  /home/mogoh/src/github.com/lwfinger/rtw89/ser.o
  LD [M]  /home/mogoh/src/github.com/lwfinger/rtw89/rtw89core.o
  CC [M]  /home/mogoh/src/github.com/lwfinger/rtw89/pci.o
  LD [M]  /home/mogoh/src/github.com/lwfinger/rtw89/rtw89pci.o
  MODPOST /home/mogoh/src/github.com/lwfinger/rtw89/Module.symvers
  CC [M]  /home/mogoh/src/github.com/lwfinger/rtw89/rtw89core.mod.o
  LD [M]  /home/mogoh/src/github.com/lwfinger/rtw89/rtw89core.ko
  BTF [M] /home/mogoh/src/github.com/lwfinger/rtw89/rtw89core.ko
Skipping BTF generation for /home/mogoh/src/github.com/lwfinger/rtw89/rtw89core.ko due to unavailability of vmlinux
  CC [M]  /home/mogoh/src/github.com/lwfinger/rtw89/rtw89pci.mod.o
  LD [M]  /home/mogoh/src/github.com/lwfinger/rtw89/rtw89pci.ko
  BTF [M] /home/mogoh/src/github.com/lwfinger/rtw89/rtw89pci.ko
Skipping BTF generation for /home/mogoh/src/github.com/lwfinger/rtw89/rtw89pci.ko due to unavailability of vmlinux
make[1]: Verzeichnis „/usr/src/kernels/5.12.11-300.fc34.x86_64“ wird verlassen
```

Am I missing some dependency?

I have a discussion thread at [ask fedora](https://ask.fedoraproject.org/t/wifi-on-thinkpad-e14-gen3-install-missing-firmware/15140) if that is of interest.

mogoh, 2021-06-24:
So I found it working anyway. But I still wonder what this means and if I should recompile it. If no one finds the time to comment, this issue could be closed.

lwfinger, 2021-06-25:
It means that the utility depmod, which determines the interdependence among modules could not find that file. The message is a warning, which you can safely ignore.

closed 2021-06-25 by lwfinger