lwfinger/rtw89 #6: Confusing names and outdated README

mogoh, 2021-06-24:
Hello, I got a bit confused by the names, so I hope some could clear things and maybe update the README.

First of all, `lspci` states:

```
03:00.0 Network controller: Realtek Semiconductor Co., Ltd. Device 8852
```

Notice, not dangling letters.

The repository description sates:  "Driver for Realtek **8852AE**, an 802.11ax device".

The readme states: "**rtw88**"

And: "Realtek **8852BE**"

And I should load it with: sudo modprobe **rtw_8723de**

So here are some Questions:

Is the chip called 8852, 8852AE, or 8852BE?

rtw89 is the name of the module, right?
So I should load it with `sudo modprobe rtw89pci`, right?
`rtw88` and `rtw_8723de` is completely wrong, right?

lwfinger, 2021-06-24:
First of all, have you noted the names of the modules that are being produced when you do the make? Those are what are important, not some extra file called README. 

Yes, you should load rtw89pci. The other names are wrong. I will look at READ.md again.

mogoh, 2021-06-25:
Thank you.

To explain my train of thought: I don't know how kernel modules work and thought the name would be somehow hidden as metadata in the files. Thus, I trusted the README more than my ability to understand the compiler-output.

Thanks for making my wifi work.

closed 2021-06-25 by lwfinger

jackm, 2021-07-01:
The main README is still outdated. The title still says "**rtw88**" and "**This repository includes drivers for the following cards: Realtek 8852BE**".

Looks to me like you copied this README from the older rtw88 repo but didn't update it to match the new Realtek card. This is quite confusing to read with all the conflicting information and it may lead some to believe that this isn't the driver they need when in fact it is.

lwfinger, 2021-07-01:
Did you do a 'git pull'? The README looks correct.

jackm, 2021-07-01:
With that latest commit now the README title is correct, however it still says "This repository includes drivers for the following cards: Realtek 8852**BE**" when it should say _8852**AE**_.