lwfinger/rtw89 #8: RTW89 buggy with Realtek 8852

rybber78, 2021-07-02:
Hi

Got a new latop with Realtek 8852 WIFI/BT module. I installed RTW89 using readme guide. It does work and I can connect to wifi, but wifi drops frequently/restarting. Bluetooth only works after running 'sudo systemctl restart bluetooth' after startup. This chip does not work OOB, so I really want to help finetune the code to make it work fully.

Some log output:

LSPCI OUTPUT:
01:00.0 Network controller: Realtek Semiconductor Co., Ltd. Device 8852
	Subsystem: Lenovo Device 4852
	Flags: bus master, fast devsel, latency 0, IRQ 74, IOMMU group 7
	I/O ports at 2000 [size=256]
	Memory at fd500000 (64-bit, non-prefetchable) [size=1M]
	Capabilities: <access denied>
	Kernel driver in use: rtw89_pci
	Kernel modules: rtw89pci

DMESG | GREP RTW89:
[   10.362499] rtw89_pci 0000:01:00.0: enabling device (0000 -> 0003)
[   10.367071] rtw89_pci 0000:01:00.0: Firmware version 0.13.8.0, cmd version 0, type 1
[   10.367076] rtw89_pci 0000:01:00.0: Firmware version 0.13.8.0, cmd version 0, type 3
[   10.384436] rtw89_pci 0000:01:00.0: chip rfe_type is 1
[   10.389697] rtw89_pci 0000:01:00.0 wlp1s0: renamed from wlan0
[   10.733585] rtw89_pci 0000:01:00.0: No TXBD consumed after DMA kicked off
[   43.973068] rtw89_pci 0000:01:00.0: failed to setup BA session for 44:ad:d9:c7:f6:06: 4: -16
[   45.041288] rtw89_pci 0000:01:00.0: failed to setup BA session for 44:ad:d9:c7:f6:06: 4: -16
[   45.041301] rtw89_pci 0000:01:00.0: failed to setup BA session for 44:ad:d9:c7:f6:06: 4: -16
[   45.041311] rtw89_pci 0000:01:00.0: failed to setup BA session for 44:ad:d9:c7:f6:06: 4: -16
[   45.435174] rtw89_pci 0000:01:00.0: failed to setup BA session for 44:ad:d9:c7:f6:06: 4: -16
[   45.437382] rtw89_pci 0000:01:00.0: failed to setup BA session for 44:ad:d9:c7:f6:06: 4: -16
[   45.437398] rtw89_pci 0000:01:00.0: failed to setup BA session for 44:ad:d9:c7:f6:06: 4: -16
[   45.580277] rtw89_pci 0000:01:00.0: failed to setup BA session for 44:ad:d9:c7:f6:06: 4: -16
[   45.584068] rtw89_pci 0000:01:00.0: failed to setup BA session for 44:ad:d9:c7:f6:06: 4: -16
[   45.584090] rtw89_pci 0000:01:00.0: failed to setup BA session for 44:ad:d9:c7:f6:06: 4: -16
[  362.477044] rtw89_pci 0000:01:00.0: failed to release fwcmd
[  724.253056] rtw89_pci 0000:01:00.0: failed to setup BA session for 44:ad:d9:c7:f6:06: 4: -11
[  725.348590] rtw89_pci 0000:01:00.0: failed to setup BA session for 44:ad:d9:c7:f6:06: 4: -11
[  728.225625] rtw89_pci 0000:01:00.0: failed to setup BA session for 44:ad:d9:c7:f6:06: 4: -16

lwfinger, 2021-07-02:
I have no idea about the BT problem, but some Lenovo BIOS code are a little funky. Try setting the following option for rtw89pci:

disable_aspm_l1=y

Other options are

disable_aspm_l1ss=y
disable_clkreq=y

rybber78, 2021-07-02:
This is the one part of the guide I didn't understand. How do I set this
exactly - not sure what to put in the brackets?

My laptop is Lenovo Slim 7 Pro AMD

lwfinger, 2021-07-02:
You need to create a file /etc/modprobe.d/50-rtw89.conf. Your editor needs to be called in 'sudo' mode. In that file add a line that says

options rtw89pci disable_aspm_l1=y

Then unload and reload the driver with

sudo modprobe -rv rtw89pci
sudo modprobe -v rtw89pci

rybber78, 2021-07-02:
Done. Still disconnects at times and has trouble reconnecting. Do you need any logs?

rybber78, 2021-07-02:
Tried 'disable_aspm_l1ss=y' seems to do the trick. Will keep testing

lwfinger, 2021-07-03:
I have just uploaded v5 of the driver, which has a number of improvements. Please do a git pull and test that version.

rybber78, 2021-07-03:
Have pulled v5. Not working well. Can't connect to any wifi now. Tried the different options too. Any ideas?

lwfinger, 2021-07-03:
All I can say is that it works for me. If you want v4, do 'git checkout 8f6b38fd88088756e1'.

rybber78, 2021-07-05:
Tried v4 and v5 now, and can't connect to anything. It tries, then prompts med for wifi password again (even if it is correct).

How can I remove it all and retry install?

Can you make anything out of my logs?:
 2717.430090] rtw89_pci 0000:01:00.0 wlp1s0: renamed from wlan0
[ 2736.380989] rtw89_pci 0000:01:00.0: timed out to flush queues
[ 2736.480781] rtw89_pci 0000:01:00.0: timed out to flush queues
[ 2749.287718] rtw89_pci 0000:01:00.0: timed out to flush queues
[ 2749.391387] rtw89_pci 0000:01:00.0: timed out to flush queues
[ 2764.378105] rtw89_pci 0000:01:00.0: timed out to flush queues
[ 2764.481716] rtw89_pci 0000:01:00.0: timed out to flush queues
[ 2779.057306] rtw89_pci 0000:01:00.0: timed out to flush queues
[ 2779.161633] rtw89_pci 0000:01:00.0: timed out to flush queues
[ 2860.043131] rtw89_pci 0000:01:00.0: No TXBD consumed after DMA kicked off
[ 2927.673017] rtw89_pci 0000:01:00.0: No TXBD consumed after DMA kicked off
[ 3158.041791] rtw89_pci 0000:01:00.0: timed out to flush queues
[ 3158.146574] rtw89_pci 0000:01:00.0: timed out to flush queues
t

lwfinger, 2021-07-06:
If you do a 'sudo make install', all the old stuff is replaced.

One possibility, which has been seen quite a bit, is that your BIOS is buggy. Try creating file /etc/modprobe.d/50-rtw89.conf with a line that says 'options rtw89pci disable_aspm_l1=y'. If that does not help, then change disable_aspm_l1 to disable_aspm_l1ss.

rybber78, 2021-07-09:
I have got it working sketchily with disable_aspm_l1ss. Takes a l,ong time to connect, sometimes not at all until rebooting. Even when connected will drop randomly but come back up soon after. Next I will try Ubuntu 21.04 live and see if it works on a fresh install.

lwfinger, 2021-07-09:
All I can say is that it never drops for me. What does the dmesg output say about the disconnect? A fresh install may not help. The fact that disable_aspm_l1ss is needed means that your BIOS is buggy.

Just in case your AP is somewhat strange, add 'options rtw89core disable_ps_mode=y' to the file you created earlier.

lunamera, 2021-07-09:
Yepp, same issue here.  Fequent disconnects, then it takes a reboot or sometimes a power off to connect again. 

I also see the "No TXBD consumed after DMA kicked off" shortly before disconnect but I also get real crashes:
[  113.671502] ------------[ cut here ]------------
[  113.671506] failed to write to DBI register, addr=0x0719
[  113.671546] WARNING: CPU: 0 PID: 548 at /home/harda/TREIBER/rtw89/pci.c:1406 rtw89_dbi_write8+0xf7/0x100 [rtw89pci]
[  113.671547] Modules linked in: uhid ccm rfcomm cmac algif_hash algif_skcipher af_alg bnep nls_iso8859_1 snd_hda_codec_realtek snd_hda_codec_generic ledtrig_audio snd_hda_codec_hdmi snd_hda_intel snd_intel_dspcfg snd_hda_codec snd_hda_core snd_hwdep rtw89pci(O) snd_pcm joydev rtw89core(O) snd_seq_midi snd_seq_midi_event snd_rawmidi mt7601u btusb snd_seq uvcvideo btrtl snd_seq_device videobuf2_vmalloc btbcm mac80211 edac_mce_amd btintel videobuf2_memops videobuf2_v4l2 kvm_amd snd_timer videobuf2_common bluetooth kvm videodev cfg80211 ecdh_generic rapl mc ecc input_leds libarc4 snd wmi_bmof serio_raw efi_pstore snd_rn_pci_acp3x hid_multitouch ucsi_acpi ccp k10temp snd_pci_acp3x soundcore typec_ucsi typec ideapad_laptop sparse_keymap mac_hid sch_fq_codel parport_pc ppdev lp parport ip_tables x_tables autofs4 dm_crypt hid_generic amdgpu iommu_v2 crct10dif_pclmul gpu_sched crc32_pclmul i2c_algo_bit ghash_clmulni_intel ttm aesni_intel drm_kms_helper crypto_simd cryptd glue_helper syscopyarea
[  113.671627]  sysfillrect sysimgblt fb_sys_fops cec rc_core drm ahci nvme libahci i2c_piix4 xhci_pci nvme_core xhci_pci_renesas wmi video i2c_hid hid
[  113.671642] CPU: 0 PID: 548 Comm: kworker/u32:11 Tainted: G           O      5.8.0-59-generic #66~20.04.1-Ubuntu
[  113.671643] Hardware name: LENOVO 82LN/LNVNB161216, BIOS H2CN16WW 06/07/2021
[  113.671656] Workqueue: phy1 rtw89_track_work [rtw89core]
[  113.671661] RIP: 0010:rtw89_dbi_write8+0xf7/0x100 [rtw89pci]
[  113.671664] Code: 00 4c 89 e7 48 8b 40 38 e8 86 e3 26 ee 84 c0 75 09 31 c0 5b 41 5c 41 5d 5d c3 41 0f b7 f5 48 c7 c7 c0 84 19 c1 e8 f1 b1 f8 ed <0f> 0b b8 92 ff ff ff eb e0 0f 1f 44 00 00 55 48 89 e5 41 55 44 0f
[  113.671666] RSP: 0018:ffffacabc067fda8 EFLAGS: 00010286
[  113.671668] RAX: 0000000000000000 RBX: 0000001a892bade6 RCX: 0000000000000027
[  113.671669] RDX: 0000000000000027 RSI: 0000000000000086 RDI: ffff8e1560e18cd8
[  113.671670] RBP: ffffacabc067fdc0 R08: ffff8e1560e18cd0 R09: 0000000000000004
[  113.671671] R10: 0000000000000000 R11: 0000000000000001 R12: ffff8e155a981ec0
[  113.671672] R13: 0000000000000719 R14: ffff8e155e4f0540 R15: ffff8e155e4f0540
[  113.671674] FS:  0000000000000000(0000) GS:ffff8e1560e00000(0000) knlGS:0000000000000000
[  113.671675] CS:  0010 DS: 0000 ES: 0000 CR0: 0000000080050033
[  113.671676] CR2: 0000562edba030a4 CR3: 000000039114c000 CR4: 0000000000340ef0
[  113.671678] Call Trace:
[  113.671685]  rtw89_dbi_write8_clr+0x50/0x70 [rtw89pci]
[  113.671689]  rtw89_pci_link_ps+0x34/0x50 [rtw89pci]
[  113.671703]  rtw89_leave_lps+0x2f/0x70 [rtw89core]
[  113.671712]  rtw89_track_work+0x67/0x130 [rtw89core]
[  113.671717]  process_one_work+0x1e8/0x3b0
[  113.671720]  worker_thread+0x4d/0x3f0
[  113.671722]  kthread+0x114/0x150
[  113.671724]  ? process_one_work+0x3b0/0x3b0
[  113.671726]  ? kthread_park+0x90/0x90
[  113.671730]  ret_from_fork+0x22/0x30
[  113.671733] ---[ end trace b9b7158cb1430b34 ]---
[  113.671739] rtw89_pci 0000:01:00.0: failed to unset ASPM L1, ret=-110
[  131.660924] wlp1s0: deauthenticating from 34:31:c4:f7:7e:3d by local choice (Reason: 3=DEAUTH_LEAVING)

lunamera, 2021-07-09:
Yepp, same issue here.  Fequent disconnects, then it takes a reboot or sometimes a power off to connect again. 

I also see the "No TXBD consumed after DMA kicked off" shortly before disconnect but I also get real crashes:
[  113.671502] ------------[ cut here ]------------
[  113.671506] failed to write to DBI register, addr=0x0719
[  113.671546] WARNING: CPU: 0 PID: 548 at /home/harda/TREIBER/rtw89/pci.c:1406 rtw89_dbi_write8+0xf7/0x100 [rtw89pci]
[  113.671547] Modules linked in: uhid ccm rfcomm cmac algif_hash algif_skcipher af_alg bnep nls_iso8859_1 snd_hda_codec_realtek snd_hda_codec_generic ledtrig_audio snd_hda_codec_hdmi snd_hda_intel snd_intel_dspcfg snd_hda_codec snd_hda_core snd_hwdep rtw89pci(O) snd_pcm joydev rtw89core(O) snd_seq_midi snd_seq_midi_event snd_rawmidi mt7601u btusb snd_seq uvcvideo btrtl snd_seq_device videobuf2_vmalloc btbcm mac80211 edac_mce_amd btintel videobuf2_memops videobuf2_v4l2 kvm_amd snd_timer videobuf2_common bluetooth kvm videodev cfg80211 ecdh_generic rapl mc ecc input_leds libarc4 snd wmi_bmof serio_raw efi_pstore snd_rn_pci_acp3x hid_multitouch ucsi_acpi ccp k10temp snd_pci_acp3x soundcore typec_ucsi typec ideapad_laptop sparse_keymap mac_hid sch_fq_codel parport_pc ppdev lp parport ip_tables x_tables autofs4 dm_crypt hid_generic amdgpu iommu_v2 crct10dif_pclmul gpu_sched crc32_pclmul i2c_algo_bit ghash_clmulni_intel ttm aesni_intel drm_kms_helper crypto_simd cryptd glue_helper syscopyarea
[  113.671627]  sysfillrect sysimgblt fb_sys_fops cec rc_core drm ahci nvme libahci i2c_piix4 xhci_pci nvme_core xhci_pci_renesas wmi video i2c_hid hid
[  113.671642] CPU: 0 PID: 548 Comm: kworker/u32:11 Tainted: G           O      5.8.0-59-generic #66~20.04.1-Ubuntu
[  113.671643] Hardware name: LENOVO 82LN/LNVNB161216, BIOS H2CN16WW 06/07/2021
[  113.671656] Workqueue: phy1 rtw89_track_work [rtw89core]
[  113.671661] RIP: 0010:rtw89_dbi_write8+0xf7/0x100 [rtw89pci]
[  113.671664] Code: 00 4c 89 e7 48 8b 40 38 e8 86 e3 26 ee 84 c0 75 09 31 c0 5b 41 5c 41 5d 5d c3 41 0f b7 f5 48 c7 c7 c0 84 19 c1 e8 f1 b1 f8 ed <0f> 0b b8 92 ff ff ff eb e0 0f 1f 44 00 00 55 48 89 e5 41 55 44 0f
[  113.671666] RSP: 0018:ffffacabc067fda8 EFLAGS: 00010286
[  113.671668] RAX: 0000000000000000 RBX: 0000001a892bade6 RCX: 0000000000000027
[  113.671669] RDX: 0000000000000027 RSI: 0000000000000086 RDI: ffff8e1560e18cd8
[  113.671670] RBP: ffffacabc067fdc0 R08: ffff8e1560e18cd0 R09: 0000000000000004
[  113.671671] R10: 0000000000000000 R11: 0000000000000001 R12: ffff8e155a981ec0
[  113.671672] R13: 0000000000000719 R14: ffff8e155e4f0540 R15: ffff8e155e4f0540
[  113.671674] FS:  0000000000000000(0000) GS:ffff8e1560e00000(0000) knlGS:0000000000000000
[  113.671675] CS:  0010 DS: 0000 ES: 0000 CR0: 0000000080050033
[  113.671676] CR2: 0000562edba030a4 CR3: 000000039114c000 CR4: 0000000000340ef0
[  113.671678] Call Trace:
[  113.671685]  rtw89_dbi_write8_clr+0x50/0x70 [rtw89pci]
[  113.671689]  rtw89_pci_link_ps+0x34/0x50 [rtw89pci]
[  113.671703]  rtw89_leave_lps+0x2f/0x70 [rtw89core]
[  113.671712]  rtw89_track_work+0x67/0x130 [rtw89core]
[  113.671717]  process_one_work+0x1e8/0x3b0
[  113.671720]  worker_thread+0x4d/0x3f0
[  113.671722]  kthread+0x114/0x150
[  113.671724]  ? process_one_work+0x3b0/0x3b0
[  113.671726]  ? kthread_park+0x90/0x90
[  113.671730]  ret_from_fork+0x22/0x30
[  113.671733] ---[ end trace b9b7158cb1430b34 ]---
[  113.671739] rtw89_pci 0000:01:00.0: failed to unset ASPM L1, ret=-110
[  131.660924] wlp1s0: deauthenticating from 34:31:c4:f7:7e:3d by local choice (Reason: 3=DEAUTH_LEAVING)

Powersave or PCI parameter don't improve the behavior.

It looks like the disconnect only occurs I a complex WLAN environment (Mesh network with three receivable stations). When I use the device in an environment with a singe access point only, then it works fine. 
But I also know, the network topology should cause PCI access crashes..

rybber78, 2021-07-09:
> All I can say is that it never drops for me. What does the dmesg output say about the disconnect? A fresh install may not help. The fact that disable_aspm_l1ss is needed means that your BIOS is [...]

You want me to add both options - one on each line?

DMESG here:
[  186.365930] rtw89_pci 0000:01:00.0: failed to setup BA session for f4:7f:35:77:96:e6: 0: -16
[  186.397204] rtw89_pci 0000:01:00.0: failed to setup BA session for f4:7f:35:77:96:e6: 0: -16
[  186.944647] rtw89_pci 0000:01:00.0: failed to setup BA session for f4:7f:35:77:96:e6: 0: -16
[  186.952530] rtw89_pci 0000:01:00.0: failed to setup BA session for f4:7f:35:77:96:e6: 0: -16
[  186.971449] rtw89_pci 0000:01:00.0: failed to setup BA session for f4:7f:35:77:96:e6: 0: -16
[  186.972118] rtw89_pci 0000:01:00.0: failed to setup BA session for f4:7f:35:77:96:e6: 0: -16
[  186.991603] rtw89_pci 0000:01:00.0: failed to setup BA session for f4:7f:35:77:96:e6: 0: -16
[  189.518150] rtw89_pci 0000:01:00.0: failed to setup BA session for f4:7f:35:77:96:e6: 0: -16
[  189.552623] rtw89_pci 0000:01:00.0: failed to setup BA session for f4:7f:35:77:96:e6: 0: -16
[  189.677904] rtw89_pci 0000:01:00.0: failed to setup BA session for f4:7f:35:77:96:e6: 0: -16
[  445.609442] rtw89_pci 0000:01:00.0: timed out to flush queues
[  449.472597] rtw89_pci 0000:01:00.0: timed out to flush queues
[  452.505206] rtw89_pci 0000:01:00.0: No TXBD consumed after DMA kicked off
[  454.757276] rtw89_pci 0000:01:00.0: timed out to flush queues
[  454.859743] rtw89_pci 0000:01:00.0: timed out to flush queues
[  464.040667] rtw89_pci 0000:01:00.0: No TXBD consumed after DMA kicked off
[  496.095124] rtw89_pci 0000:01:00.0: No TXBD consumed after DMA kicked off
[  525.081135] rtw89_pci 0000:01:00.0: No TXBD consumed after DMA kicked off
[  557.306647] rtw89pci: unknown parameter 'disable_ps_mode' ignored
[  557.331582] rtw89_pci 0000:01:00.0: Firmware version 0.13.8.0, cmd version 0, type 1
[  557.331603] rtw89_pci 0000:01:00.0: Firmware version 0.13.8.0, cmd version 0, type 3
[  557.352551] rtw89_pci 0000:01:00.0: chip rfe_type is 1
[  557.367997] rtw89_pci 0000:01:00.0 wlp1s0: renamed from wlan0
[  578.711257] rtw89_pci 0000:01:00.0: No TXBD consumed after DMA kicked off

rybber78, 2021-07-09:
My preliminary test corroborates lunameras finding - connecting to my company mesh wifi is a mesh of dropouts and resets. Connecting to my phone hotspot is almost perfect, with no dropouts.

Bluetooth however is sketchy, and only working when running 'sudo systemctl restart bluetooth'

lwfinger, 2021-07-09:
Yes, add both options, one line for each.

I will pass on the problems with a mesh network to Realtek.

lwfinger, 2021-07-09:
I just noticed the "rtw89pci: unknown parameter 'disable_ps_mode' ignored". That parameter is for rtw89core, not rtw89pci.

'

rybber78, 2021-07-09:
Okay I will test this. Do you have any idea of how fast a realtek addresses
these issues? My Laptop is pretty useless without wifi. Even to the point
where I'm considering reinstalling win10 🙄

lwfinger, 2021-07-11:
It will depend on how much other work they have, if they can duplicate the problem, and how difficult it will be to fix. What did you expect me to say?

riquezjp, 2021-07-11:
> Even to the point where I'm considering reinstalling win10
rybber78, have you tried some other linux distos? I have Lenovo Thinkbook 14 g2 with this 8852 wifi chip & Ubuntu Gnome 20.04 LTS & 21.04 both work really fine with v5.

rybber78, 2021-07-12:
> It will depend on how much other work they have, if they can duplicate the problem, and how difficult it will be to fix. What did you expect me to say?

Well, your previous answer could indicate that you have experience contacting Realtek. Anyway, your effort is appreciated since the alternative is Windows.

I will keep testing, and also try other distros as suggested.

lwfinger, 2021-07-13:
I do, and my answer is based on previous contacts.

closed 2021-07-13 by lwfinger

lunamera, 2021-07-13:
> have you tried some other linux distos?

Just to add, I'm using Ubuntu 20.4 LTS. 
Dropout also occur with the original patchwork patch on 5.13 as well as with the driver from this repo on a 5.11 kernel.

TheFloatingBrain, 2022-01-23:
I tried both `disable_aspm_l1ss=1` and `disable_aspm_l1=1` I am still getting `rtw89_pci 0000:02:00.0: timed out to flush queues` :( Connection still keeps being touch-and-go. I used another similar computer with this driver  and it was fine.

lwfinger, 2022-01-24:
Actually, these kinds of errors are the result of a buggy BIOS that is not handling the PCIe devices correctly. How "similar" was that other computer?

Try add "disable_clkreq=y" to your options.

TheFloatingBrain, 2022-01-26:
> Actually, these kinds of errors are the result of a buggy BIOS that is not handling the PCIe devices correctly. How "similar" was that other computer?
> 
> Try add "disable_clkreq=y" to your options.

It was in the same product line

After I added that and  the other lines it worked, my bad sorry about that.

nfp0, 2023-10-13:
@TheFloatingBrain Are you sure that is the correct syntax for the file? Shouldn't it read like this instead?
```
options rtw89_core disable_ps_mode=y
options rtw89_pci disable_aspm_l1=y disable_aspm_l1ss=y disable_clkreq=y
```

lwfinger, 2023-10-14:
Yes it should. Furthermore, those options only apply to the in-kernel driver. If you are using the drivers from this repo, the module names need to be rtw89core and rtw89pci. Even better, add rules for both kinds of drivers.

pevogam, 2024-03-14:
Hi @lwfinger I used 
```
$ sudo cat /etc/modprobe.d/20-wifi.conf 
options rtw89_pci disable_aspm_l1ss=y disable_aspm_l1=y disable_clkreq=y
options rtw89_core disable_ps_mode=y
```
as
```
$ lsmod | grep rtw
rtw89_8852be           12288  0
rtw89_8852b           364544  1 rtw89_8852be
rtw89_pci              81920  1 rtw89_8852be
rtw89_core            577536  2 rtw89_8852b,rtw89_pci
mac80211             1568768  2 rtw89_core,rtw89_pci
cfg80211             1335296  3 rtw89_8852b,rtw89_core,mac80211
```
but this doesn't fix my issue which also looks like
```
[ 1794.542266] rtw89_8852be 0000:02:00.0: FW does not process h2c registers
[ 1794.542269] rtw89_8852be 0000:02:00.0: HW scan failed with status: -110
[ 1794.935328] rtw89_8852be 0000:02:00.0: xtal si not ready(W): offset=90 val=10 mask=10
[ 1795.006323] rtw89_8852be 0000:02:00.0: xtal si not ready(W): offset=90 val=10 mask=10
[ 1795.006330] rtw89_8852be 0000:02:00.0: mac init fail, ret:-110
```
Is there an easy way to verify the options have taken effect? I grepped dmesg for the names of the parameters and also tried
```
[pevogam@vagrant ~]$ sudo modinfo rtw89_core | grep parm
parm:           disable_ps_mode:Set Y to disable low power mode (bool)
[pevogam@vagrant ~]$ sudo modinfo rtw89_pci | grep parm
parm:           disable_clkreq:Set Y to disable PCI clkreq support (bool)
parm:           disable_aspm_l1:Set Y to disable PCI ASPM L1 support (bool)
parm:           disable_aspm_l1ss:Set Y to disable PCI L1SS support (bool)
```
which doesn't seem to show the current boolean value of the parameters, just "Set Y to disable". My kernel version is also within the problematic range (`6.7.4-100.fc38.x86_64`) so if the values are set could the issue be something else?

lwfinger, 2024-03-14:
sudo modprobe -r rtw89_8852be
sudo modprobe -r rtw89_core
sudo modprobe -v rtw89_8852be

That last one should show the options when it loads rtw89_pci and rtw89_core.

pevogam, 2024-03-14:
> That last one should show the options when it loads rtw89_pci and rtw89_core.

This produced
```
insmod /lib/modules/6.7.4-100.fc38.x86_64/kernel/net/mac80211/mac80211.ko.xz 
insmod /lib/modules/6.7.4-100.fc38.x86_64/kernel/drivers/net/wireless/realtek/rtw89/rtw89_core.ko.xz disable_ps_mode=y 
insmod /lib/modules/6.7.4-100.fc38.x86_64/kernel/drivers/net/wireless/realtek/rtw89/rtw89_pci.ko.xz disable_aspm_l1ss=y disable_aspm_l1=y disable_clkreq=y 
insmod /lib/modules/6.7.4-100.fc38.x86_64/kernel/drivers/net/wireless/realtek/rtw89/rtw89_8852b.ko.xz 
insmod /lib/modules/6.7.4-100.fc38.x86_64/kernel/drivers/net/wireless/realtek/rtw89/rtw89_8852be.ko.xz 
```
so then I assume all options are indeed there. However, the `xtal is not ready` still appears after reloading and the device is likely in a bad state once the error is already encountered:
```
[12876.478139] rtw89_8852be 0000:02:00.0: loaded firmware rtw89/rtw8852b_fw-1.bin
[12876.478249] rtw89_8852be 0000:02:00.0: Unable to change power state from D3cold to D0, device inaccessible
[12876.529327] rtw89_8852be 0000:02:00.0: xtal si not ready(R): offset=41
[12876.529336] rtw89_8852be 0000:02:00.0: no suitable firmware found
[12876.529339] rtw89_8852be 0000:02:00.0: failed to recognize firmware
[12876.529342] rtw89_8852be 0000:02:00.0: failed to setup chip information
[12876.530074] rtw89_8852be: probe of 0000:02:00.0 failed with error -2
```
Do you think this is the same or a different problem? I assume the issue was closed as reported to Realtek?

lwfinger, 2024-03-14:
I have no idea. We have exhaust3ed my steps for recovery from the buggy BIOS code found in Lenovo and HP laptops. You need to read and follow the instructions in README.md concerning reporting problems to the linux-wireless mailing list. That will get the attention of the driver's developers.