lwfinger/rtw89 #11: when i make, error occured

lutien-zhou, 2021-07-03:
/home/lutien/Git/rtw89/mac.c: In function ‘rtw89_mac_port_cfg_bss_color’:
/home/lutien/Git/rtw89/mac.c:3096:28: error: ‘struct ieee80211_bss_conf’ has no member named ‘he_bss_color’; did you mean ‘bss_color’?
  bss_color = vif->bss_conf.he_bss_color.color;
                            ^~~~~~~~~~~~
                            bss_color
scripts/Makefile.build:269: recipe for target '/home/lutien/Git/rtw89/mac.o' failed
make[2]: *** [/home/lutien/Git/rtw89/mac.o] Error 1
Makefile:1762: recipe for target '/home/lutien/Git/rtw89' failed
make[1]: *** [/home/lutien/Git/rtw89] Error 2
make[1]: Leaving directory '/usr/src/linux-headers-5.4.0-77-generic'
Makefile:52: recipe for target 'all' failed

Richard-LXA, 2021-07-03:
thx, i ve already solved

lwfinger, 2021-07-03:
How did you solve it? I just submitted patches that fixed builds for kernel v5.4 and newer. Previously, the driver would only build for kernel 5.7 and newer.

lutien-zhou, 2021-07-03:
> How did you solve it? I just submitted patches that fixed builds for kernel v5.4 and newer. Previously, the driver would only build for kernel 5.7 and newer.

Actually I upgrade to Ubuntu 20.04 for building it. At least, it works anyway! (

Gerjen100, 2021-07-14:
Im still experiencing this issue. Im currently using ubuntu 18.04 with the 5.4.0-77-generic kernel. I get the following output.

`make`
make -C /lib/modules/5.4.0-77-generic/build M=/home/gerjen/rtw89 modules
make[1]: Entering directory '/usr/src/linux-headers-5.4.0-77-generic'
  Building modules, stage 2.
  MODPOST 2 modules
make[1]: Leaving directory '/usr/src/linux-headers-5.4.0-77-generic'

`sudo make install`
make -C /lib/modules/5.4.0-77-generic/build M=/home/gerjen/rtw89 modules
make[1]: Entering directory '/usr/src/linux-headers-5.4.0-77-generic'
  Building modules, stage 2.
  MODPOST 2 modules
make[1]: Leaving directory '/usr/src/linux-headers-5.4.0-77-generic'
cp *.bin /lib/firmware/rtw89/.
cp: cannot create regular file '/lib/firmware/rtw89/.': No such file or directory
Makefile:54: recipe for target 'install' failed
make: *** [install] Error 1

lwfinger, 2021-07-14:
There was a step missing in the install section of the Make file. Do a 'git pull' followed by 'sudo make install', and the error should be gone.

Gerjen100, 2021-07-15:
Its working now. Thanks a lot!

closed 2021-07-17 by lwfinger

Kazi-Rakib-Hasan-Jawwad, 2022-11-11:
Hello. I am using ubuntu 18.04. My wifi driver is: Realtek 8852AE.
When I enter the command: 'cd rtw89' -> 'make' ; This happens:
'''
make -C /lib/modules/5.4.0-131-generic/build M=/home/rakib/rtw89 modules
make[1]: Entering directory '/usr/src/linux-headers-5.4.0-131-generic'
  CC [M]  /home/rakib/rtw89/core.o
In file included from /home/rakib/rtw89/cam.h:8:0,
                 from /home/rakib/rtw89/core.c:7:
/home/rakib/rtw89/core.h:4431:18: warning: ‘struct cfg80211_tid_config’ declared inside parameter list will not be visible outside of this definition or declaration
           struct cfg80211_tid_config *tid_config);
                  ^~~~~~~~~~~~~~~~~~~
/home/rakib/rtw89/core.c:2609:19: warning: ‘struct cfg80211_tid_cfg’ declared inside parameter list will not be visible outside of this definition or declaration
            struct cfg80211_tid_cfg *tid_conf)
                   ^~~~~~~~~~~~~~~~
/home/rakib/rtw89/core.c: In function ‘_rtw89_core_set_tid_config’:
/home/rakib/rtw89/core.c:2613:21: error: dereferencing pointer to incomplete type ‘struct cfg80211_tid_cfg’
  u32 mask = tid_conf->mask;
                     ^~
In file included from ./include/linux/bitops.h:5:0,
                 from ./include/linux/kernel.h:12,
                 from ./include/linux/skbuff.h:13,
                 from ./include/linux/ip.h:16,
                 from /home/rakib/rtw89/core.c:4:
/home/rakib/rtw89/core.c:2628:18: error: ‘NL80211_TID_CONFIG_ATTR_AMPDU_CTRL’ undeclared (first use in this function); did you mean ‘NL80211_MESHCONF_ATTR_MAX’?
   if (mask & BIT(NL80211_TID_CONFIG_ATTR_AMPDU_CTRL)) {
                  ^
./include/linux/bits.h:8:30: note: in definition of macro ‘BIT’
 #define BIT(nr)   (UL(1) << (nr))
                              ^~
/home/rakib/rtw89/core.c:2628:18: note: each undeclared identifier is reported only once for each function it appears in
   if (mask & BIT(NL80211_TID_CONFIG_ATTR_AMPDU_CTRL)) {
                  ^
./include/linux/bits.h:8:30: note: in definition of macro ‘BIT’
 #define BIT(nr)   (UL(1) << (nr))
                              ^~
/home/rakib/rtw89/core.c:2629:27: error: ‘NL80211_TID_CONFIG_ENABLE’ undeclared (first use in this function); did you mean ‘NL80211_CMD_CONN_FAILED’?
    if (tid_conf->ampdu == NL80211_TID_CONFIG_ENABLE) {
                           ^~~~~~~~~~~~~~~~~~~~~~~~~
                           NL80211_CMD_CONN_FAILED
/home/rakib/rtw89/core.c: At top level:
/home/rakib/rtw89/core.c:2654:18: warning: ‘struct cfg80211_tid_config’ declared inside parameter list will not be visible outside of this definition or declaration
           struct cfg80211_tid_config *tid_config)
                  ^~~~~~~~~~~~~~~~~~~
/home/rakib/rtw89/core.c:2652:6: error: conflicting types for ‘rtw89_core_set_tid_config’
 void rtw89_core_set_tid_config(struct rtw89_dev *rtwdev,
      ^~~~~~~~~~~~~~~~~~~~~~~~~
In file included from /home/rakib/rtw89/cam.h:8:0,
                 from /home/rakib/rtw89/core.c:7:
/home/rakib/rtw89/core.h:4429:6: note: previous declaration of ‘rtw89_core_set_tid_config’ was here
 void rtw89_core_set_tid_config(struct rtw89_dev *rtwdev,
      ^~~~~~~~~~~~~~~~~~~~~~~~~
/home/rakib/rtw89/core.c: In function ‘rtw89_core_set_tid_config’:
/home/rakib/rtw89/core.c:2658:28: error: dereferencing pointer to incomplete type ‘struct cfg80211_tid_config’
  for (i = 0; i < tid_config->n_tid_conf; i++)
                            ^~
/home/rakib/rtw89/core.c: In function ‘rtw89_core_register_hw’:
/home/rakib/rtw89/core.c:3357:13: error: ‘struct wiphy’ has no member named ‘tid_config_support’; did you mean ‘bss_select_support’?
  hw->wiphy->tid_config_support.vif |= BIT(NL80211_TID_CONFIG_ATTR_AMPDU_CTRL);
             ^~~~~~~~~~~~~~~~~~
             bss_select_support
In file included from ./include/linux/bitops.h:5:0,
                 from ./include/linux/kernel.h:12,
                 from ./include/linux/skbuff.h:13,
                 from ./include/linux/ip.h:16,
                 from /home/rakib/rtw89/core.c:4:
/home/rakib/rtw89/core.c:3357:43: error: ‘NL80211_TID_CONFIG_ATTR_AMPDU_CTRL’ undeclared (first use in this function); did you mean ‘NL80211_MESHCONF_ATTR_MAX’?
  hw->wiphy->tid_config_support.vif |= BIT(NL80211_TID_CONFIG_ATTR_AMPDU_CTRL);
                                           ^
./include/linux/bits.h:8:30: note: in definition of macro ‘BIT’
 #define BIT(nr)   (UL(1) << (nr))
                              ^~
/home/rakib/rtw89/core.c:3358:13: error: ‘struct wiphy’ has no member named ‘tid_config_support’; did you mean ‘bss_select_support’?
  hw->wiphy->tid_config_support.peer |= BIT(NL80211_TID_CONFIG_ATTR_AMPDU_CTRL);
             ^~~~~~~~~~~~~~~~~~
             bss_select_support
scripts/Makefile.build:270: recipe for target '/home/rakib/rtw89/core.o' failed
make[2]: *** [/home/rakib/rtw89/core.o] Error 1
Makefile:1762: recipe for target '/home/rakib/rtw89' failed
make[1]: *** [/home/rakib/rtw89] Error 2
make[1]: Leaving directory '/usr/src/linux-headers-5.4.0-131-generic'
Makefile:86: recipe for target 'all' failed
make: *** [all] Error 2
'''
I just installed ubuntu and know very little. Hope I can get some help!

lwfinger, 2022-11-11:
There are too many changes between kernel 5.4 and 5.5 for me to support the older version. You should install Ubuntu 20.04.