lwfinger/rtw89 #14: logs flooding wpa_supplicant notice.

riquezjp, 2021-07-06:
I've noticed my log is flooding with wpa_supplicant notice. About 5 times per second! It resumes after restart.
Although I do not have any noticable loss of connection.
Is this related to rtw89? I am still on v4. 

`7月 06 12:36:07 Sulaco wpa_supplicant[598]: wlp4s0: CTRL-EVENT-SIGNAL-CHANGE above=1 signal=0 noise=9999 txrate=144400
 7月 06 12:36:07 Sulaco wpa_supplicant[598]: wlp4s0: CTRL-EVENT-SIGNAL-CHANGE above=1 signal=0 noise=9999 txrate=144400
 7月 06 12:36:07 Sulaco wpa_supplicant[598]: wlp4s0: CTRL-EVENT-SIGNAL-CHANGE above=1 signal=0 noise=9999 txrate=144400
 7月 06 12:36:07 Sulaco wpa_supplicant[598]: wlp4s0: CTRL-EVENT-SIGNAL-CHANGE above=1 signal=0 noise=9999 txrate=144400
 7月 06 12:36:07 Sulaco wpa_supplicant[598]: wlp4s0: CTRL-EVENT-SIGNAL-CHANGE above=1 signal=0 noise=9999 txrate=144400
`

(Ubuntu 20.04, Lenovo Thinkbook 14 g2 amd)

riquezjp, 2021-07-06:
I was able to able to quiet the flood of notices editing wpa_supplicant.service following [guidance here](https://askubuntu.com/questions/1177443/wpa-supplicant-is-floading-journal)

```
sudo systemctl edit wpa_supplicant.service
```

add the following to the top editable area (-q option is additional to default) & save

```
[Service]
ExecStart=
ExecStart=/sbin/wpa_supplicant -q -u -s -0 /run/wpa_supplicant
```

restart wpa_supplicant
```
sudo systemctl restart wpa_supplicant
```

Turn wi-fi off/on to reconnect. Excessive log is silenced.

lwfinger, 2021-07-06:
I do not use the supplicant directly, but use NetworkManager, which calls it. I have no idea what options it uses, but I have never seen any such messages in my logs.

I do noe see the TXBD messages. You should switch to v5 of the driver, which is now in branch main.

closed 2021-07-06 by lwfinger