lwfinger/rtw89 #15: Frequent reboots with kernel panic

mrjoes, 2021-07-08:
Hi!

 Not sure if I should send this here or to the patch authors. Most likely patch authors, as it seems that it is a hardware (?) timeout issue.

```
Jul 08 09:43:09 meerkat kernel: ------------[ cut here ]------------
Jul 08 09:43:09 meerkat kernel: failed to write to DBI register, addr=0x0719
Jul 08 09:43:09 meerkat kernel: WARNING: CPU: 4 PID: 231164 at /home/serge/misc/aur/rtw89/pci.c:1406 rtw89_dbi_write8+0xe9/0x100 [rtw89pci]
Jul 08 09:43:09 meerkat kernel: Modules linked in: ipmi_devintf ipmi_msghandler tun loop cmac ccm snd_seq_dummy snd_hrtimer snd_seq snd_seq_device joydev mousedev snd_soc_dmic snd_acp3x_rn s>
Jul 08 09:43:09 meerkat kernel:  tpm_tis_core roles wmi soundcore video mac_hid pinctrl_amd acpi_cpufreq i2c_scmi drm fuse agpgart bpf_preload ip_tables x_tables ext4 crc32c_generic crc16 mb>
Jul 08 09:43:09 meerkat kernel: CPU: 4 PID: 231164 Comm: kworker/u32:5 Tainted: G        W  OE     5.12.14-arch1-1 #1
Jul 08 09:43:09 meerkat kernel: Hardware name: LENOVO 20XK0019US/20XK0019US, BIOS R1MET35W (1.05 ) 04/23/2021
Jul 08 09:43:09 meerkat kernel: Workqueue: phy0 rtw89_track_work [rtw89core]
Jul 08 09:43:09 meerkat kernel: RIP: 0010:rtw89_dbi_write8+0xe9/0x100 [rtw89pci]
Jul 08 09:43:09 meerkat kernel: Code: 10 00 00 48 89 ef 48 8b 40 38 e8 12 5c bc e4 84 c0 75 07 31 c0 5b 5d 41 5c c3 41 0f b7 f4 48 c7 c7 00 14 e4 c0 e8 97 64 80 e4 <0f> 0b b8 92 ff ff ff eb >
Jul 08 09:43:09 meerkat kernel: RSP: 0018:ffffb87e83c7be10 EFLAGS: 00010282
Jul 08 09:43:09 meerkat kernel: RAX: 0000000000000000 RBX: 0000172c2f395bc7 RCX: 0000000000000027
Jul 08 09:43:09 meerkat kernel: RDX: ffff9c485ed186e8 RSI: 0000000000000001 RDI: ffff9c485ed186e0
Jul 08 09:43:09 meerkat kernel: RBP: ffff9c4583801f40 R08: 0000000000000000 R09: ffffb87e83c7bc40
Jul 08 09:43:09 meerkat kernel: R10: ffffb87e83c7bc38 R11: ffffffffa66cc4a8 R12: 0000000000000719
Jul 08 09:43:09 meerkat kernel: R13: ffff9c45ca5f7900 R14: 0000000000000000 R15: 0000000000000009
Jul 08 09:43:09 meerkat kernel: FS:  0000000000000000(0000) GS:ffff9c485ed00000(0000) knlGS:0000000000000000
Jul 08 09:43:09 meerkat kernel: CS:  0010 DS: 0000 ES: 0000 CR0: 0000000080050033
Jul 08 09:43:09 meerkat kernel: CR2: 00001b29080b600a CR3: 0000000083c10000 CR4: 0000000000750ee0
Jul 08 09:43:09 meerkat kernel: PKRU: 55555554
Jul 08 09:43:09 meerkat kernel: Call Trace:
Jul 08 09:43:09 meerkat kernel:  rtw89_pci_link_ps+0xc2/0xe0 [rtw89pci]
Jul 08 09:43:09 meerkat kernel:  rtw89_leave_lps+0x2a/0x60 [rtw89core]
Jul 08 09:43:09 meerkat kernel:  rtw89_track_work+0x5f/0x130 [rtw89core]
Jul 08 09:43:09 meerkat kernel:  process_one_work+0x1e3/0x3b0
Jul 08 09:43:09 meerkat kernel:  worker_thread+0x50/0x3b0
Jul 08 09:43:09 meerkat kernel:  ? process_one_work+0x3b0/0x3b0
Jul 08 09:43:09 meerkat kernel:  kthread+0x133/0x150
Jul 08 09:43:09 meerkat kernel:  ? kthread_associate_blkcg+0xc0/0xc0
Jul 08 09:43:09 meerkat kernel:  ret_from_fork+0x22/0x30
Jul 08 09:43:09 meerkat kernel: ---[ end trace 95cff8cf27d790a8 ]---
Jul 08 09:43:09 meerkat kernel: rtw89_pci 0000:03:00.0: failed to unset ASPM L1, ret=-110
```

lwfinger, 2021-07-10:
That is not a PANIC. It is merely a WARNING. Create a file (using sudo) named /etc/modprobe.d/50-rtw89.conf and add a line that says 'options rtw89pci  disable_aspm_l1ss=y', without the quotes. Then reboot. 

Let me know the results.

mrjoes, 2021-07-11:
This is the last log message before each and every reboot.

Added the option, we'll see how it goes. Thank you!

mrjoes, 2021-07-13:
It's been stable so far. Sometimes I'm losing wi-fi connectivity, but the system is no longer rebooting on its own. Closing for now.

closed 2021-07-13 by mrjoes