lwfinger/rtw89 #16: upstreaming

roidelapluie, 2021-07-08:
Hello,

Thank you for work work. The driver works correctly.

I am wondering if we can help you to get this upstream. Is it something you plan to do? Can we assist you?

Thanks!

andir, 2021-07-13:
There is already an effort ongoing to upstream this driver. See for example [here](https://lore.kernel.org/linux-wireless/20201230044223.14085-1-pkshih@realtek.com/) for the (AFAIK) initial submission to the ML. They are currently going through revisions of the patchset.

lwfinger, 2021-07-13:
Where do you think I got the source?

andir, 2021-07-13:
On 06:29 13.07.21, lwfinger wrote:
> Where do you think I got the source?

Sorry, I didn't mean to make it sound like there is another effort
ongoing.

I just wanted to clarify for those that opened the issue that there is
already an effort ongoing to get this upstreamed. That effort is the
source of this repository AFAIK.

jorp, 2021-07-13:
also see [this](https://www.spinics.net/lists/linux-wireless/msg212019.html)

closed 2021-07-17 by lwfinger

dotnwat, 2022-12-14:
Just curious about an update on the state of upstreaming. I am using Fedora with a 6.0.10 kernel and it looks like this driver is not available yet. Is it still under review?