lwfinger/rtw89 #18: Wifi modul loads but unable to connect

Maximus1996, 2021-07-14:
Hello,

I tried to compile the packages and everything works fine. I was able to (finally) load the Wifi module but everytime I try to connect to the local network, the Manager shows a message that it fails and keep re-trying without success.

Thank you.

lunamera, 2021-07-14:
Hi,

do you use one access point only or multiple AP (i.e. a mesh network)?

regards

Maximus1996, 2021-07-14:
Thank you. It´s just one access point (The router).

lwfinger, 2021-07-14:
Would you mind sharing the message(s) about the failure?

Maximus1996, 2021-07-14:
Thank you for the reply. Is there a specific command I can run? Because as far as I can tell, it´s just a notification in Gnome saying "connection has failed", but I don´t know what is happening in the background.

lwfinger, 2021-07-14:
You need to run the command 'dmesg | less'. The messages(s) will be near the end.

Maximus1996, 2021-07-14:
The following message keeps repeating:

`[   77.525768] wlo1: authenticate with 74:da:38:fd:20:0f
[   77.706886] wlo1: send auth to 74:da:38:fd:20:0f (try 1/3)
[   77.734321] wlo1: send auth to 74:da:38:fd:20:0f (try 2/3)
[   77.761957] wlo1: send auth to 74:da:38:fd:20:0f (try 3/3)
[   77.790452] wlo1: authentication with 74:da:38:fd:20:0f timed out
[   82.141666] wlo1: authenticate with 74:da:38:fd:20:0f
[   82.322832] wlo1: send auth to 74:da:38:fd:20:0f (try 1/3)
[   82.353773] wlo1: send auth to 74:da:38:fd:20:0f (try 2/3)
[   82.381524] wlo1: send auth to 74:da:38:fd:20:0f (try 3/3)
[   82.410094] wlo1: authentication with 74:da:38:fd:20:0f timed out
[   87.261533] wlo1: authenticate with 74:da:38:fd:20:0f
[   87.444153] wlo1: send auth to 74:da:38:fd:20:0f (try 1/3)
[   87.471576] wlo1: send auth to 74:da:38:fd:20:0f (try 2/3)
[   87.500168] wlo1: send auth to 74:da:38:fd:20:0f (try 3/3)
[   87.528650] wlo1: authentication with 74:da:38:fd:20:0f timed out
[  103.106536] wlo1: authenticate with 74:da:38:fd:20:0f
[  103.285728] wlo1: send auth to 74:da:38:fd:20:0f (try 1/3)
[  103.314353] wlo1: send auth to 74:da:38:fd:20:0f (try 2/3)
[  103.343251] wlo1: send auth to 74:da:38:fd:20:0f (try 3/3)
[  103.372260] wlo1: authentication with 74:da:38:fd:20:0f timed out
[  117.233535] wlo1: authenticate with 74:da:38:fd:20:0f
[  117.414689] wlo1: send auth to 74:da:38:fd:20:0f (try 1/3)
[  117.442764] wlo1: send auth to 74:da:38:fd:20:0f (try 2/3)
[  117.469964] wlo1: send auth to 74:da:38:fd:20:0f (try 3/3)
[  117.497480] wlo1: authentication with 74:da:38:fd:20:0f timed out
[  132.035080] wlo1: authenticate with 74:da:38:fd:20:0f
[  132.215293] wlo1: send auth to 74:da:38:fd:20:0f (try 1/3)
[  132.243994] wlo1: send auth to 74:da:38:fd:20:0f (try 2/3)
[  132.275955] wlo1: send auth to 74:da:38:fd:20:0f (try 3/3)
[  132.303924] wlo1: authentication with 74:da:38:fd:20:0f timed out
[  146.161946] wlo1: authenticate with 74:da:38:fd:20:0f
[  146.342970] wlo1: send auth to 74:da:38:fd:20:0f (try 1/3)
[  146.371028] wlo1: send auth to 74:da:38:fd:20:0f (try 2/3)
[  146.399824] wlo1: send auth to 74:da:38:fd:20:0f (try 3/3)
[  146.430614] wlo1: authentication with 74:da:38:fd:20:0f timed out
lines 1228-1284/1284 (END)
`

lwfinger, 2021-07-14:
That looks like a bad authentication secret. Please do 'dmesg -t > dmesg.txt' and attach file dmesg.txt to this issue.

Maximus1996, 2021-07-14:
[dmesg.txt](https://github.com/lwfinger/rtw89/files/6818216/dmesg.txt)
Done, thanks!

lwfinger, 2021-07-14:
Everything looks normal except for the authentication failures. Are you sure you have the wifi secret correctly entered? If so, then please describe the network including the AP make and model and the type of connection (for example wpa2, wpa3, etc.)

Please run the command 'sudo iw dev wlo1 scan > scan.txt' and attach file scan.txt. Please indicate the AP you are trying to connect.

Maximus1996, 2021-07-14:
Yes, the wifi code is correct. I´m sure because I checked several times and because I noticed that after several reboots it might happen that it connects one time, and then fails again.
It´s my local wifi network, the router is " Edimax Wireless Access Point Wifi Router N300" (https://www.notebooksbilliger.de/edimax+n300+micro+usb+powered+5+in+1+wlan+router), connection type seems to be WPA,WPA2

In the attached files, it should be the first device.
[scan.txt](https://github.com/lwfinger/rtw89/files/6818524/scan.txt)

Thank you again!

lwfinger, 2021-07-14:
Yes, the connection methods are:
	WPA:	 * Version: 1
		 * Group cipher: TKIP
		 * Pairwise ciphers: TKIP
		 * Authentication suites: PSK
	RSN:	 * Version: 1
		 * Group cipher: TKIP
		 * Pairwise ciphers: CCMP
		 * Authentication suites: PSK
		 * Capabilities: 1-PTKSA-RC 1-GTKSA-RC (0x0000)
My AP's all use the following:
        RSN:     * Version: 1
                 * Group cipher: CCMP
                 * Pairwise ciphers: CCMP
                 * Authentication suites: PSK
                 * Capabilities: 16-PTKSA-RC 1-GTKSA-RC (0x000c)
                 * 
Do you control that AP? I wonder about the "Group cipher: TKIP" part of your RSN details.

Why is WPA version 1 enabled? That is so easily cracked that it essentially unsecured.

Maximus1996, 2021-07-14:
I have access to it, but since it was preconfigured by the internet provider of our building and always worked fine, I never changed anything in it.

evilphish, 2021-07-14:
I am having the same problem on a brand new Lenovo T14. Using Kernel 5.11.0 with Linux Mint 20.2. Compilation and installation of wifi module works like a charm, I can see wifi networks but every time I try to connect I get the same "authentication with ... timed out" messages in the log. 

Also 100% sure the secret is correct :)

lwfinger, 2021-07-14:
It may be that there is a problem. Please run 'iw dev' and find your interface name. Then

sudo iw dev <name from above> scan > scan.txt

Attach file scan.txt.

Maximus1996, 2021-07-15:
The response for iw dev is:
`phy#0
	Interface wlo1
		ifindex 2
		wdev 0x1
		addr a2:97:68:8a:66:25
		type managed
		txpower 20.00 dBm
		multicast TXQ:
			qsz-byt	qsz-pkt	flows	drops	marks	overlmt	hashcoltx-bytes	tx-packets
			0	0	0	0	0	0	0	00
`
And attached is the scan.
[scan2.txt](https://github.com/lwfinger/rtw89/files/6820396/scan2.txt)

Pencil-Yao, 2021-07-15:
the command for you @Maximus1996  should be
`sudo iw dev wlo1 scan > scan.txt`

I also have the same problem.

Maximus1996, 2021-07-15:
Oh thanks Pencil-Yao.
Here is an updated file with the command.
[scan3.txt](https://github.com/lwfinger/rtw89/files/6820862/scan3.txt)

evilphish, 2021-07-15:
Also a scan from my side
[scan.txt](https://github.com/lwfinger/rtw89/files/6821079/scan.txt)
SSID to be connected to is PenguinDen (not the Guest one).

sgmiller, 2021-07-15:
Having the same issue.  It worked fine seemingly until the laptop went into standby.  After resuming I get the authentication timeouts.  Also a new Lenovo T14.

sgmiller, 2021-07-15:
Scratch that, same issue after a reboot without standby involved.

lwfinger, 2021-07-15:
@Maximus1996: Your scan shows the same security options, namely

	WPA:	 * Version: 1
		 * Group cipher: TKIP
		 * Pairwise ciphers: TKIP
		 * Authentication suites: PSK
	RSN:	 * Version: 1
		 * Group cipher: TKIP
		 * Pairwise ciphers: CCMP
		 * Authentication suites: PSK
		 * Capabilities: 1-PTKSA-RC 1-GTKSA-RC (0x0000)

I have been unable to set any of my APs to this configuration. I would suspect that the network control software, not the driver, should be the ones to handle this case.

@evilphish: Your security info is

	RSN:	 * Version: 1
		 * Group cipher: CCMP
		 * Pairwise ciphers: CCMP
		 * Authentication suites: PSK FT/PSK
		 * Capabilities: 1-PTKSA-RC 1-GTKSA-RC (0x0000)

That matches mine, except that I have

Capabilities: 16-PTKSA-RC 1-GTKSA-RC (0x000c)

I see multiple APs broadcasting the same SSID. Is this a mesh network?

Maximus1996, 2021-07-15:
> @Maximus1996: Your scan shows the same security options, namely
>
> ```
> WPA:	 * Version: 1
> 	 * Group cipher: TKIP
> 	 * Pairwise ciphers: TKIP
> 	 * Authentication suites: PSK
> RSN:	 * Version: 1
> 	 * Group [...]

Thanks. I´m not sure how I can change that.
Also, the capability that evilphish has is also similar to mine:
1-PTKSA-RC 1-GTKSA-RC (0x0000)
(just in case that helps to norrow things down).

lwfinger, 2021-07-15:
Yes, I knew that. I'm not sure the meaning of that difference.

@Maximus1996: I can confirm that IPv6 fails with this device.

Maximus1996, 2021-07-15:
So I changed encryption from "mixed" to AES in the router options, and the group cipher is also CCMP for me now. But sadly no change on the behavior.

Maximus1996, 2021-07-15:
I´m really sorry for the double post and inconvenience. I wanted to just add that I tried creating a Hotspot using my iPhone and the laptop seems to have no trouble connecting to it and guess what, the capability is set to 16-PTKSA-RC 1-GTKSA-RC (0x000c) for the network that the iPhone created.
I hope this helps maybe that you could understand more the source cause of the problem.

Thank you again for your patience with me.

evilphish, 2021-07-15:
@lwfinger Correct, I run a Unifi setup with 4 access points around the property

sgmiller, 2021-07-15:
FWIW I have a single Asus router, mesh isn't the cause for me.

sgmiller, 2021-07-15:
FWIW my setup is a single Asus RT-AC66U router, no mesh.

evilphish, 2021-07-16:
I tried the phone hotspot test that @Maximus1996 did using a OnePlus Android phone. I can connect to that network just fine. Capabilities are listed as: 16-PTKSA-RC 1-GTKSA-RC (0x000c) for the android hot spot as well.

lwfinger, 2021-07-16:
I have received a new patch from the Realtek developers. Please try it and report back.

evilphish, 2021-07-16:
Very nice, now I can associate to my network!

sgmiller, 2021-07-16:
Yep, worked for me too.

lwfinger, 2021-07-17:
I think this Issue can be closed.

closed 2021-07-17 by lwfinger