lwfinger/rtw89 #19: Anybody else has ipv6 issues with this driver?

madushan1000, 2021-07-15:
The wifi interface doesn't seem to get a ipv6 address when configured with dhcp. 
I tried to manually configure with `dhclient -6 <interface name>` but I can see that the client never gets a response from the gateway and client keep sending request packets. Unfortunately my gateway doesn't let me reserve ipv6 addresses so I can't test with a static address.  
The same router works okay for ipv6/dhcpv6 for all the other devices(android phones and a macbook). I'm just trying to isolate if it's a driver issue before raising a bug report to the kernel mailing list.

Maximus1996, 2021-07-15:
I´m no expert, but it seems to me that you´re facing the same timeout problem others (including me) are facing [here](https://github.com/lwfinger/rtw89/issues/18).

madushan1000, 2021-07-15:
@Maximus1996 actually I should have mentioned in the ticket that I can authenticate with the wifi network and get ipv4 address. I can even serf internet. My problemiis only with ipv6.

lwfinger, 2021-07-15:
I can confirm that IPv6 fails with this driver, but works with an Intel Wireless 7260. I will report this finding upstream.

madushan1000, 2021-07-15:
Oh thanks!

jreusch, 2021-07-16:
It seems i have the same/similar issue on a new Thinkpad E14 Gen3 (AMD) Realtek 8852 , but here i'm using (irrc) slaac and not dhcpv6.

- wifi connection works
- ipv4 works
- i get a link local ipv6 though (fe80::...)

i've tested it with the shipped debian bullseye kernel ((5.10.0-7-amd64) and a custom build (same config) 5.13 from kernel.org

happy to assist if i could help here any further. i'm behind a carrier grade nat for v4 of my provider and the connection quality is really bad/slow.

thanks for providing this module&hope this gets resolved :)

lwfinger, 2021-07-16:
At the moment, the driver has no provisions for IPv6. I will let you know when that feature is added. Your description of "really bad/slow" is not helpful. Can you provide more details? Using a local version of LibreSpeed, I get up and download speeds of roughly 250 Mbps.

madushan1000, 2021-07-16:
BTW latest commit broke wifi for me, it doesn't even detect the card(Thinkpad L14 Gen2(Intel) Realtek 8852) anymore. No reference rtw89 or 8852 in the dmesg. Had to revert back to commit `0f026585e7d7434c0597e4dbd05f160be5b6088d`.
Thanks for looking in to the ipv6 issue :)

jreusch, 2021-07-16:
> Your description of "really bad/slow" is not helpful. Can you provide more details? Using a local version of LibreSpeed, I get up and download speeds of roughly 250 Mbps.

sorry for the confusion, the ipv4 performance of the wifi driver is perfect, it's my internet service provider which has the issues, that's why not having ipv6 currently makes it painful. so nothing to worry here (and the reason why i did not give more details than "bad/slow" :) )

lwfinger, 2021-07-16:
@madushan1000: I see nothing in the patch that should affect detection of the NIC. It initializes some registers in the device that were not previously set, but that only happens after the kernel's probe routines have detected the device, and the driver and its associated firmware are loaded.

madushan1000, 2021-07-16:
So the pci device is actually detected, I can see it when I run lspci, 
```
09:00.0 Network controller: Realtek Semiconductor Co., Ltd. Device 8852
```
the kernel modules get loaded without any issues too. I can also use bluetooth which I believe handled by the same device?. But the bellow lines are absent from dmesg, which are present when the wifi interface is properly detected. And the wifi interface itself if absent when I run `ip link`

```
[    9.223330] rtw89_pci 0000:09:00.0: enabling device (0000 -> 0003)
[    9.226623] rtw89_pci 0000:09:00.0: Firmware version 0.13.24.0, cmd version 0, type 1
[    9.226626] rtw89_pci 0000:09:00.0: Firmware version 0.13.24.0, cmd version 0, type 3
[    9.257904] rtw89_pci 0000:09:00.0: chip rfe_type is 1
[    9.276321] rtw89_pci 0000:09:00.0 wlp9s0: renamed from wlan0
``` 

I guess the firmware gets loaded but it never initialize properly due to new changes? Not sure how to debug this, happy to help if I can.

lwfinger, 2021-07-17:
The first of those lines is logged before the firmware is loaded. Up to that point, the kernel has scanned the device, similar to lspci, detected a device whose probe routine matches the PCI ID found in the scan, and loads the driver. None of that stuff depends on the content of the driver other than that probe routine. For that reason, the changes made in the last commit, which are invoked much later, cannot cause the problem you are having. I would not have the slightest idea what has gone wrong, and I have no idea how to debug it.

The BT device is part of the chip, but it is connected to a special USB hub also on the chip. The PCIe wifi device and the USB hub with attached BT device are separate.

basilarchia, 2021-07-19:
I can also confirm IPv6 doesn't work (ThinkPad T14 Rev2 / Realtek Device 8852). I was able to force fe80:: pings to start working by adding the routes manually:

        ip addr  add fd5d:345:dcd7::a13/128                                 dev wifi0
        ip route add fd5d:345:dcd7::/64                                     dev wifi0
        ip route add fd5d:345:dcd7::/48        via fe80::3023:3ff:fede:5d38 dev wifi0
        ip route add ::/0                      via fe80::3023:3ff:fede:5d38 dev wifi0

at which point network-manager will occasionally finish the dhcp request. I suppose this is similar to hard coding the arp table in IPv4. IPv6 traffic does then work, but it will disconnect easily and/or forget how to do discovery.

My guess is this bug is related to 'ICMP6 neighbor solicitation' and 'ICMP6 neighbor advertisement' packets not actually being sent.

jreusch, 2021-08-11:
Hey hey
just as heads up: one person at upstream is working on a patch on this.

up until now the attached patch fixes the issue for meat least for now.
if no regression are found i'd hope that this will make it's proper way soon into this driver, up until then whoever is interested i just attach it here
```From a3c2e27f4319f495e4bed7f2e382f1c7ca26c1e7 Mon Sep 17 00:00:00 2001
From: root <root>
Date: Thu, 12 Aug 2021 00:03:05 +0200
Subject: [PATCH] Possible fix for ipv6 issue

---
 reg.h | 2 +-
 1 file changed, 1 insertion(+), 1 deletion(-)

diff --git a/reg.h b/reg.h
index bbb0ff0..b6b8b87 100644
--- a/reg.h
+++ b/reg.h
@@ -1274,7 +1274,7 @@
 #define B_AX_SNIFFER_MODE BIT(0)
 #define DEFAULT_AX_RX_FLTR (B_AX_A_A1_MATCH | B_AX_A_BC | B_AX_A_MC |	       \
 			    B_AX_A_UC_CAM_MATCH | B_AX_A_BC_CAM_MATCH |	       \
-			    B_AX_A_MC_LIST_CAM_MATCH |			       \
+                            B_AX_A_PWR_MGNT | B_AX_A_FTM_REQ |                 \
 			    u32_encode_bits(3, B_AX_UID_FILTER_MASK) |	       \
 			    B_AX_A_BCN_CHK_EN)
 
-- 
2.30.2

```

lwfinger, 2021-08-11:
That patch does enable IPv6 capability, at least as tested by https://ipv6-test.com/ 

https://test-ipv6.com/ reports the following:
Danger! IPv6 sorta works - however, large packets appear to fail, giving the appearance of a broken website. If a publisher publishes to IPv6, you will believe their web site to be broken. Ask your ISP about MTU issues; possibly with your tunnel. Check your firewall to make sure that ICMPv6 messages are allowed (in particular, Type 2 or Packet Too Big).

I get that same "error" when I use the wire connection, thus there may be a problem with my TCP/IP stack. In any case, I will push the patch to the repo. Thanks for the patch.

9er, 2021-08-12:
I just tested the latest version (current main is at https://github.com/lwfinger/rtw89/commit/914f028d224f10eb66268376f0e3d506a1714a06) and it still doesn't work for me. Using a 8852 on Thinkpad P14s G2 AMD, kernel 5.13.9.

I have noticed that the device is sending ICMPv6 router solicitation (IPv6 multicast, destination MAC address `33:33:00:00:00:02`) every once in a while, and I can see that packet on the wire (between the WiFi AP and the router). I can also see the response from the router (router advertisement, IPv6 multicast to `33:33:00:00:00:01`). But that packet is not received on the laptop.

IPv6 works - I have only really tested unicast traffic - when I manually configure an address and route, so it looks like the issue is receiving mcast packets.

9er, 2021-08-12:
Whoops, quick update. Apparently `rmmod` + `modprobe` wasn't sufficient. After a reboot it works \o/

```
16:37:58.745094 IP6 fe80::bd0e:a261:5c9a:8586 > ff02::2: ICMP6, router solicitation, length 8
16:37:58.745462 IP6 _gateway > ff02::1: ICMP6, router advertisement, length 112
```
Side note: I don't have any issues with MTU or anything. The IPv6-Test shows 10/10. It is (sadly) quite common for ISPs to filter ICMP in general, because some people believe that ICMP is dangerous somehow. Filtering ICMPv6 (in that case probably "packet too big" messages) is usually not a good idea, and it may be the cause of your IPv6 report.

Thanks for fixing this!

lwfinger, 2021-08-12:
The chip likely needed a power cycle. Glad it works for you.