lwfinger/rtw89 #20: Connection crashes after a short period of time

mogoh, 2021-07-16:
After some minutes the connection crashes. Here is dmsg output:

[dmesg.txt](https://github.com/lwfinger/rtw89/files/6830494/dmesg.txt)

lwfinger, 2021-07-16:
Unfortunately, your dmesg buffer had wrapped around so that I missed the initial cause of failure.

Please use an editor to create (as root) a file named /etc/modprobe.d/50-rtw89.conf and add a single line that says
options rtw89pci disable_aspm_l1ss=y

If that does not help, try changing the last part to disable_aspm_l1=y.

mogoh, 2021-07-19:
I am sorry, I cut off the necessary information.
I have problems reproducing my own problem.
I have tried all options until the network “crashed” again, but that could be of any reason.
Be warned, that this dmesg output was gathered while using my notebook, that might interfere with its output.
So this could be a non-issue.

That being said, if still find the network “crashing” that does not feel like normal connection loss.

[dmesg.txt](https://github.com/lwfinger/rtw89/files/6843930/dmesg.txt)
[dmesg_disable_aspm_l1ss.txt](https://github.com/lwfinger/rtw89/files/6843929/dmesg_disable_aspm_l1ss.txt)
[dmesg_disable_aspm_l1.txt](https://github.com/lwfinger/rtw89/files/6843928/dmesg_disable_aspm_l1.txt)

lwfinger, 2021-07-20:
The kinds of things that you are seeing have generally been caused by some unusual interaction with your BIOS. These kinds of things are usually fixed with one or the other of those options.

This is a problem that I will need to send to Realtek. I'll let you know the response.

lwfinger, 2021-07-20:
The report has been sent to Realtek.

Please try a run with both options set. The line in /etc/modprobe.d/50-rtw89.conf should read

options rtw89pci disable_aspm_l1ss=y disable_aspm_l1=y

lwfinger, 2021-07-21:
I heard back from Realtek already. A trial patch has been added to the code. Do a 'git pull', 'make' and 'sudo make install' and retry. If it still fails, post the dmesg output.

mogoh, 2021-07-22:
Wow, thank you for your fast help. I'll try it and let you know!

mogoh, 2021-08-20:
I have tested the changes for a while it works really well. Thanks Again. I'll close this issue.

closed 2021-08-20 by mogoh