lwfinger/rtw89 #21: Null pointer dereference when disconnecting

pca006132, 2021-07-19:
I was using KDE Plasma with networkmanager. The driver crashed after I clicked disconnect for a network. I was unable to connect to WiFi afterward. I was able to reproduce the issue but only under a specific network environment, after failure connecting to a school WiFi network (eduroam, which uses WPA2 enterprise with PEAP (EAP-MSCHAPv2)).

The kernel version is 5.12.13-xanmod1-cacule. rtw89 version is 501166f.

dmesg output:
<details>

```
[Mon Jul 19 16:25:16 2021] wlan0: deauthenticating from 58:41:20:79:6f:f1 by local choice (Reason: 3=DEAUTH_LEAVING)
[Mon Jul 19 16:25:16 2021] BUG: kernel NULL pointer dereference, address: 0000000000000000
[Mon Jul 19 16:25:16 2021] #PF: supervisor read access in kernel mode
[Mon Jul 19 16:25:16 2021] #PF: error_code(0x0000) - not-present page
[Mon Jul 19 16:25:16 2021] PGD 1171c3067 P4D 1171c3067 PUD 116867067 PMD 0
[Mon Jul 19 16:25:16 2021] Oops: 0000 [#1] SMP NOPTI
[Mon Jul 19 16:25:16 2021] CPU: 2 PID: 1299 Comm: iwd Tainted: G           O      5.12.13-xanmod1-cacule #1-NixOS
[Mon Jul 19 16:25:16 2021] Hardware name: LENOVO 82MS/LNVNB161216, BIOS GZCN14WW 02/03/2021
[Mon Jul 19 16:25:16 2021] RIP: 0010:rtw89_cam_sec_key_del+0x45/0xa0 [rtw89core]
[Mon Jul 19 16:25:16 2021] Code: c2 48 83 c0 06 48 8b ac c6 18 03 00 00 f0 48 0f b3 96 30 03 00 00 48 c7 84 c6 18 03 00 00 00 00 00 00 45 31 e4 45 84 c0 75 26 <0f> b6 45 00 0f b6 d0 f0 48 0f b3 93 d8 02 00 00 f6 45 03 10 75 28
[Mon Jul 19 16:25:16 2021] RSP: 0018:ffffbc72819c38a8 EFLAGS: 00010246
[Mon Jul 19 16:25:16 2021] RAX: 0000000000000000 RBX: ffff9ea55a801f00 RCX: 00000000ffffff02
[Mon Jul 19 16:25:16 2021] RDX: 0000000000000002 RSI: 00000000fffffe01 RDI: ffffffffc1976662
[Mon Jul 19 16:25:16 2021] RBP: 0000000000000000 R08: 0000000000000000 R09: 0000000000000a20
[Mon Jul 19 16:25:16 2021] R10: 0000000000004000 R11: 000000000000005c R12: 0000000000000000
[Mon Jul 19 16:25:16 2021] R13: 0000000000000000 R14: ffff9ea5de720218 R15: ffff9ea55a8020c8
[Mon Jul 19 16:25:16 2021] FS:  00007fab48504740(0000) GS:ffff9ea7dfe80000(0000) knlGS:0000000000000000
[Mon Jul 19 16:25:16 2021] CS:  0010 DS: 0000 ES: 0000 CR0: 0000000080050033
[Mon Jul 19 16:25:16 2021] CR2: 0000000000000000 CR3: 000000011b860000 CR4: 0000000000750ee0
[Mon Jul 19 16:25:16 2021] PKRU: 55555554
[Mon Jul 19 16:25:16 2021] Call Trace:
[Mon Jul 19 16:25:16 2021]  ? rtw89_ops_set_key+0x9f/0x100 [rtw89core]
[Mon Jul 19 16:25:16 2021]  ? drv_set_key+0x6d/0x140 [mac80211]
[Mon Jul 19 16:25:16 2021]  ? ieee80211_key_replace+0x67f/0x830 [mac80211]
[Mon Jul 19 16:25:16 2021]  ? ieee80211_key_free+0x37/0x50 [mac80211]
[Mon Jul 19 16:25:16 2021]  ? ieee80211_del_key+0x82/0xc0 [mac80211]
[Mon Jul 19 16:25:16 2021]  ? __cfg80211_disconnected+0x164/0x3e0 [cfg80211]
[Mon Jul 19 16:25:16 2021]  ? cfg80211_process_deauth+0xc4/0xe0 [cfg80211]
[Mon Jul 19 16:25:16 2021]  ? ieee80211_report_disconnect+0x60/0xb0 [mac80211]
[Mon Jul 19 16:25:16 2021]  ? ieee80211_mgd_deauth.cold+0x68/0x1cb [mac80211]
[Mon Jul 19 16:25:16 2021]  ? cfg80211_mlme_deauth+0xb1/0x1b0 [cfg80211]
[Mon Jul 19 16:25:16 2021]  ? cfg80211_disconnect+0x99/0x1f0 [cfg80211]
[Mon Jul 19 16:25:16 2021]  ? nl80211_disconnect+0x69/0xb0 [cfg80211]
[Mon Jul 19 16:25:16 2021]  ? genl_family_rcv_msg_doit+0xea/0x150
[Mon Jul 19 16:25:16 2021]  ? genl_rcv_msg+0xde/0x1d0
[Mon Jul 19 16:25:16 2021]  ? nl80211_register_mgmt+0xc0/0xc0 [cfg80211]
[Mon Jul 19 16:25:16 2021]  ? genl_get_cmd+0xd0/0xd0
[Mon Jul 19 16:25:16 2021]  ? netlink_rcv_skb+0x50/0xf0
[Mon Jul 19 16:25:16 2021]  ? genl_rcv+0x24/0x40
[Mon Jul 19 16:25:16 2021]  ? netlink_unicast+0x201/0x2c0
[Mon Jul 19 16:25:16 2021]  ? netlink_sendmsg+0x225/0x460
[Mon Jul 19 16:25:16 2021]  ? sock_sendmsg+0x5e/0x60
[Mon Jul 19 16:25:16 2021]  ? __sys_sendto+0xee/0x150
[Mon Jul 19 16:25:16 2021]  ? ep_item_poll.isra.0+0x2d/0x50
[Mon Jul 19 16:25:16 2021]  ? __x64_sys_sendto+0x25/0x30
[Mon Jul 19 16:25:16 2021]  ? do_syscall_64+0x33/0x40
[Mon Jul 19 16:25:16 2021]  ? entry_SYSCALL_64_after_hwframe+0x44/0xae
[Mon Jul 19 16:25:16 2021] Modules linked in: snd_seq_dummy snd_hrtimer snd_seq snd_seq_device af_packet rfcomm bnep uvcvideo btusb btrtl videobuf2_vmalloc btbcm videobuf2_memops videobuf2_v4l2 btintel videobuf2_common bluetooth videodev ecdh_generic ecc crc16 mc xt_MASQUERADE nf_conntrack_netlink nfnetlink xfrm_user xfrm_algo xt_addrtype br_netfilter ccm algif_aead cbc des_generic libdes ecb algif_skcipher hid_sensor_als hid_sensor_trigger industrialio_triggered_buffer kfifo_buf hid_sensor_iio_common industrialio cmac hid_sensor_custom sha512_ssse3 sha512_generic md4 algif_hash hid_sensor_hub af_alg ip6table_nat joydev mousedev iptable_nat intel_rapl_msr nf_nat xt_conntrack nf_conntrack rtw89pci(O) nf_defrag_ipv6 nf_defrag_ipv4 rtw89core(O) ip6t_rpfilter ipt_rpfilter ip6table_raw iptable_raw mac80211 xt_pkttype nf_log_ipv6 snd_hda_codec_realtek nls_iso8859_1 nf_log_ipv4 nls_cp437 nf_log_common edac_mce_amd hid_multitouch snd_hda_codec_generic vfat xt_LOG wmi_bmof hid_generic edac_core xt_tcpudp fat
[Mon Jul 19 16:25:16 2021]  ledtrig_audio snd_hda_codec_hdmi intel_rapl_common crc32_pclmul ghash_clmulni_intel ip6table_filter evdev aesni_intel ip6_tables snd_hda_intel snd_intel_dspcfg libaes crypto_simd snd_intel_sdw_acpi sch_fq_codel snd_hda_codec input_leds iptable_filter mac_hid cryptd cfg80211 rapl deflate snd_hda_core serio_raw ideapad_laptop snd_hwdep msr platform_profile ctr sparse_keymap snd_pcm led_class efi_pstore tpm_crb ucsi_acpi loop tpm_tis typec_ucsi i2c_hid_acpi battery tpm_tis_core i2c_hid tun tpm hid tap video macvlan typec snd_timer veth acpi_cpufreq rfkill bridge roles tiny_power_button wmi snd snd_rn_pci_acp3x libarc4 sp5100_tco rng_core watchdog button soundcore stp llc snd_pci_acp3x vboxnetflt(O) pinctrl_amd i2c_designware_platform i2c_piix4 i2c_designware_core ac vboxnetadp(O) vboxdrv(O) kvm_amd kvm irqbypass fuse pstore configfs efivarfs ip_tables x_tables autofs4 xhci_pci xhci_pci_renesas xhci_hcd usbcore nvme atkbd libps2 nvme_core t10_pi crc_t10dif crct10dif_generic
[Mon Jul 19 16:25:16 2021]  usb_common i8042 crct10dif_pclmul crct10dif_common rtc_cmos serio dm_mod btrfs blake2b_generic libcrc32c crc32c_generic crc32c_intel xor raid6_pq amdgpu drm_ttm_helper ttm iommu_v2 gpu_sched i2c_algo_bit drm_kms_helper syscopyarea sysfillrect sysimgblt fb_sys_fops drm i2c_core backlight agpgart
[Mon Jul 19 16:25:16 2021] CR2: 0000000000000000
[Mon Jul 19 16:25:16 2021] ---[ end trace b0bbdd40ef76e5c6 ]---
[Mon Jul 19 16:25:16 2021] RIP: 0010:rtw89_cam_sec_key_del+0x45/0xa0 [rtw89core]
[Mon Jul 19 16:25:16 2021] Code: c2 48 83 c0 06 48 8b ac c6 18 03 00 00 f0 48 0f b3 96 30 03 00 00 48 c7 84 c6 18 03 00 00 00 00 00 00 45 31 e4 45 84 c0 75 26 <0f> b6 45 00 0f b6 d0 f0 48 0f b3 93 d8 02 00 00 f6 45 03 10 75 28
[Mon Jul 19 16:25:16 2021] RSP: 0018:ffffbc72819c38a8 EFLAGS: 00010246
[Mon Jul 19 16:25:16 2021] RAX: 0000000000000000 RBX: ffff9ea55a801f00 RCX: 00000000ffffff02
[Mon Jul 19 16:25:16 2021] RDX: 0000000000000002 RSI: 00000000fffffe01 RDI: ffffffffc1976662
[Mon Jul 19 16:25:16 2021] RBP: 0000000000000000 R08: 0000000000000000 R09: 0000000000000a20
[Mon Jul 19 16:25:16 2021] R10: 0000000000004000 R11: 000000000000005c R12: 0000000000000000
[Mon Jul 19 16:25:16 2021] R13: 0000000000000000 R14: ffff9ea5de720218 R15: ffff9ea55a8020c8
[Mon Jul 19 16:25:16 2021] FS:  00007fab48504740(0000) GS:ffff9ea7dfe80000(0000) knlGS:0000000000000000
[Mon Jul 19 16:25:16 2021] CS:  0010 DS: 0000 ES: 0000 CR0: 0000000080050033
[Mon Jul 19 16:25:16 2021] CR2: 0000000000000000 CR3: 000000011b860000 CR4: 0000000000750ee0
[Mon Jul 19 16:25:16 2021] PKRU: 55555554
```

</details>

lwfinger, 2021-07-19:
Thanks for your excellent report. It appears as if there is a race condition concerning shutdown. This problem happens infrequently, thus it may take a while to debug.

pca006132, 2021-09-28:
Still occur in 17d957accfb1302cd989b4fb5645c1fd338e2736 with a very similar stack trace.
It happens frequently when the connection is poor. I triggered it during a class for three times :(

lwfinger, 2021-09-28:
I added a check for null pointers in the place I think this happens. Do a 'git pull', 'make', and 'sudo make install' to try the new code. Report any further occurrences.

pca006132, 2021-09-29:
> I added a check for null pointers in the place I think this happens. Do a 'git pull', 'make', and 'sudo make install' to try the new code. Report any further occurrences.

Sorry but it seems that there is no new commit?

lwfinger, 2021-09-29:
Yes, I missed a push in branch min. It is there now.

lwfinger, 2021-10-01:
Did this patch help? The reviews are are now arriving, and I want to send any updates to the developer.

pca006132, 2021-10-04:
> Did this patch help? The reviews are are now arriving, and I want to send any updates to the developer.

Sorry for the late reply, I did not experience any problem recently, but I'm not sure if this is due to the patch or improved network environment at the university.

pca006132, 2021-10-07:
> Did this patch help? The reviews are are now arriving, and I want to send any updates to the developer.

I think this patch fixes the problem. Previously when the WiFi keep disconnecting, it would trigger the null pointer dereference, but this does not happen anymore with this patch. I think maybe we can close this issue for now?