lwfinger/rtw89 #23: Double Wi-Fi icon in status bar with Cinnamon

masemoel, 2021-07-21:
Hello,
I really appreciate your job as your driver fixed my Wi-Fi issues on my laptop with Ubuntu Cinnamon Remix 21.04.
I have the following issue: when I have Wi-Fi working there are two icons in my Cinnamon status bar. They disappear when I update my kernel (so no Wi-Fi until I reinstall the driver) nor with other PCs with other Wi-Fi drivers, so I guess it's something related to this specific driver. One of those icons appear as a XApp, and the other one is the default network icon.

lwfinger, 2021-07-21:
This "problem" is due to something in your Ubuntu Cinnamon user space. On my openSUSE Tumbleweed system with NetworkManager controlling the network devices, I see only a single network icon no matter which connection I am using.

No one else has reported this kind of issue.

masemoel, 2021-07-21:
Ok, keep up the good work, I'll try to find the fix for this weird bug then ;)

closed 2021-07-21 by masemoel