lwfinger/rtw89 #24: Broken module after resume

Vliro, 2021-07-23:
I have the latest IdeaPad 5 Pro with 5800H and the relevant wifi module. The module works just fine, however after resuming from sleep I lose wifi with many error outputs to the kernel log.

Output from `sudo dmesg | grep rtw89`:

`[  349.403734] rtw89_pci 0000:03:00.0: can't change power state from D3hot to D0 (config space inaccessible)
[  351.483789] WARNING: CPU: 14 PID: 2059 at /home/albert/rtw89/pci.c:1454 rtw89_dbi_read8.constprop.0+0xdf/0xf0 [rtw89pci]
[  351.483808] Modules linked in: ccm uinput snd_seq_dummy snd_hrtimer snd_seq snd_seq_device rfcomm hid_logitech_hidpp cmac algif_hash algif_skcipher af_alg hid_logitech_dj bnep joydev mousedev uvcvideo videobuf2_vmalloc videobuf2_memops videobuf2_v4l2 btusb videobuf2_common btrtl btbcm intel_rapl_msr btintel videodev intel_rapl_common hid_sensor_custom mc bluetooth snd_hda_codec_realtek ecdh_generic snd_hda_codec_generic rtw89pci(OE) ecc hid_sensor_hub hid_multitouch edac_mce_amd usbhid crc16 wmi_bmof ledtrig_audio rtw89core(OE) snd_hda_codec_hdmi amdgpu snd_hda_intel kvm_amd snd_intel_dspcfg mac80211 gpu_sched snd_intel_sdw_acpi vfat i2c_algo_bit fat drm_ttm_helper kvm cfg80211 snd_hda_codec ideapad_laptop ttm platform_profile irqbypass snd_hda_core sparse_keymap crct10dif_pclmul snd_hwdep drm_kms_helper crc32_pclmul ghash_clmulni_intel snd_pcm aesni_intel ucsi_acpi cec snd_timer ccp typec_ucsi syscopyarea crypto_simd snd sysfillrect typec sp5100_tco cryptd snd_rn_pci_acp3x rfkill
[  351.483991] RIP: 0010:rtw89_dbi_read8.constprop.0+0xdf/0xf0 [rtw89pci]
[  351.484056]  rtw89_pci_resume+0x14b/0x1d0 [rtw89pci]
[  351.484456] WARNING: CPU: 14 PID: 2059 at /home/albert/rtw89/pci.c:1454 rtw89_dbi_read8.constprop.0+0xdf/0xf0 [rtw89pci]
[  351.484466] Modules linked in: ccm uinput snd_seq_dummy snd_hrtimer snd_seq snd_seq_device rfcomm hid_logitech_hidpp cmac algif_hash algif_skcipher af_alg hid_logitech_dj bnep joydev mousedev uvcvideo videobuf2_vmalloc videobuf2_memops videobuf2_v4l2 btusb videobuf2_common btrtl btbcm intel_rapl_msr btintel videodev intel_rapl_common hid_sensor_custom mc bluetooth snd_hda_codec_realtek ecdh_generic snd_hda_codec_generic rtw89pci(OE) ecc hid_sensor_hub hid_multitouch edac_mce_amd usbhid crc16 wmi_bmof ledtrig_audio rtw89core(OE) snd_hda_codec_hdmi amdgpu snd_hda_intel kvm_amd snd_intel_dspcfg mac80211 gpu_sched snd_intel_sdw_acpi vfat i2c_algo_bit fat drm_ttm_helper kvm cfg80211 snd_hda_codec ideapad_laptop ttm platform_profile irqbypass snd_hda_core sparse_keymap crct10dif_pclmul snd_hwdep drm_kms_helper crc32_pclmul ghash_clmulni_intel snd_pcm aesni_intel ucsi_acpi cec snd_timer ccp typec_ucsi syscopyarea crypto_simd snd sysfillrect typec sp5100_tco cryptd snd_rn_pci_acp3x rfkill
[  351.484621] RIP: 0010:rtw89_dbi_read8.constprop.0+0xdf/0xf0 [rtw89pci]
[  351.484682]  rtw89_pci_resume+0x166/0x1d0 [rtw89pci]
[  351.485005] WARNING: CPU: 14 PID: 2059 at /home/albert/rtw89/pci.c:1454 rtw89_dbi_read8.constprop.0+0xdf/0xf0 [rtw89pci]
[  351.485015] Modules linked in: ccm uinput snd_seq_dummy snd_hrtimer snd_seq snd_seq_device rfcomm hid_logitech_hidpp cmac algif_hash algif_skcipher af_alg hid_logitech_dj bnep joydev mousedev uvcvideo videobuf2_vmalloc videobuf2_memops videobuf2_v4l2 btusb videobuf2_common btrtl btbcm intel_rapl_msr btintel videodev intel_rapl_common hid_sensor_custom mc bluetooth snd_hda_codec_realtek ecdh_generic snd_hda_codec_generic rtw89pci(OE) ecc hid_sensor_hub hid_multitouch edac_mce_amd usbhid crc16 wmi_bmof ledtrig_audio rtw89core(OE) snd_hda_codec_hdmi amdgpu snd_hda_intel kvm_amd snd_intel_dspcfg mac80211 gpu_sched snd_intel_sdw_acpi vfat
[  351.485249] RIP: 0010:rtw89_dbi_read8.constprop.0+0xdf/0xf0 [rtw89pci]
[  351.485290]  rtw89_pci_link_cfg+0x89/0xc0 [rtw89pci]
[  351.485296]  rtw89_pci_resume+0x172/0x1d0 [rtw89pci]
[  351.485354] rtw89_pci 0000:03:00.0: failed to set CLKREQ_L1, ret=-5
[  353.486193] rtw89_pci 0000:03:00.0: [ERR] Polling timeout
[  353.486200] rtw89_pci 0000:03:00.0: [ERR] addr: 5, 5
[  353.486204] rtw89_pci 0000:03:00.0: [ERR] val: FF, 0
[  355.486357] rtw89_pci 0000:03:00.0: [ERR] Polling timeout
[  355.486363] rtw89_pci 0000:03:00.0: [ERR] addr: 5, 5
[  355.486367] rtw89_pci 0000:03:00.0: [ERR] val: FF, 0
[  357.486979] rtw89_pci 0000:03:00.0: [ERR] Polling timeout
[  357.486999] rtw89_pci 0000:03:00.0: [ERR] addr: 5, 5
[  357.487003] rtw89_pci 0000:03:00.0: [ERR] val: FF, 0
[  359.487205] rtw89_pci 0000:03:00.0: [ERR] Polling timeout
[  359.487213] rtw89_pci 0000:03:00.0: [ERR] addr: 5, 5
[  359.487217] rtw89_pci 0000:03:00.0: [ERR] val: FF, 0
[  359.487220] rtw89_pci 0000:03:00.0: mac init fail, ret:-16
[  359.487341] Modules linked in: ccm uinput snd_seq_dummy snd_hrtimer snd_seq snd_seq_device rfcomm hid_logitech_hidpp cmac algif_hash algif_skcipher af_alg hid_logitech_dj bnep joydev mousedev uvcvideo videobuf2_vmalloc videobuf2_memops videobuf2_v4l2 btusb videobuf2_common btrtl btbcm intel_rapl_msr btintel videodev intel_rapl_common hid_sensor_custom mc bluetooth snd_hda_codec_realtek ecdh_generic snd_hda_codec_generic rtw89pci(OE) ecc hid_sensor_hub hid_multitouch edac_mce_amd usbhid crc16 wmi_bmof ledtrig_audio rtw89core(OE) snd_hda_codec_hdmi amdgpu snd_hda_intel kvm_amd snd_intel_dspcfg mac80211 gpu_sched snd_intel_sdw_acpi vfat i2c_algo_bit fat drm_ttm_helper kvm cfg80211 snd_hda_codec ideapad_laptop ttm platform_profile irqbypass snd_hda_core sparse_keymap crct10dif_pclmul snd_hwdep drm_kms_helper crc32_pclmul ghash_clmulni_intel snd_pcm aesni_intel ucsi_acpi cec snd_timer ccp typec_ucsi syscopyarea crypto_simd snd sysfillrect typec sp5100_tco cryptd snd_rn_pci_acp3x rfkill
[  359.488159] Modules linked in: ccm uinput snd_seq_dummy snd_hrtimer snd_seq snd_seq_device rfcomm hid_logitech_hidpp cmac algif_hash algif_skcipher af_alg hid_logitech_dj bnep joydev mousedev uvcvideo videobuf2_vmalloc videobuf2_memops videobuf2_v4l2 btusb videobuf2_common btrtl btbcm intel_rapl_msr btintel videodev intel_rapl_common hid_sensor_custom mc bluetooth snd_hda_codec_realtek ecdh_generic snd_hda_codec_generic rtw89pci(OE) ecc hid_sensor_hub hid_multitouch edac_mce_amd usbhid crc16 wmi_bmof ledtrig_audio rtw89core(OE) snd_hda_codec_hdmi amdgpu snd_hda_intel kvm_amd snd_intel_dspcfg mac80211 gpu_sched snd_intel_sdw_acpi vfat i2c_algo_bit fat drm_ttm_helper kvm cfg80211 snd_hda_codec ideapad_laptop ttm platform_profile irqbypass snd_hda_core sparse_keymap crct10dif_pclmul snd_hwdep drm_kms_helper crc32_pclmul ghash_clmulni_intel snd_pcm aesni_intel ucsi_acpi cec snd_timer ccp typec_ucsi syscopyarea crypto_simd snd sysfillrect typec sp5100_tco cryptd snd_rn_pci_acp3x rfkill
[  359.488996] Modules linked in: ccm uinput snd_seq_dummy snd_hrtimer snd_seq snd_seq_device rfcomm hid_logitech_hidpp cmac algif_hash algif_skcipher af_alg hid_logitech_dj bnep joydev mousedev uvcvideo videobuf2_vmalloc videobuf2_memops videobuf2_v4l2 btusb videobuf2_common btrtl btbcm intel_rapl_msr btintel videodev intel_rapl_common hid_sensor_custom mc bluetooth snd_hda_codec_realtek ecdh_generic snd_hda_codec_generic rtw89pci(OE) ecc hid_sensor_hub hid_multitouch edac_mce_amd usbhid crc16 wmi_bmof ledtrig_audio rtw89core(OE) snd_hda_codec_hdmi amdgpu snd_hda_intel kvm_amd snd_intel_dspcfg mac80211 gpu_sched snd_intel_sdw_acpi vfat i2c_algo_bit fat drm_ttm_helper kvm cfg80211 snd_hda_codec ideapad_laptop ttm platform_profile irqbypass snd_hda_core sparse_keymap crct10dif_pclmul snd_hwdep drm_kms_helper crc32_pclmul ghash_clmulni_intel snd_pcm aesni_intel ucsi_acpi cec snd_timer ccp typec_ucsi syscopyarea crypto_simd snd sysfillrect typec sp5100_tco cryptd snd_rn_pci_acp3x rfkill
[  361.769352] rtw89_pci 0000:03:00.0: [ERR] Polling timeout
[  361.769359] rtw89_pci 0000:03:00.0: [ERR] addr: 5, 5
[  361.769363] rtw89_pci 0000:03:00.0: [ERR] val: FF, 0
[  363.769690] rtw89_pci 0000:03:00.0: [ERR] Polling timeout
[  363.769697] rtw89_pci 0000:03:00.0: [ERR] addr: 5, 5
[  363.769702] rtw89_pci 0000:03:00.0: [ERR] val: FF, 0
[  365.769716] rtw89_pci 0000:03:00.0: [ERR] Polling timeout
[  365.769723] rtw89_pci 0000:03:00.0: [ERR] addr: 5, 5
[  365.769727] rtw89_pci 0000:03:00.0: [ERR] val: FF, 0
[  367.770975] rtw89_pci 0000:03:00.0: [ERR] Polling timeout
[  367.770983] rtw89_pci 0000:03:00.0: [ERR] addr: 5, 5
[  367.770987] rtw89_pci 0000:03:00.0: [ERR] val: FF, 0
[  367.770990] rtw89_pci 0000:03:00.0: mac init fail, ret:-16
[  369.775363] rtw89_pci 0000:03:00.0: [ERR] Polling timeout
[  369.775371] rtw89_pci 0000:03:00.0: [ERR] addr: 5, 5
[... cut ...]
[  387.784317] rtw89_pci 0000:03:00.0: [ERR] val: FF, 0
[  389.784360] rtw89_pci 0000:03:00.0: [ERR] Polling timeout
[  389.784367] rtw89_pci 0000:03:00.0: [ERR] addr: 5, 5
[  389.784372] rtw89_pci 0000:03:00.0: [ERR] val: FF, 0
[  391.784523] rtw89_pci 0000:03:00.0: [ERR] Polling timeout
[  391.784531] rtw89_pci 0000:03:00.0: [ERR] addr: 5, 5
[  391.784535] rtw89_pci 0000:03:00.0: [ERR] val: FF, 0
[  391.784539] rtw89_pci 0000:03:00.0: mac init fail, ret:-16
`

In order to have Wifi again I have to reboot, at which point it works fine.

lwfinger, 2021-07-23:
That message "can't change power state from D3hot" usually occurs with a buggy BIOS.

The work-around is to add a script that unloads the modules on entering sleep/hibernation, and reloads upon resumption. Fortunately, systemd provides the necessary hook. Do a 'git pull' and the following command:

sudo cp suspend_rtw89 /usr/lib/systemd/system-sleep/.

That should handle your problem.

Vliro, 2021-07-23:
Hi lwfinger,

I just changed some stuff in bios for some other unrelated thing and this time it did work upon resume. One thing of note is that I changed from "Intelligent Cooling" preset in BIOS to Performance, although no guarantee that was the cause. 

The suspension you mentioned I will also try out. If I see no issue I'll close the issue. Thanks for your response!

lwfinger, 2021-07-23:
Yes, it was in the v5 branch. I have now added it to the main branch, but as long as your BIOS changes fixed the problem, you can ignore the script.

Vliro, 2021-07-25:
Hi again lwfinger,

After a few days of no issues I encountered a broken module again, this time after letting the laptop sleep for a few hours.

Output from running command `sudo dmesg | grep rtw89 > file`

> [38882.552777] rtw89_pci 0000:03:00.0: Firmware version 0.13.8.0, cmd version 0, type 1
[38882.552825] rtw89_pci 0000:03:00.0: Firmware version 0.13.8.0, cmd version 0, type 3
[38882.570452] rtw89_pci 0000:03:00.0: chip rfe_type is 1
[38882.581087] rtw89_pci 0000:03:00.0 wlo1: renamed from wlan0
[45422.166758] rtw89_pci 0000:03:00.0: Firmware version 0.13.8.0, cmd version 0, type 1
[45422.166771] rtw89_pci 0000:03:00.0: Firmware version 0.13.8.0, cmd version 0, type 3
[45422.183683] rtw89_pci 0000:03:00.0: chip rfe_type is 1
[45422.194135] rtw89_pci 0000:03:00.0 wlo1: renamed from wlan0
[87981.483455] rtw89_pci 0000:03:00.0: Firmware version 0.13.8.0, cmd version 0, type 1
[87981.483475] rtw89_pci 0000:03:00.0: Firmware version 0.13.8.0, cmd version 0, type 3
[87981.499765] rtw89_pci 0000:03:00.0: chip rfe_type is 1
[87981.516827] rtw89_pci 0000:03:00.0 wlo1: renamed from wlan0
[127162.508641] rtw89_pci 0000:03:00.0: no suitable firmware found
[127162.508648] rtw89_pci 0000:03:00.0: failed to recognize firmware
[127162.508652] rtw89_pci 0000:03:00.0: failed to setup chip information
[127162.509801] rtw89_pci: probe of 0000:03:00.0 failed with error -2
[127230.168443] rtw89_pci 0000:03:00.0: no suitable firmware found
[127230.168452] rtw89_pci 0000:03:00.0: failed to recognize firmware
[127230.168456] rtw89_pci 0000:03:00.0: failed to setup chip information
[127230.169720] rtw89_pci: probe of 0000:03:00.0 failed with error -2

lwfinger, 2021-07-25:
The sleep time length cannot make a difference. Your BIOS is still broken, just not as badly as before.

closed 2023-10-02 by lwfinger