lwfinger/rtw89 #26: Bluetooth issues

aelray, 2021-07-30:
Hi,

I'm using the driver provided here and it works flawlessly on my Debian (11) system for wifi. Has anybody tested Bleutooth should it work? I'm getting a bluetooth listed with:
```
root@thinkpadL15:~# lsusb | grep Realtek
Bus 005 Device 003: ID 0bda:4852 Realtek Semiconductor Corp. Bluetooth Radio
```

However with `bluetoothctl` I can't scan for devices and can't find the device e.g. from my mobile.

erikhubers, 2021-08-01:
Can confirm, having the same issue on Ubuntu 20.04.2

bedrankara, 2021-08-01:
Bluetooth works perfectly for me on Fedora 34 with 5.13 Kernel

lwfinger, 2021-08-01:
Interesting. On my openSUSE Tumbleweed system, the device was not recognized, and the BT driver was not loaded. When I added USB ID 04ca:4006 to the kernel tables, the device was recognized and the firmware was loaded; however, the device still did not work!

I sent an Email to Realtek, but I have not heard back.

lwfinger, 2021-08-04:
Device 04ca:4006 was not in the device tables. The following patch, which has been accepted, and will be propagated to all stable kernels. adds the device

[add_id.txt](https://github.com/lwfinger/rtw89/files/6932491/add_id.txt)

Of course, if you do not generate a custom kernel, the new device should be available soon in your distro's kernel.

ryan1288, 2021-08-08:
Hello,
I have the exact same issue as aelray:
```
lsusb | grep Realtek
Bus 003 Device 003: ID 0bda:4852 Realtek Semiconductor Corp. 
```
Really appreciate what you're doing to improve compatibility with our system. I'm not very knowledgeable in this field. Does "Of course, if you do not generate a custom kernel, the new device should be available soon in your distro's kernel." mean that it'll come as a pop-up update for the Ubuntu distro for my 5.4 kernel soon?

Secondly, would that fix the problem? Since you stated "the device still did not work!" even if it was recognized. Should I be concerned about bluetooth compatibility in the coming months?

Thank you so much.

lwfinger, 2021-08-08:
No, your device is unlikely to "popup" in a 5.4 kernel. As the driver is in kernel review, I cannot predict when it will be merged, but the earliest will be 5.16. When Ubuntu reaches the point where that kernel is their base, then it will be available without building an out-of-kernel driver.

I did, in fact, get the Bluetooth part to work. It requires a patch to the kernel code, which will be part of kernel 5.15, otherwise your device will not work. For anyone that wishes to use the BT part of their device, the patch is attached.
[Realtek_BT_patch.txt](https://github.com/lwfinger/rtw89/files/6951132/Realtek_BT_patch.txt)

closed 2021-08-08 by lwfinger