lwfinger/rtw89 #28: Won't compile

charlesrwest, 2021-08-03:
Upon hitting make, I first get a message about /lib/modules/5.10.53-1-MANJARO/build not being present.  After making that directory, I get:

make -C /lib/modules/5.10.53-1-MANJARO/build M=/home/charlesrwest/cpp/libraries/rtw89 modules
make[1]: Entering directory '/usr/lib/modules/5.10.53-1-MANJARO/build'
make[1]: *** No rule to make target 'modules'.  Stop.
make[1]: Leaving directory '/usr/lib/modules/5.10.53-1-MANJARO/build'
make: *** [Makefile:52: all] Error 2

I'm not really good at debugging make files.  Is there something wrong with my system?

lwfinger, 2021-08-03:
Yes. It is not sufficient to create the build directory. It also needs to point to the kernel headers. You need to install the kernel header package. A quick search of the web says that the correct command is 'sudo pacman -S linux-headers'. That information came from https://linuxconfig.org/manjaro-linux-kernel-headers-installation/.

charlesrwest, 2021-08-03:
That did it.  Thank you!

closed 2021-08-03 by lwfinger

shadiakiki1986, 2021-12-02:
I'm on debian 11, ran `apt-get install linux-headers-amd64`, rebooted, and am still getting `No rule to make target 'modules'`. What else do I need to do?