lwfinger/rtw89 #29: Sporadic de-authentication and disconnecting (4WAY_HANDSHAKE_TIMEOUT)

covertg, 2021-08-04:
Firstly, thanks so much for all of your work on this module.

On a Lenovo 16ACH6 ("Ideapad 5 Pro 16"), I am experiencing somewhat regular de-authentication which breaks the connection. This is on the latest Arch Linux, using NetworkManager in GNOME. The module is installed with via [`rtw89-dkms-git`](https://aur.archlinux.org/packages/rtw89-dkms-git/) in the AUR. The access point in question is a pretty common Netgear, C3700-100NAS using WPA2 personal.

```
$ uname -a                                                                                                                                                                                              ~
Linux opia 5.13.7-arch1-1 #1 SMP PREEMPT Sat, 31 Jul 2021 13:18:52 +0000 x86_64 GNU/Linux

$ lspci | grep -i net                                                                                                                                                                               ~
03:00.0 Network controller: Realtek Semiconductor Co., Ltd. Device 8852

$ pacman -Qs rtw89                                                   ~
local/rtw89-dkms-git r87.febb639-1
    Driver for Realtek 8852AE, an 802.11ax device

# dmidecode -s bios-version                                                                                                                         ~
GSCN24WW
```

This behavior is sporadic, so I haven't been yet able to neatly replicate it. Sometimes an hour or two will go by with no issue. Other times it will disconnect within a couple minutes of logging in, or shortly after reconnecting.

I've attached a couple of full dmesg logs, one from a disconnection a few minutes after startup, and another disconnection sometime after waking up from suspend. Could lines such as `wlo1: deauthenticated from 3c:37:86:56:56:ab (Reason: 15=4WAY_HANDSHAKE_TIMEOUT)` be indicators here?

[dmesg-startup.log](https://github.com/lwfinger/rtw89/files/6928135/dmesg-startup.log)
[dmesg-wakeup.log](https://github.com/lwfinger/rtw89/files/6928137/dmesg-wakeup.log)

Happy to provide more info or test around if it'd help.

lwfinger, 2021-08-04:
I am seeing some problems with authenticating. That may be due to low signal levels. Please run the following and post the output:

sudo iw dev wlo1 scan | egrep "signal|SSID"

covertg, 2021-08-04:
Sure thing:
```
	signal: -64.00 dBm
	SSID: TP-Link_E030
	signal: -49.00 dBm
	SSID: PizzaParty24
	signal: -73.00 dBm
	SSID: freeloaders
	signal: -55.00 dBm
	SSID: ARRIS-CD69
	signal: -82.00 dBm
	SSID: PizzaParty24
		 * SSID List
	signal: -89.00 dBm
	SSID: DIRECT-5E-HP ENVY 5660 series
	signal: -64.00 dBm
	SSID: ARRIS-CD69-5G
	signal: -52.00 dBm
	SSID: 
		 * SSID List
	signal: -84.00 dBm
	SSID: 
		 * SSID List
	signal: -82.00 dBm
	SSID: 
		 * SSID List
	signal: -84.00 dBm
	SSID: 705naranja
	signal: -84.00 dBm
	SSID: 
		 * SSID List
	signal: -83.00 dBm
	SSID: RobertT
		 * SSID List
	signal: -83.00 dBm
	SSID: Hanna
		 * SSID List
	signal: -84.00 dBm
	SSID: 
		 * SSID List
	signal: -84.00 dBm
	SSID: 
		 * SSID List
	signal: -93.00 dBm
	SSID: 705naranja_EXT
	signal: -78.00 dBm
	SSID: NETGEAR17
	signal: -73.00 dBm
	SSID: 
	signal: -74.00 dBm
	SSID: Rice
		 * SSID List
	signal: -74.00 dBm
	SSID: XFINITY
		 * SSID List
	signal: -52.00 dBm
	SSID: Stevens
		 * SSID List
	signal: -74.00 dBm
	SSID: 
		 * SSID List
	signal: -52.00 dBm
	SSID: 
		 * SSID List
	signal: -83.00 dBm
	SSID: 
		 * SSID List
	signal: -46.00 dBm
	SSID: 
		 * SSID List
	signal: -71.00 dBm
	SSID: TP-Link_E030_5G
	signal: -74.00 dBm
	SSID: 
		 * SSID List
	signal: -74.00 dBm
	SSID: massimo
	signal: -78.00 dBm
	SSID: 
		 * SSID List
	signal: -79.00 dBm
	SSID: 
		 * SSID List
	signal: 0.00 dBm
	SSID: PizzaParty24
		 * SSID List
	signal: -74.00 dBm
	SSID: XFINITY
	signal: -74.00 dBm
	SSID: \x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00
	signal: -83.00 dBm
	SSID: Modem
		 * SSID List
	signal: -73.00 dBm
	SSID: apt2
	signal: -56.00 dBm
	SSID: 
		 * SSID List
	signal: -73.00 dBm
	SSID: 
	signal: -73.00 dBm
	SSID: 
	signal: -83.00 dBm
	SSID: \x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00
	signal: -73.00 dBm
	SSID: 
	signal: -82.00 dBm
	SSID: 
		 * SSID List
	signal: -84.00 dBm
	SSID: UBNT2G
	signal: -46.00 dBm
	SSID: 
		 * SSID List
	signal: -46.00 dBm
	SSID: Stevens
		 * SSID List
	signal: -91.00 dBm
	SSID: NETGEAR36
		 * SSID List
	signal: -70.00 dBm
	SSID: apt2
	signal: -71.00 dBm
	SSID: 
	signal: -71.00 dBm
	SSID: 
	signal: -71.00 dBm
	SSID: xfinitywifi
	signal: -82.00 dBm
	SSID: RobertT
		 * SSID List
	signal: -89.00 dBm
	SSID: 
		 * SSID List
	signal: -80.00 dBm
	SSID: 
		 * SSID List
	signal: -81.00 dBm
	SSID: 
		 * SSID List
	signal: -69.00 dBm
	SSID: PizzaParty50
	signal: -73.00 dBm
	SSID: freeloaders
	signal: -78.00 dBm
	SSID: massimo
	signal: -73.00 dBm
	SSID: 
	signal: -79.00 dBm
	SSID: \x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00
	signal: -78.00 dBm
	SSID: \x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00
	signal: -82.00 dBm
	SSID: Krubenaganstein
		 * SSID List
	signal: -88.00 dBm
	SSID: Hotspot380D
	signal: -71.00 dBm
	SSID: 
	signal: -92.00 dBm
```
Crowded space over here. The ssid in question is "freeloaders" (-55dbm)

covertg, 2021-08-04:
A few more details:
You're right, it does seem like this may be related to signal levels. When close to the access point (<10ft), there seems to be no issue. With my phone hotspot nearby, also no issue. Getting just a little bit further from the AP seems to be the struggle—the issue occurs even when just ~20ft and a couple of small rooms away, which is not what I'd expect.

I will say, that in my case it does seem like this AP is not strong. However, when booted into Windows from the same location the laptop can keep a stable connection, and other devices seem to connect to the AP fine from this location as well.

lwfinger, 2021-08-04:
In the list, the signal is logged before the SSID. The freeloaders AP has a strength of -73 dBm - very weak!

As this driver is new, there will likely be advances that help stability. If Windows can do it, then the hardware can.

If you run a speed test, what value do you get? Using a local LibreTest server at about 1m, I get 300 Mbps down and 493 up. The AP in question is a Linksys EA6900 AC1900. Switching to the 2G band using the same router, I get 115 down and 109 up. Switching to a Netgear WNDR4500 N900 router at 2m, I get 102 down and 121 up. Switching to the 2G band on that router, I get 36 Mbps down and 30 up. My 2G space is as crowded as yours, but given the short-range of 5G, I have those bands to myself.

I am particularly happy with that upload speed to the AC1900 router. That is nearly half the speed of my gigabit wire! The fact that the RX rate is significantly less than TX shows that there is more speed  to be gained, and likely more stability with weak signals.

covertg, 2021-08-06:
Aha, thanks for catching me on the iw scan. LibreSpeed is neat - here's the results of a few quick tests, but apologies in advance for a lot more data than was asked for.

I ran the server on a separate device, kept stationary ~10ft away from the AP. Then I tried a few speedtests on the 8852 device from a few different locations, retrying in both Windows and Linux.

Room 1:
```
Down	Up	Ping	Jitter

[Linux]
[2.4 GHz -58dbm]
98.4	101	15	51.3
109	69.6	15	2.66
107	98.8	16	30.3
[5GHz -60dbm]
161 	118	15	6.58
162	130	13	192
192	132	18	355

[Windows]
[2.4 GHz -64dbm]
104	77.5	1	372
102	84.5	11	277
104	58.8	9	4.94
[5 GHz -70 dbm]
109	145	10	347
179	141	14	64.6
174	131	12	17.5
```

Room 2:
```
Down	Up	Ping	Jitter

[Linux]
[2.4GHz -62dbm]
77.5	91.8	17	132
70	92.5	16	197
76.6	97.9	15	50.7	
[5GHz -76dbm]
25.6	8.68	15	3.78
29.2	10.7	15	123.0
21.2	12.7	15	13.1

[Windows]
[2.4 GHz -78dbm]
55.4	50.2	8	4.44
78.1	66.3	10	121
69.6	51.5	9	6.67
[5GHz -81dbm]
34	63	11	1.98
57	54.2	11	2.59
59.9	53.3	10	2.48
```

Room 3:
```
Down	Up	Ping	Jitter

[Linux]
[2.4GHz -80dbm]
5.51	2.69	19	61.1
5.09	7.54	14	22.3
2.85	2.81	16	176
[5GHz n/a]

[Windows]
[2.4GHz -86 dbm]
6.16	10.7	9	8.56
7.6	10.4	14	4.09
7.36	6.85	11	171
[5GHz n/a]
```

"Room 1" is the closest to the router, then "Room 2," then "Room 3." Certainly this could be more rigorous (wishing now that I had wired the server into the AP), but a few interesting notes I see:

* Comparing Windows to Linux when close to the router seems on-par, at least in terms of upload/download. Nice!
* The tx/rx point you made is a cool observation, but it seems more complicated for me. Up close, it does seem that TX > RX. But at a distance, it seems to vary.
* The Room 3 data seems to indicate how Windows is able to maintain a more stable connection (despite, yes, the quite bad signal).

(Also, the rssi is measured in windows with [vistumbler](https://www.vistumbler.net/), so it might not make sense to directly compare the reported signal strength across Windows-Linux.

For whatever that's worth! I do hope this might help the developers pinpoint stability issues specifically related to weaker connections.

fschrempf, 2021-08-21:
I'm also seeing deauthentication issues on my Lenovo Thinkpad P14s AMD Gen2. As described above it happens in places with low signal. A different device with an Intel WiFi 6 AX200 in the same spot with the same poor signal strength keeps the connection up forever, while with the Realtek 8852AE it fails sooner or later.

In contrast to the reports above, in my case I don't see `4WAY_HANDSHAKE_TIMEOUT`, but rather `PREV_AUTH_NOT_VALID`. Therefore it asks for the key again. Disabling and re-enabling the connection is enough to make it work for some time again. I've seen this behavior with the v5 and v6 branch.

```
Aug 21 13:59:31 fs-work kernel: wlp3s0: associated
Aug 21 13:59:31 fs-work NetworkManager[574]: <info>  [1629547171.6284] device (wlp3s0): supplicant interface state: associating -> 4way_handshake
Aug 21 13:59:35 fs-work kernel: wlp3s0: deauthenticated from aa:bb:cc:dd:ee:ff (Reason: 2=PREV_AUTH_NOT_VALID)
Aug 21 13:59:35 fs-work wpa_supplicant[3451]: wlp3s0: CTRL-EVENT-DISCONNECTED bssid=aa:bb:cc:dd:ee:ff reason=2
Aug 21 13:59:35 fs-work wpa_supplicant[3451]: wlp3s0: WPA: 4-Way Handshake failed - pre-shared key may be incorrect
Aug 21 13:59:35 fs-work wpa_supplicant[3451]: wlp3s0: CTRL-EVENT-SSID-TEMP-DISABLED id=0 ssid="Test" auth_failures=1 duration=10 reason=WRONG_KEY
Aug 21 13:59:35 fs-work wpa_supplicant[3451]: wlp3s0: CTRL-EVENT-REGDOM-CHANGE init=CORE type=WORLD
Aug 21 13:59:35 fs-work NetworkManager[574]: <info>  [1629547175.6154] device (wlp3s0): supplicant interface state: 4way_handshake -> disconnected
Aug 21 13:59:35 fs-work NetworkManager[574]: <info>  [1629547175.6156] device (wlp3s0): Activation: (wifi) disconnected during association, asking for new key
Aug 21 13:59:35 fs-work NetworkManager[574]: <info>  [1629547175.6157] device (wlp3s0): state change: activated -> need-auth (reason 'supplicant-disconnect', sys-iface-state: 'managed')
```

closed 2023-10-02 by lwfinger