lwfinger/rtw89 #30: Slow transfer?

domenkozar, 2021-08-04:
```
$ speedtest
Retrieving speedtest.net configuration...
Testing from A1 Slovenija (146.212.93.147)...
Retrieving speedtest.net server list...
Selecting best server based on ping...
Hosted by Telemach d.o.o. (Ljubljana) [0.36 km]: 3.872 ms
Testing download speed................................................................................
Download: 31.23 Mbit/s
Testing upload speed......................................................................................................
Upload: 22.27 Mbit/s
```

On the same wifi, from the same location, my phone can do 400Mbits/100Mbits

lwfinger, 2021-08-05:
Using a local LibreTest server at about 1m, I get 300 Mbps down and 493 up. The AP in question is a Linksys EA6900 AC1900. My iPhone gets 491 down and 453 up. Yes, the phone is a bit better, but not that much.

What signal strength do you have? The driver does tend to slow with weak signals, but there is little I can do about that. I am a volunteer that makes this driver available.

domenkozar, 2021-08-05:
```
wlp3s0    IEEE 802.11  ESSID:"XXX"  
          Mode:Managed  Frequency:5.5 GHz  Access Point: 2C:3A:FD:45:58:49   
          Bit Rate=585.1 Mb/s   Tx-Power=20 dBm   
          Retry short limit:7   RTS thr:off   Fragment thr:off
          Power Management:off
          Link Quality=55/70  Signal level=-55 dBm  
          Rx invalid nwid:0  Rx invalid crypt:0  Rx invalid frag:0
          Tx excessive retries:0  Invalid misc:7096   Missed beacon:0
```

Even if I move closer the speed doesn't improve much.

Note that I'm using https://github.com/lwfinger/rtw89/commit/cebafc6dc839e66c725b92c0fabf131bc908f607

Don't take this issue as that you have to fix it :)

domenkozar, 2021-08-06:
Upgrading to latest master doesn't improve the speed.

Artturin, 2021-08-07:
Try setting the wifi country 
```nix
  networking = {
    localCommands =
      ''
        ${pkgs.iw}/bin/iw reg set SI
      '';
  };
```

Non-nixos
https://wiki.archlinux.org/title/Network_configuration/Wireless#Respecting_the_regulatory_domain

domenkozar, 2021-08-10:
I'm in touch with realtek developers now, running some tests.

@Artturin thanks for the suggestion, that didn't help.

I get reasonable speed on 2.4GHz networks, so it has to do something with 5GHz - @lwfinger which one did you run the tests on?

domenkozar, 2021-08-11:
[image]

With these settings on the AP I get 200mbit/100mbit, but it's not reproducible what exactly triggered the change, need to do more testing.

domenkozar, 2021-08-11:
I've used WiFi4 as a test, going back to WiFi5 as it doesn't make a difference.

I found out that if I just reconnect to the wifi, 1 out of 10 times I get 500mbps while 9 out of 10 times I get 30mbps connection.

How do I debug what is different?

domenkozar, 2021-08-11:
Currently:

```
$ speedtest
Retrieving speedtest.net configuration...
Testing from A1 Slovenija (146.212.73.11)...
Retrieving speedtest.net server list...
Selecting best server based on ping...
Hosted by Bostjan Industries Inc. (Ljubljana) [0.36 km]: 4.309 ms
Testing download speed................................................................................
Download: 466.67 Mbit/s
Testing upload speed......................................................................................................
Upload: 95.54 Mbit/s
```

lwfinger, 2021-08-11:
I use a local LibreSpeed instance connected to my router with Gigabit cable. With it, I get a download speed of 600 Mbps, and upload of 380 Mbps. These results are obtained using the 5GHz band and connecting to a WiFi5 router about 1m from my computer. In the much more crowded 2.4 GHz, I get 81 Mbps down and 54 up. Using Speedtest.net, my broadband connection would limit me to 200/12.

Connecting to a WiFi4 router at 2m from my computer using the 5GHz band, I get 218/204 Mbps - definitely slower than the WiFi6 router. Using the 2.4GHz band of that router, I get 95/65.

Soulwest, 2021-09-10:
I have the same issue (speed more than 10 times lower than on a nearby phone). What info do you need to debug?

domenkozar, 2021-09-16:
Gave up - ordered Intel AX200.

Ledarium, 2022-03-01:
I am getting incredibly low bitrate on 5Ghz iperf test, like 10-12 mbits. OpenWRT router shows bitrates as such:
[image]
Other devices can do 100+ mbits on same network. No error messages in dmesg. Any updates on this?

lwfinger, 2022-03-01:
I get speeds of ~800 Mbps on a 5G connection to my 802.11ac router, but I have signal strengths of -40 dBm. That -66 is pretty low.

Can you get closer to the router?

This kind of question is outside my knowledge. You need to follow the instructions of the last paragraph of README.md.

Ledarium, 2022-03-01:
> Can you get closer to the router?
Yes, I've tried. No change in speed.

aathan, 2022-08-02:
I'm seeing 50% the transfer speed on a simple scp test between a mac os x host and a lenovo thinkpad running rtw89_pci compared with the same transfer between the same mac os x host and a mac pro laptop, all on the same wifi network (5G)

```
  *-network
       description: Wireless interface
       product: RTL8852AE 802.11ax PCIe Wireless Network Adapter
       vendor: Realtek Semiconductor Co., Ltd.
       physical id: 0
       bus info: pci@0000:03:00.0
       logical name: wlo1
       version: 00
       serial: 14:5a:fc:01:c2:xx
       width: 64 bits
       clock: 33MHz
       capabilities: pm msi pciexpress bus_master cap_list ethernet physical wireless
       configuration: broadcast=yes driver=rtw89_pci driverversion=5.17.0-1014-oem firmware=N/A ip=192.168.x.x latency=0 link=yes multicast=yes wireless=IEEE 802.11
       resources: irq:64 ioport:2000(size=256) memory:d1600000-d16fffff
 ```

lwfinger, 2022-08-02:
I tried your scp test with a 204 MB file. Using the 5G band on a wifi5 router (AC1900), I got 49.2 MBps (399 Mbps) for uploads and 64.0 MBps (511 Mbps) for downloads. The other end was a desktop connected to the network via a wire. Using my local speedtest, I got 587 Mbps down and 481 Mbps up. With file reads and writes, scp should be a little slower than speedtest. My signal reported by 'iw dev ... scan' is -45 dBm.

You did not tell me what your kernel is, but from the name (rtw89_pci), I see that you are using the in-kernel version. The one from this repo, which I am using, would be rtw89pci with no underscore.

closed 2023-10-02 by lwfinger