lwfinger/rtw89 #32: Power Consumption on Thinkpad P14s AMD G2

stettberger, 2021-08-06:
Hi!

I'm using the latest HEAD (febb6391e7e6ca8f04adbf2af56ad123defb4d77) with an Debian unstable kernel (Linux obelix 5.10.0-8-amd64 ). And it looks like the driver/the device requires a lot of energy. At least powertop says:

```
Power est.              Usage       Events/s    Category       Description
  22.7 W     39.5 pkts/s                Device         Network interface: wlp3s0 (rtw89_pci)
```

In the dmesg, I see:

```
[33713.163160] rtw89_pci 0000:03:00.0: firmware: direct-loading firmware rtw89/rtw8852a_fw.bin
[33713.164878] rtw89_pci 0000:03:00.0: Firmware version 0.13.24.0, cmd version 0, type 1
[33713.164880] rtw89_pci 0000:03:00.0: Firmware version 0.13.24.0, cmd version 0, type 3
[33713.184564] rtw89_pci 0000:03:00.0: chip rfe_type is 1
[33713.189958] rtw89_pci 0000:03:00.0 wlp3s0: renamed from wlan0
[33721.789295] wlp3s0: authenticate with f0:b0:14:a0:e9:83
[33721.985640] wlp3s0: send auth to f0:b0:14:a0:e9:83 (try 1/3)
[33721.987043] wlp3s0: authenticated
[33721.987233] wlp3s0: associate with f0:b0:14:a0:e9:83 (try 1/3)
[33721.990160] wlp3s0: RX AssocResp from f0:b0:14:a0:e9:83 (capab=0x1511 status=0 aid=1)
[33722.098853] wlp3s0: associated
[33722.098884] wlp3s0: Limiting TX power to 20 (23 - 3) dBm as advertised by f0:b0:14:a0:e9:83
```

23W seems a little much for my understanding of mobile devices. Is this a known issue?

lwfinger, 2021-08-06:
I was not aware of PowerTop.. On my system (7-year old Toshiba i7 laptop), powertop shows power consumption about 7.5W. Running s flood ping to my router with over 1000 packets per sec, the power usage goes to 10W. It appears that the  usage of the 8852AE is about 3W when if is busy sending and receiving. 

On my system, the 8852AE is on an extender. It does not feel warm. I never found out how to get the wattage use by devices with powertop.

stettberger, 2021-08-09:
I'm not sure if this power consumption is an actual issue. However, I have the problem that the Laptop drains (for my feeling) too much energy and the rtw89 driver is the only driver that I use from a secondary source.

My device is build in and lspci says:

```
03:00.0 Network controller: Realtek Semiconductor Co., Ltd. Device 8852
	Subsystem: Lenovo Device 4852
	Physical Slot: 0
	Flags: bus master, fast devsel, latency 0, IRQ 88, IOMMU group 13
	I/O ports at 3000 [size=256]
	Memory at fd700000 (64-bit, non-prefetchable) [size=1M]
	Capabilities: [40] Power Management version 3
	Capabilities: [50] MSI: Enable+ Count=1/1 Maskable- 64bit+
	Capabilities: [70] Express Endpoint, MSI 00
	Capabilities: [100] Advanced Error Reporting
	Capabilities: [148] Device Serial Number 00-e0-4c-ff-fe-88-52-01
	Capabilities: [158] Latency Tolerance Reporting
	Capabilities: [160] L1 PM Substates
	Kernel driver in use: rtw89_pci
	Kernel modules: rtw89pci
```

I'm not sure how I can give more useful information about this.

covertg, 2021-08-11:
@stettberger, perhaps try recalibrating powertop to make sure that its calculations are relatively accurate? From my understanding, powertop's wattage calculations are all estimates, which rely on it taking a lot of readings from the device while going through different types of load (all while on battery power). See:
* https://access.redhat.com/documentation/en-us/red_hat_enterprise_linux/8/html/monitoring_and_managing_system_status_and_performance/managing-power-consumption-with-powertop_monitoring-and-managing-system-status-and-performance
* https://01.org/sites/default/files/page/powertop_users_guide_201412.pdf

I had also been experiencing significant battery drain on a new ideapad x AMD laptop, so did some exploring with powertop as well. For me though, the primary culprit was hybrid graphics. Setting up [tlp](https://linrunner.de/tlp/) helped a good bit as well.

Now, after running `powertop --calibrate` (and letting powertop run in the background for some hours after that by now, as well), the ~idle power consumption reported by powertop for rtw89_pci on a ranges from 1.4-1.5W (this is on 2.4ghz, a bit away from the router). A ping flood to my router (on 5ghz, nearby) pulls 2.4-2.8W with about 550 packets/sec.

stettberger, 2021-08-11:
I'm using it in the sense that bluetooth is activated. However, when I deactivate it in the UI, such that the device disappears in  `hcitool dev`, strange things happened. First of all: The reported energy consumption did not change.

But my device became hell of unstable. `iwlist sc` sporadically said that it cannot see any networks. All networks reported with a signal strength of 0 db and my notebook could not authenticate to the accesspoint anymore. It said that the AP did not answer after 3 tries: timeout.

Reloading the modules did not help. Rebooting did not help. I almost went crazy, as there was no indicator why things were going downhill.

Then I rebooted to Windows, connected to a Wifi, and everything ist fine by now.

domenkozar, 2023-05-29:
Possible fix https://patchwork.kernel.org/project/linux-wireless/patch/20230527082939.11206-3-pkshih@realtek.com/

lwfinger, 2023-05-29:
If you had looked at 'git log', you would have seen that the patch was merged last Saturday.

I work closely with Ping-Ke Shih, khew that the patch was coming, and merged it here as soon as he pushed it to wireless-next@vger.kernel.org.

closed 2023-05-29 by lwfinger