lwfinger/rtw89 #33: Are version tags planned in this repo?

realnickel, 2021-08-10:
Hi!
Thank you for your work.
I'd like to create an RPM package with the driver and wish to know which versioning scheme do you plan to use?
Would it be "v5", "v6" or something different?
Do you plan to tag releases?

lwfinger, 2021-08-10:
Until the driver is accepted into mainline, I will create a new branch to match the "revision" level of the submission. After testing, that code will be copied into the main branch, thus main is the only branch you need to use. There will not be  an official version, and the releases will not be tagged. Once the driver is accepted into mainline, only the main branch will remain, and updates to the kernel code will be ported to this repo. Set up your RPM spec file to pull this repo.

realnickel, 2021-08-11:
Thanks a lot.

closed 2021-08-11 by realnickel