lwfinger/rtw89 #34: system hangs with task wpa_supplicant:973 blocked for more than 120 seconds

rouckas, 2021-08-12:
I am not sure if this is caused by the rtw89 driver, but I have no other idea. Maybe you will understand what is going on. My configuration is: Lenovo Yoga Slim 7 Pro 14ACH5 with ubuntu 21.04. It happens to me apparently randomly about once a week of daily work. The system becomes partly unresponsive - GUI freezes, I can switch to terminal, where I see with htop that Firefox is stuck in D (disk sleep) state and in syslog I see the output below. Apparently, the frozen tasks are all in the D state, waiting for some I/O. Could this be related to the wifi driver?
```
Aug 12 10:41:42 yoga-stepan kernel: [103459.236435] INFO: task NetworkManager:936 blocked for more than 120 seconds.
Aug 12 10:41:42 yoga-stepan kernel: [103459.237242]       Tainted: G        W  OE     5.11.0-25-generic #27-Ubuntu
Aug 12 10:41:42 yoga-stepan kernel: [103459.238030] "echo 0 > /proc/sys/kernel/hung_task_timeout_secs" disables this message.
Aug 12 10:41:42 yoga-stepan kernel: [103459.238836] task:NetworkManager  state:D stack:    0 pid:  936 ppid:     1 flags:0x00000000
Aug 12 10:41:42 yoga-stepan kernel: [103459.238841] Call Trace:
Aug 12 10:41:42 yoga-stepan kernel: [103459.238845]  __schedule+0x23d/0x670
Aug 12 10:41:42 yoga-stepan kernel: [103459.238851]  ? ttwu_do_activate+0x6e/0xd0
Aug 12 10:41:42 yoga-stepan kernel: [103459.238856]  schedule+0x4f/0xc0
Aug 12 10:41:42 yoga-stepan kernel: [103459.238858]  schedule_preempt_disabled+0xe/0x10
Aug 12 10:41:42 yoga-stepan kernel: [103459.238860]  __mutex_lock.constprop.0+0x2ce/0x4a0
Aug 12 10:41:42 yoga-stepan kernel: [103459.238863]  __mutex_lock_slowpath+0x13/0x20
Aug 12 10:41:42 yoga-stepan kernel: [103459.238865]  mutex_lock+0x34/0x40
Aug 12 10:41:42 yoga-stepan kernel: [103459.238867]  rtnl_lock+0x15/0x20
Aug 12 10:41:42 yoga-stepan kernel: [103459.238872]  nl80211_dump_station+0x3b/0x240 [cfg80211]
Aug 12 10:41:42 yoga-stepan kernel: [103459.238906]  ? __mod_lruvec_page_state+0x69/0x80
Aug 12 10:41:42 yoga-stepan kernel: [103459.238910]  ? __kmalloc_node_track_caller+0x235/0x2b0
Aug 12 10:41:42 yoga-stepan kernel: [103459.238913]  ? kmem_cache_alloc_node+0x10e/0x220
Aug 12 10:41:42 yoga-stepan kernel: [103459.238916]  ? netlink_dump+0xb0/0x340
Aug 12 10:41:42 yoga-stepan kernel: [103459.238920]  ? ksize+0x14/0x20
Aug 12 10:41:42 yoga-stepan kernel: [103459.238923]  ? __alloc_skb+0xa4/0x200
Aug 12 10:41:42 yoga-stepan kernel: [103459.238927]  netlink_dump+0x17d/0x340
Aug 12 10:41:42 yoga-stepan kernel: [103459.238930]  __netlink_dump_start+0x212/0x2f0
Aug 12 10:41:42 yoga-stepan kernel: [103459.238932]  genl_family_rcv_msg_dumpit+0x8f/0x110
Aug 12 10:41:42 yoga-stepan kernel: [103459.238935]  ? genl_family_rcv_msg_doit+0x150/0x150
Aug 12 10:41:42 yoga-stepan kernel: [103459.238938]  ? nl80211_send_station+0xef0/0xef0 [cfg80211]
Aug 12 10:41:42 yoga-stepan kernel: [103459.238963]  ? genl_family_rcv_msg_dumpit+0x110/0x110
Aug 12 10:41:42 yoga-stepan kernel: [103459.238966]  genl_family_rcv_msg+0xeb/0x160
Aug 12 10:41:42 yoga-stepan kernel: [103459.238968]  ? nl80211_dump_station+0x240/0x240 [cfg80211]
Aug 12 10:41:42 yoga-stepan kernel: [103459.238993]  ? nl80211_send_station+0xef0/0xef0 [cfg80211]
Aug 12 10:41:42 yoga-stepan kernel: [103459.239017]  genl_rcv_msg+0x4c/0xa0
Aug 12 10:41:42 yoga-stepan kernel: [103459.239020]  ? genl_family_rcv_msg+0x160/0x160
Aug 12 10:41:42 yoga-stepan kernel: [103459.239023]  netlink_rcv_skb+0x55/0x100
Aug 12 10:41:42 yoga-stepan kernel: [103459.239025]  genl_rcv+0x29/0x40
Aug 12 10:41:42 yoga-stepan kernel: [103459.239027]  netlink_unicast+0x221/0x330
Aug 12 10:41:42 yoga-stepan kernel: [103459.239030]  netlink_sendmsg+0x221/0x440
Aug 12 10:41:42 yoga-stepan kernel: [103459.239033]  sock_sendmsg+0x65/0x70
Aug 12 10:41:42 yoga-stepan kernel: [103459.239037]  ____sys_sendmsg+0x257/0x2a0
Aug 12 10:41:42 yoga-stepan kernel: [103459.239039]  ? import_iovec+0x31/0x40
Aug 12 10:41:42 yoga-stepan kernel: [103459.239043]  ? sendmsg_copy_msghdr+0x7e/0xa0
Aug 12 10:41:42 yoga-stepan kernel: [103459.239045]  ? __pollwait+0xd0/0xd0
Aug 12 10:41:42 yoga-stepan kernel: [103459.239048]  ___sys_sendmsg+0x82/0xc0
Aug 12 10:41:42 yoga-stepan kernel: [103459.239050]  ? __pollwait+0xd0/0xd0
Aug 12 10:41:42 yoga-stepan kernel: [103459.239052]  ? __pollwait+0xd0/0xd0
Aug 12 10:41:42 yoga-stepan kernel: [103459.239054]  ? __fget_files+0x5f/0x90
Aug 12 10:41:42 yoga-stepan kernel: [103459.239058]  ? __fget_light+0x32/0x80
Aug 12 10:41:42 yoga-stepan kernel: [103459.239061]  __sys_sendmsg+0x62/0xb0
Aug 12 10:41:42 yoga-stepan kernel: [103459.239063]  __x64_sys_sendmsg+0x1f/0x30
Aug 12 10:41:42 yoga-stepan kernel: [103459.239065]  do_syscall_64+0x38/0x90
Aug 12 10:41:42 yoga-stepan kernel: [103459.239068]  entry_SYSCALL_64_after_hwframe+0x44/0xa9
Aug 12 10:41:42 yoga-stepan kernel: [103459.239071] RIP: 0033:0x7f4016717f5d
Aug 12 10:41:42 yoga-stepan kernel: [103459.239073] RSP: 002b:00007ffecb67efe0 EFLAGS: 00000293 ORIG_RAX: 000000000000002e
Aug 12 10:41:42 yoga-stepan kernel: [103459.239076] RAX: ffffffffffffffda RBX: 00007ffecb67f0b0 RCX: 00007f4016717f5d
Aug 12 10:41:42 yoga-stepan kernel: [103459.239078] RDX: 0000000000000000 RSI: 00007ffecb67f020 RDI: 000000000000000b
Aug 12 10:41:42 yoga-stepan kernel: [103459.239079] RBP: 0000558dc76d9300 R08: 0000000000000000 R09: 0000000000000011
Aug 12 10:41:42 yoga-stepan kernel: [103459.239080] R10: 0000558dc76388e0 R11: 0000000000000293 R12: 0000558dc76d9300
Aug 12 10:41:42 yoga-stepan kernel: [103459.239082] R13: 0000558dc757fd80 R14: 00007ffecb67f2f0 R15: 00007ffecb67f2f4
Aug 12 10:41:42 yoga-stepan kernel: [103459.239095] INFO: task wpa_supplicant:973 blocked for more than 120 seconds.
Aug 12 10:41:42 yoga-stepan kernel: [103459.239906]       Tainted: G        W  OE     5.11.0-25-generic #27-Ubuntu
Aug 12 10:41:42 yoga-stepan kernel: [103459.240699] "echo 0 > /proc/sys/kernel/hung_task_timeout_secs" disables this message.
Aug 12 10:41:42 yoga-stepan kernel: [103459.241444] task:wpa_supplicant  state:D stack:    0 pid:  973 ppid:     1 flags:0x00004000
Aug 12 10:41:42 yoga-stepan kernel: [103459.241448] Call Trace:
Aug 12 10:41:42 yoga-stepan kernel: [103459.241450]  __schedule+0x23d/0x670
Aug 12 10:41:42 yoga-stepan kernel: [103459.241454]  schedule+0x4f/0xc0
Aug 12 10:41:42 yoga-stepan kernel: [103459.241456]  schedule_preempt_disabled+0xe/0x10
Aug 12 10:41:42 yoga-stepan kernel: [103459.241458]  __mutex_lock.constprop.0+0x2ce/0x4a0
Aug 12 10:41:42 yoga-stepan kernel: [103459.241460]  ? __mutex_unlock_slowpath.constprop.0+0xc7/0x120
Aug 12 10:41:42 yoga-stepan kernel: [103459.241463]  __mutex_lock_slowpath+0x13/0x20
Aug 12 10:41:42 yoga-stepan kernel: [103459.241465]  mutex_lock+0x34/0x40
[... 135 lines cut ...]
Aug 12 10:41:42 yoga-stepan kernel: [103459.248125]  __mutex_lock.constprop.0+0x2ce/0x4a0
Aug 12 10:41:42 yoga-stepan kernel: [103459.248127]  ? rtw89_traffic_stats_calc+0x1a0/0x1a0 [rtw89core]
Aug 12 10:41:42 yoga-stepan kernel: [103459.248136]  __mutex_lock_slowpath+0x13/0x20
Aug 12 10:41:42 yoga-stepan kernel: [103459.248138]  mutex_lock+0x34/0x40
Aug 12 10:41:42 yoga-stepan kernel: [103459.248141]  ieee80211_iterate_interfaces+0x2d/0x60 [mac80211]
Aug 12 10:41:42 yoga-stepan kernel: [103459.248174]  rtw89_track_work+0x91/0x130 [rtw89core]
Aug 12 10:41:42 yoga-stepan kernel: [103459.248183]  process_one_work+0x220/0x3c0
Aug 12 10:41:42 yoga-stepan kernel: [103459.248187]  worker_thread+0x50/0x370
Aug 12 10:41:42 yoga-stepan kernel: [103459.248189]  kthread+0x12f/0x150
Aug 12 10:41:42 yoga-stepan kernel: [103459.248192]  ? process_one_work+0x3c0/0x3c0
Aug 12 10:41:42 yoga-stepan kernel: [103459.248194]  ? __kthread_bind_mask+0x70/0x70
Aug 12 10:41:42 yoga-stepan kernel: [103459.248197]  ret_from_fork+0x22/0x30
Aug 12 10:41:42 yoga-stepan systemd-resolved[882]: Using degraded feature set UDP instead of TCP for DNS server 138.232.1.4.
Aug 12 10:41:47 yoga-stepan systemd-resolved[882]: Using degraded feature set TCP instead of UDP for DNS server 138.232.1.4.
Aug 12 10:41:58 yoga-stepan systemd-resolved[882]: Using degraded feature set UDP instead of TCP for DNS server 138.232.1.4.
Aug 12 10:42:03 yoga-stepan systemd-resolved[882]: Using degraded feature set TCP instead of UDP for DNS server 138.232.1.4.
Aug 12 10:42:23 yoga-stepan systemd-resolved[882]: Using degraded feature set UDP instead of TCP for DNS server 138.232.1.4.
[... cut ...]
Aug 12 10:43:05 yoga-stepan systemd-resolved[882]: Using degraded feature set TCP instead of UDP for DNS server 138.232.1.4.
Aug 12 10:43:25 yoga-stepan systemd-resolved[882]: Using degraded feature set UDP instead of TCP for DNS server 138.232.1.4.
Aug 12 10:43:30 yoga-stepan systemd-resolved[882]: Using degraded feature set TCP instead of UDP for DNS server 138.232.1.4.
Aug 12 10:43:43 yoga-stepan kernel: [103580.008660] INFO: task NetworkManager:936 blocked for more than 241 seconds.
```
etc...

lwfinger, 2021-08-19:
This is a new complaint. The system is clearly doing wifi work, the calls from cfg80211 and the involvement of wpa_supplicant confirm that. The involvement of rtw89 does not appear until several sets of dumps later. I find that only the first is reliable.

What complicates matters is that when rtw89 passes a packet off to the mac80211 code (including cfg80211), the packet is complete, thus cfg80211 is not waiting for data from this driver.

Is your system fully updated? As wpa_supplicant gets its information from NetworkManager through Dbus calls, I would suspect some problem with the supplicant, NetworkManager, or Dbus.

rouckas, 2021-10-13:
FYI, this problem doesn't occur on my system since I switched from Wayland to Xorg.

neon12345, 2021-11-27:
I have this issue using RTL8852AE on Debian Bullseye with Wayland and kernel Linux 5.10.0-9-amd64 #1 SMP Debian 5.10.70-1 (2021-09-30) x86_64 GNU/Linux

The same system (ASUS notebook) works fine with the Intel AX200.

lwfinger, 2021-11-27:
That clearly indicates a PCI bus problem that are usually due to BIOS bugs.. Most of these are fixed with modules options. As root, create file /etc/modprobe.d/50-rtw89.conf containing the following line:
options rtw89pci disable_aspm_l1=y disable_aspm_l1ss=y

You may not need both options, but try both first.

neon12345, 2021-11-28:
The options do not fix the problem for me. I have compiled my kernel with lockdep to check for mutex deadlocks but I have no experience using it.

lwfinger, 2021-11-28:
You need to report this issue, with as many pertinent logs as possible, to the mailing list linux-wireless@vger.kernel.org. Two things to note: Such a posting will expose your E-mail address in a public forum, and you must submit plain text - no html mail.

There are no static mutex deadlocks in the driver. My kernel also has all such tests enabled. If anything is detected, it will be logged in the dmesg output.

neon12345, 2021-11-28:
Is this possibly related to RTNL locking? [some RTNL patch](https://patchwork.kernel.org/project/linux-wireless/patch/1383730993-18259-1-git-send-email-johannes@sipsolutions.net/)

I have no problems so far running with a small test fix and no options, but I do not know if it is valid:

In rtw89_traffic_stats_calc@core.c replace

`ieee80211_queue_work(rtwdev->hw, &rtwdev->btc.wl_sta_notify_work);`

with

`schedule_work(&rtwdev->btc.wl_sta_notify_work);`

lwfinger, 2021-11-29:
I have no idea what making that change would do. I am not the author of the driver - just the one that makes it available for people with older kernels.

It is hard to believe that there is a locking problem with rtw89. I have run the driver hard starting with the v3 submission, and have never seen a wpa_supplicant lockup such as you report. My BIOS is solid, and I never need to run any options.

neon12345, 2021-11-29:
From version v5 to v7 there was a change in rtw89_traffic_stats_calc that removed the ieee80211_queue_work. Perhaps a newer kernel will work then for me.

lwfinger, 2021-11-29:
Are you not using v7 already?

closed 2021-11-29 by lwfinger

neon12345, 2021-11-29:
No, I normally start with the master branch.