lwfinger/rtw89 #35: Non buildable on ix86 as of git.febb639

realnickel, 2021-08-12:
Build log part: 

```
[00:00:08]   LD [M]  /usr/src/RPM/BUILD/kernel-source-rtw89-0.0/rtw89core.o
[00:00:08] make[1]: *** Deleting file '/usr/src/RPM/BUILD/kernel-source-rtw89-0.0/Module.symvers'
[00:00:08]   MODPOST /usr/src/RPM/BUILD/kernel-source-rtw89-0.0/Module.symvers
[00:00:08] ERROR: modpost: "__udivdi3" [/usr/src/RPM/BUILD/kernel-source-rtw89-0.0/rtw89core.ko] undefined!
[00:00:08] make[1]: *** [scripts/Makefile.modpost:124: /usr/src/RPM/BUILD/kernel-source-rtw89-0.0/Module.symvers] Error 1
[00:00:08] make: *** [Makefile:1726: modules] Error 2
[00:00:08] error: Bad exit status from /usr/src/tmp/rpm-tmp.1605 (%build)
```

Objdump gives us a hint where to search the caller:

```

$ objdump -Sr rtw89core.o |grep "__udivdi3"
     fd0:       e8 fc ff ff ff          call   fd1 <rtw89_traffic_stats_calc+0x121> fd1: R_386_PC32  __udivdi3
    1051:       e8 fc ff ff ff          call   1052 <rtw89_traffic_stats_calc+0x1a2>1052: R_386_PC32 __udivdi3
```

Calls are from core.c:1522
```
stats->tx_avg_len = (u32)(stats->tx_cnt ? stats->tx_unicast / stats->tx_cnt : 0);
stats->rx_avg_len = (u32)(stats->rx_cnt ? stats->rx_unicast / stats->rx_cnt : 0);
```
We may cast u64 args to u32 before division to prevent compiler to choose __udivdi3 to perform 64-bit division, and code will compile fine, but loss of precision may occur I guess.

Another options are described here:

https://stackoverflow.com/questions/55999728/how-to-fix-linker-warning-function-undefined-in-kernel-space

 
```
[...]
    The compiler will generate code that calls functions in libgcc in order to perform certain operations. 
    In this case, they are being called to perform the regular division (/) and modulo (%) operations on 64-bit quantities.
    Correct, the kernel doesn't use libgcc. 
    Kernel code needs to be written to prevent the compiler making calls to functions in libgcc (in this case for division and module operations on 64-bit quantities).
[...]
    For new code, it is better to avoid the old do_div(n,base) macro and make use of the functions declared by #include <linux/math64.h>. 
    There are various functions defined there for signed and unsigned arguments and for both 32-bit and 64-bit remainders.   You need to pick the most appropriate function to match the requirements of your code.
```

lwfinger, 2021-08-12:
I am not sure why one should avoid do_div(). The latest kernel uses it roughly 1300 times.

Does the following patch fix your problem? It fixes the build on a 32-bit PowerPC platform. Note that I cannot test on the old Mac.

diff --git a/core.c b/core.c
index 636a9ac..bcc4d9a 100644
--- a/core.c
+++ b/core.c
@@ -1519,8 +1519,8 @@ static void rtw89_traffic_stats_calc(struct rtw89_dev *rtwdev,
                                                   stats->tx_cnt);
        stats->rx_tfc_lv = rtw89_get_traffic_level(rtwdev, stats->rx_throughput,
                                                   stats->rx_cnt);
-       stats->tx_avg_len = (u32)(stats->tx_cnt ? stats->tx_unicast / stats->tx_cnt : 0);
-       stats->rx_avg_len = (u32)(stats->rx_cnt ? stats->rx_unicast / stats->rx_cnt : 0);
+       stats->tx_avg_len = (u32)(stats->tx_cnt ? do_div(stats->tx_unicast, stats->tx_cnt) : 0);
+       stats->rx_avg_len = (u32)(stats->rx_cnt ? do_div(stats->rx_unicast, stats->rx_cnt) : 0);
 
        stats->tx_unicast = 0;
        stats->rx_unicast = 0;

lwfinger, 2021-08-12:
It has been added.

closed 2021-08-12 by lwfinger