lwfinger/rtw89 #36: poor latency performance

paulmil1, 2021-08-16:
Lenovo legion 5 15ach6h, fresh new ubuntu install, didnt have wifi, so i installed the driver from here 

`  *-network                 
       description: Ethernet interface
       product: RTL8111/8168/8411 PCI Express Gigabit Ethernet Controller
       vendor: Realtek Semiconductor Co., Ltd.
       physical id: 0
       bus info: pci@0000:03:00.0
       logical name: eno1
       version: 15
       serial: 38:f3:ab:b7:a4:33
       capacity: 1Gbit/s
       width: 64 bits
       clock: 33MHz
       capabilities: pm msi pciexpress msix bus_master cap_list ethernet physical tp mii 10bt 10bt-fd 100bt 100bt-fd 1000bt-fd autonegotiation
       configuration: autonegotiation=on broadcast=yes driver=r8169 driverversion=5.11.0-25-generic firmware=rtl8168h-2_0.0.2 02/26/15 latency=0 link=no multicast=yes port=twisted pair
       resources: irq:42 ioport:3000(size=256) memory:d1704000-d1704fff memory:d1700000-d1703fff
  *-network
       description: Wireless interface
       product: Realtek Semiconductor Co., Ltd.
       vendor: Realtek Semiconductor Co., Ltd.
       physical id: 0
       bus info: pci@0000:04:00.0
       logical name: wlp4s0
       version: 00
       serial: 74:4c:a1:d1:2e:8f
       width: 64 bits
       clock: 33MHz
       capabilities: pm msi pciexpress bus_master cap_list ethernet physical wireless
       configuration: broadcast=yes driver=rtw89_pci driverversion=5.11.0-25-generic firmware=N/A ip=192.168.1.147 latency=0 link=yes multicast=yes wireless=IEEE 802.11
       resources: irq:78 ioport:2000(size=256) memory:d1600000-d16fffff
`

very noticable thing is the poor performance of WIFI when compared to Windows (i got two systems)
[image]
tried both 2.4 GHz/ 5.0 GHz, same results, on Windows ping is constantly in 32-36 ms range

lwfinger, 2021-08-16:
Pinging off the Google nameserver I get for a 100 ping run:
--- 8.8.8.8 ping statistics ---
100 packets transmitted, 99 received, 1% packet loss, time 99147ms
rtt min/avg/max/mdev = 26.552/152.069/304.084/49.422 ms

Switching to my router I get:
--- 192.168.1.1 ping statistics ---
100 packets transmitted, 100 received, 0% packet loss, time 99279ms
rtt min/avg/max/mdev = 0.791/53.856/205.208/45.003 ms

I then tested ping with the -s 1023 so that 1031 bytes were in each paclet:
--- 192.168.1.1 ping statistics ---
100 packets transmitted, 100 received, 0% packet loss, time 99142ms
rtt min/avg/max/mdev = 1.672/53.424/115.303/30.384 ms

That does indeed look bad, but when I look at the performance against a LibreSpeed instance on my local network, I get
ping 1.40 jitter 0.68
Download 596 Mbps Upload 482 Mbps

All results were obtained using an 802.11ac router at a distance of 1m and in the 5GHz band. Packet aggregation really helps.

I will pass this issue upstream to Realtek to see what they have to say. All I can say is that the driver seems to be better in real-life situations than when sending 84 byte packets.

lissyx, 2021-09-01:
Hello,

Just received and transfered my old Ubuntu 21.04 setup from ThinkPad T480s to brand new ThinkPad P14s MAD Gen2, and i'm also seeing similar behavior.
```
$ mtr --report -4 serveur.home
Start: 2021-09-01T21:59:43+0200
HOST: portable-alex               Loss%   Snt   Last   Avg  Best  Wrst StDev
  1.|-- serveur.home               0.0%    10   66.7 108.9   1.3 259.4  99.9
$ mtr --report -4 livebox.home
Start: 2021-09-01T22:00:16+0200
HOST: portable-alex               Loss%   Snt   Last   Avg  Best  Wrst StDev
  1.|-- livebox.home               0.0%    10   17.3 118.0   1.7 265.3  97.5
```

My WiFi setup is Ubiquiti-based, I'm just below the NanoHD AP, connected with good signal:
```
$ iw dev wlp3s0 link
Connected to xx:xx:xx:xx:xx:xx (on wlp3s0)
        SSID: Livebox-xxxx
        freq: 5200
        RX: 948960108 bytes (442782 packets)
        TX: 730634198 bytes (403251 packets)
        signal: -39 dBm
        rx bitrate: 866.7 MBit/s VHT-MCS 9 80MHz short GI VHT-NSS 2
        tx bitrate: 866.7 MBit/s VHT-MCS 9 80MHz short GI VHT-NSS 2

        bss flags:      short-preamble short-slot-time
        dtim period:    3
        beacon int:     100
```

I do feel the latency a lot when connecting to remove hosts over SSH.

danryu, 2021-10-30:
There are zero other routers in my environment - I'm not in a built-up area.
Also, even if there were routers or other external factors involved, those factors would also be present when running the Windows Realtek driver for the device, and they would also impact other devices (eg my HP notebook and my Android phone) when running the same ping test on the same network.

(If you have a Windows installation with this device, that is a good general comparison test to run against the rtw89 driver). 

I have done isolated tests to verify the higher latency, and it confirms what multiple other technical users have demonstrated here.

(Note: I have noticed that it improves/degrades as the performance profile changes on the host - partially confirming what @lissyx reported. I don't have a replication scenario yet though.)

Therefore, I would consider this NOT to be a red herring, and it was definitely worthwhile submitting the issue to Realtek.
I will register the issue with linux-wireless when I get a chance,.

@lwfinger I appreciate your effort on this project, and I hope things improve as the driver is merged into the upcoming mainline kernel (as I read is happening here: https://www.phoronix.com/scan.php?page=news_item&px=Realtek-802.11ax-rtw89).

lissyx, 2021-11-15:
FTR, uptodate Ubuntu/Hirsute on ThinkPad P14s Gen2 AMD, current `v7` branch, I have removed the workaround for `disable_ps_mode=Y` and I am now seeing normal pings:
```
$ mtr --report -4 -n serveur.home
Start: 2021-11-15T10:56:39+0100
HOST: portable-alex               Loss%   Snt   Last   Avg  Best  Wrst StDev
  1.|-- 192.168.1.98               0.0%    10    1.7   1.6   0.9   2.1   0.4
```

lissyx, 2021-11-15:
> Why had you added the PS mode workaround?

For the latency issues i reported above.

lwfinger, 2021-11-15:
What fixed it? Anything else changed?

lissyx, 2021-11-15:
> What fixed it? Anything else changed?

I am unsure exactly, but the first reports was done against `main` branch from the date of that time, I think it was some v5-based. There has been several kernel upgrades (majors ones when I upgraded from ubuntu 21.04 to 21.10). Firmware is still the same. I'm currently running on your current `v7` branch.

At least, I can state that I was experiencing the issue on the older setup, and it is not there anymore. I'm sorry I can't be more definitive on the actual fix.

lissyx, 2021-12-10:
It seems to be back using the `rtw89` module bundled in current 5.13.0-23 from Ubuntu 21.10, after a set of suspend/resume, even though the modules (core and pci) were both unloaded prior suspend and reloaded after resume. I'll make sure it is known to the distro packager, I can't remember for sure what commits they have in their tree, but I think it might be useful for others to know.

lammas, 2022-01-07:
I'm having the exact same issue and I've tried `disable_ps_mode=Y` and v7 branch in this repo.
I'm running Debian 11 on a ThinkPad P14s Gen 2a.

The only thing that did reduce the latency a little was commenting out the defines enabling debug output in the Makefile:

```
EXTRA_CFLAGS += -DCONFIG_RTW89_DEBUGMSG
EXTRA_CFLAGS += -DCONFIG_RTW89_DEBUGFS
```

It's still horrible, but the latency spikes went from 200-500ms to around 80-100ms.

ldx63, 2022-01-15:
Same issue here, Linux Mint, kernel 5.13.0-1026-oem, but ONLY in 5 GHz mode:

PING dsldevice.lan (192.168.0.1) 56(84) bytes of data.
64 bytes from dsldevice.lan (192.168.0.1): icmp_seq=1 ttl=64 time=333 ms
64 bytes from dsldevice.lan (192.168.0.1): icmp_seq=2 ttl=64 time=98.3 ms
64 bytes from dsldevice.lan (192.168.0.1): icmp_seq=3 ttl=64 time=684 ms
64 bytes from dsldevice.lan (192.168.0.1): icmp_seq=4 ttl=64 time=128 ms
64 bytes from dsldevice.lan (192.168.0.1): icmp_seq=5 ttl=64 time=1226 ms
64 bytes from dsldevice.lan (192.168.0.1): icmp_seq=6 ttl=64 time=215 ms
64 bytes from dsldevice.lan (192.168.0.1): icmp_seq=7 ttl=64 time=342 ms
64 bytes from dsldevice.lan (192.168.0.1): icmp_seq=8 ttl=64 time=380 ms
64 bytes from dsldevice.lan (192.168.0.1): icmp_seq=9 ttl=64 time=914 ms
64 bytes from dsldevice.lan (192.168.0.1): icmp_seq=10 ttl=64 time=326 ms
64 bytes from dsldevice.lan (192.168.0.1): icmp_seq=11 ttl=64 time=756 ms

While in 2.4 GHz mode, ping looks quite good:

64 bytes from dsldevice.lan (192.168.0.1): icmp_seq=1 ttl=64 time=3.56 ms
64 bytes from dsldevice.lan (192.168.0.1): icmp_seq=2 ttl=64 time=2.44 ms
64 bytes from dsldevice.lan (192.168.0.1): icmp_seq=3 ttl=64 time=6.90 ms
64 bytes from dsldevice.lan (192.168.0.1): icmp_seq=4 ttl=64 time=6.77 ms
64 bytes from dsldevice.lan (192.168.0.1): icmp_seq=5 ttl=64 time=12.4 ms
64 bytes from dsldevice.lan (192.168.0.1): icmp_seq=6 ttl=64 time=2.58 ms
64 bytes from dsldevice.lan (192.168.0.1): icmp_seq=7 ttl=64 time=3.09 ms
64 bytes from dsldevice.lan (192.168.0.1): icmp_seq=8 ttl=64 time=1.99 ms

This is an excerpt from lswh:
           *-network
                description: Wireless interface
                product: Realtek Semiconductor Co., Ltd.
                vendor: Realtek Semiconductor Co., Ltd.
                physical id: 0
                bus info: pci@0000:03:00.0
                logical name: wlp3s0
                version: 00
                serial: 74:4c:a1:de:c4:99
                width: 64 bits
                clock: 33MHz
                capabilities: pm msi pciexpress bus_master cap_list ethernet physical wireless
                configuration: broadcast=yes driver=rtw89_pci driverversion=5.13.0-1026-oem firmware=N/A ip=192.168.0.39 latency=0 l
ink=yes multicast=yes wireless=IEEE 802.11
                resources: irq:80 ioport:2000(size=256) memory:fd500000-fd5fffff

Regards
Aldo

lwfinger, 2022-01-20:
I have no idea what is wrong. On the 5G band, I get

--- 192.168.1.1 ping statistics ---
50 packets transmitted, 50 received, 0% packet loss, time 49060ms
rtt min/avg/max/mdev = 0.927/1.580/3.257/0.652 ms

Those are good statistics. I am running the latest code from this repo. There are no special kernel commands, nor am I loading with any module parameters.

lissyx, 2022-03-25:
Keeping an eye on that, updated to latest `main` yesterday alongside with latest 1.15 UEFI (P14s AMD Gen2) especially since release notes mentions Linux S3 improvements, I'm always connected on 5GHz, roaming seems not to be an issue.

I've disabled the hack to disable power sleep:
```
$ cat /etc/modprobe.d/rtw89_thinkpad.conf
#options rtw89core disable_ps_mode=Y
```

And so far, under similar conditions as before, it seems to behave adequately:
```
$ mtr --report -6 serveur.home; mtr --report -4 serveur.home
Start: 2022-03-25T09:41:48+0100
HOST: portable-alex               Loss%   Snt   Last   Avg  Best  Wrst StDev
  1.|-- serveur.home               0.0%    10    1.6   1.6   1.2   2.2   0.3
Start: 2022-03-25T09:42:04+0100
HOST: portable-alex               Loss%   Snt   Last   Avg  Best  Wrst StDev
  1.|-- serveur.home               0.0%    10    1.2   1.5   1.1   2.1   0.3
```

milan-markovic, 2022-07-23:
Using AUR repo (on Manjaro kernel 5.19), and also experiencing random latency spikes. I've noticed that running iperf3 actually reduces the latency. I've used Waveform Bufferbloat test, since it shows detailed latency tests (ignore bufferbloat results):

[Without running iperf in background](https://www.waveform.com/tools/bufferbloat?test-id=74bff7a3-36be-4aa3-a551-b884d74cf651)

[With running iperf in background](https://www.waveform.com/tools/bufferbloat?test-id=9b9f8e15-ea01-43d8-b57e-50607cb9c4c5)

As you can see, the overall latency results (mean values) are better when I have iperf3 running in the background. Basically, the "warm up" latency is the issue which is far greater in first test than second. Iperf3 is running with limited bandwidth `-b 600K`, when running below 500K, the results start to become worse, but didn't test more thoroughly.

I've seen people reporting the issues are fixed in recent commits, but I'm not sure which branch should I use? Any clarification on v5,6,7 branches vs main? I believe AUR uses main:

```
# Maintainer: Jerry Xiao <aur at mail.jerryxiao.cc>

_pkgbase=rtw89
_srcname=rtw89
_branch=main
pkgname=${_pkgbase}-dkms-git
pkgver=r178.f3ea327
pkgrel=1
epoch=1
pkgdesc="Driver for Realtek 8852AE, an 802.11ax device"
arch=('x86_64')
url="https://github.com/lwfinger/rtw89"
license=('GPL2')
makedepends=('git' 'xz')
depends=('dkms')
provides=("${_pkgbase}")
conflicts=("${_pkgbase}")
source=("$_srcname::git+https://github.com/lwfinger/rtw89.git#branch=${_branch}"
        'dkms.conf')
sha256sums=('SKIP'
            '3261b90709a41e7d1030d2365d8ee0d821275f8deba4825307c4de35013dbf8e')
install='rtw89-dkms-git.install'

build() {
  cd "$srcdir/${_srcname}"
  rm -fv dkms.conf
}
pkgver() {
  cd "$srcdir/${_srcname}"
  printf "r%s.%s" "$(git rev-list --count HEAD)" "$(git rev-parse --short HEAD)"
}

package() {
  # Copy dkms.conf
  install -Dt "${pkgdir}/usr/src/${_pkgbase}-${pkgver}" -m644 dkms.conf

  # Set name and version
  sed -e "s/@_PKGBASE@/${_pkgbase}/" \
      -e "s/@PKGVER@/${pkgver}/" \
      -i "${pkgdir}"/usr/src/${_pkgbase}-${pkgver}/dkms.conf

  # Copy sources (including Makefile)
  cp -rT "$_srcname" "${pkgdir}/usr/src/${_pkgbase}-${pkgver}"
  # This isn't the best solution but it works
  # and does not require additional dependencies
  rm -rfv "${pkgdir}/usr/src/${_pkgbase}-${pkgver}"/{.git,debian,.gitignore,README.md}
}
```

lwfinger, 2022-07-23:
Branch main is the only one up to date.

This kind of problem is beyond my abilities to fix and should be reported in the manner stated in the last paragraph of README.md

aathan, 2022-08-02:
Horrible interactive performance on ssh, still looking for a solution, on lenovo thinkpad using linux-oem-22.04a (5.17) kernel on ubuntu.

```
  *-network
       description: Wireless interface
       product: RTL8852AE 802.11ax PCIe Wireless Network Adapter
       vendor: Realtek Semiconductor Co., Ltd.
       physical id: 0
       bus info: pci@0000:03:00.0
       logical name: wlo1
       version: 00
       serial: 14:5a:fc:01:c2:xx
       width: 64 bits
       clock: 33MHz
       capabilities: pm msi pciexpress bus_master cap_list ethernet physical wireless
       configuration: broadcast=yes driver=rtw89_pci driverversion=5.17.0-1014-oem firmware=N/A ip=192.168.x.x latency=0 link=yes multicast=yes wireless=IEEE 802.11
       resources: irq:64 ioport:2000(size=256) memory:d1600000-d16fffff
       ```

mikaelhg, 2022-09-25:
So far I've had the best latency results with rtw89_8852ae with kernel 6.0rc6 and all modprobe options for the module removed.

    01:00.0 Network controller: Realtek Semiconductor Co., Ltd. RTL8852AE 802.11ax PCIe Wireless Network Adapter

gfokkema, 2022-09-25:
I own a Lenovo legion 5 15ach6h as well, been watching this issue for a bit over a year now.
After a few months of annoyances I bought an Intel WiFi NIC last year for about €20 and called it a day, never had any issues since.

mandalorian-one, 2022-11-24:
same issue with rtl8852be wireless module on my new Xiaomi Redmi Book
i have very unstable ping latency with my Mikrotik router located in 2m. I had never this issue with other devices and wireless controllers in the same environment.
also I have very poor download performance (5-10Mb/s) in comparison upload performance (50-60Mb/s) in 2.4Ghz networks
Some users reported that these issues is happen with some old routers especially with 2.4 band. Also they have the same issue in Windows.
As I see, there are some specificity of working Realtek wifi module with some routers and some wireless configurations that explains why some users have this issues but some not
Some users say that changing some options in windows driver solve the problem with performance partially: roaming aggressiveness -> disable, 802.11d -> off 

I see a lot of options in Linux kernel module, and think we can solve the problem by just tuning some options, but I don't see  options description and I don't know where to start

lwfinger, 2022-11-24:
If you see a lot of options, then you are running the obsolete driver from https://github.com/lwfinger/rtw8852be.git. There are very few options for this repo - https://githib.com/lwfinger/rtw89.git, and all are documented in modinfo. The former driver will be deleted next week due to its inferior design and performance.

mandalorian-one, 2022-11-24:
> If you see a lot of options, then you are running the obsolete driver from https://github.com/lwfinger/rtw8852be.git. There are very few options for this repo - [...]

Of course it is not the question of this topic, but
Does rtw89 support 6.0 kernel? I tried it, but driver isn't loaded because of errors:

> [150030.906258] rtw89_8852be 0000:02:00.0: Direct firmware load for rtw89/rtw8852b_fw.bin failed with error -2
> [150030.906266] rtw89_8852be 0000:02:00.0: failed to early request firmware: -2
> [150030.906406] rtw89_8852be 0000:02:00.0: Direct firmware load for rtw89/rtw8852b_fw.bin failed with error -2
> [150030.906408] rtw89_8852be 0000:02:00.0: failed to request firmware
> [150030.907953] rtw89_8852be 0000:02:00.0: failed to wait firmware completion
> [150030.907955] rtw89_8852be 0000:02:00.0: failed to setup chip information
> [150030.908545] rtw89_8852be: probe of 0000:02:00.0 failed with error -22
> 

but old rtw8852be is working.

lwfinger, 2022-11-24:
If your rtw89 driver does not see the firmware for the 8852be, then sudo {apt,dnf,zypper} install *firmware-realtek. If that repo does not exist, complain to your distro. The firmware for rtw8852be has been in the linux-firmware repo at vger.kernel.org since October 27. An alternative source is https://lwfinger.com/download/rtw8852b_fw.bin.

lwfinger, 2022-12-01:
What utility produces those graphs? That is new for me.

henk717, 2022-12-01:
That is cloudflare's speedtest available at speed.cloudflare.com, interestingly enough after taking the Windows screenshot and using Windows for a bit I booted back to Linux to see if I could resolve the issue but was unable to reproduce it afterwards, despite having rebooted prior between the different screenshots. So for now it is behaving correctly (With the powersafe still off) and I do not yet know what the difference is between the two test runs.

henk717, 2022-12-02:
Got an update on the situation, the remaining difference in loss could be explained by a firefox vs chrome difference. Using the same browser engine with the disable_ps_mode=Y tweak results in a stable experience. I no longer experience the loss and latency issues I was having in games. Neither do I have the constant buffering I had on videos. It has been stable all evening now.

So the disable_ps_mode=Y is a must, and I do recommend to make this the default value (or tweak the powersaving mode if possible) since without it the experience is very poor for the end user.

lwfinger, 2022-12-02:
Which browser is better?

You would have to take up the disable_ps_mode issue with the authors through the linux-wireless mailing list <linux-wireless@vger.kernel.org>, but as this is not a problem for most people, and battery life is a bigger concern, I doubt that they will agree with you. In any case, my code follows the upstream driver as present in the wireless-next repo at vger.kernel.org. The only differences are those needed to build on older kernels. The repo always has the code for the next kernel. It now has v6.2 wireless stuff.

henk717, 2022-12-02:
Chrome based browsers are better on that speedtest, i think firefox just struggles with the test rather than it being an actual issue in the wifi on firefox.

The issues are pretty extreme on the power saving mode, in game I saw constant loss, but I could also not properly watch a livestream because of the constant long buffering. With the power saving mode disabled that was no issue.

On the windows driver I do not see a power saving ability like some other network cards have so I think on Windows its already taken out.

I thought your repository was the one being upstreamed, if that is not the case the only relevant info for this repo is that the power management disabling indeed worked.

lwfinger, 2022-12-03:
You have it a little backwards. There is a Linux group at Realtek that writes the drivers for inclusion in Linux. I have a good working arrangement with their leader - I'm kind of like a grandfather to him. I take that code, add the modifications needed to build on older kernels and post it here. Without this repo, users of RTW8852BE would need to wait until kernel 6.2 came out roughly 2 months from now.

The module parameters is these drivers are also part of the upstream drivers. Thus parameter disable_ps_mode applied to driver rtw89core from this repo will do exactly the same with rtw89_core from kernel 6.2. If your system needs that option with drivers from this repo, it will need it with kernel 6.2. The drivers are essentially identical!

My contact did send me their code before it was sent to the kernel, and I did some beta-level debugging for them, but I did not post it here until they submitted it to wireless-next. They also provided me with the 8852be hardware.

So there is a connection between upstream and this repo, but upstream is the source for this code.

mikaelhg, 2023-07-18:
In 2023, using the `rtw89_8852ae` driver with the `6.4.2-060402-generic` kernel, the factors which allowed me to minimize the latency of the wifi connection were:

1. Upgrading to a `6.2`+ kernel.

2. Disabling `NetworkManager`'s enabling of WiFi powersaving.

On Ubuntu, `NetworkManager`'s configuration file was `/etc/NetworkManager/conf.d/default-wifi-powersave-on.conf`, and the correct setting `wifi.powersave = 2`. Then `sudo systemctl restart NetworkManager`.

closed 2023-11-02 by lwfinger

Rexcrazy804, 2024-06-02:
> In 2023, using the `rtw89_8852ae` driver with the `6.4.2-060402-generic` kernel, the factors which allowed me to minimize the latency of the wifi connection were:
> 
>     1. Upgrading to a `6.2`+ kernel.
> 
>     2. Disabling `NetworkManager`'s enabling of WiFi powersaving.
> 
> 
> On Ubuntu, `NetworkManager`'s configuration file was `/etc/NetworkManager/conf.d/default-wifi-powersave-on.conf`, and the correct setting `wifi.powersave = 2`. Then `sudo systemctl restart NetworkManager`.
> 
> Pre-`6.2`, there were weekly kernel lockups requiring a REISUB restart, associated with very long-running Cisco VPN connections. Disabling wifi power saving also gave a noticeable boost to wifi throughput.

I have tried disabling powersave as this comment highlighted but sadly while its gotten somewhat more consistent [still pretty high] in ipv4, ipv6 latency is remains pretty terrible. I am currently on the latest kernel version and I have tried rolling back to the latest lts 6.6 but the issue persists.

Semulia, 2024-08-22:
> In 2023, using the `rtw89_8852ae` driver with the `6.4.2-060402-generic` kernel, the factors which allowed me to minimize the latency of the wifi connection were:
>
> 1. Upgrading to a `6.2`+ kernel. [...]

You're awesome. Fedora 40. Lenovo LOQ laptop.
Just created *wifi-powersave-off.conf* file with following content:
`# File to be place under /etc/NetworkManager/conf.d
[connection]
# Values are 0 (use default), 1 (ignore/don't touch), 2 (disable) or 3 (enable).
wifi.powersave = 2`

And restart NetworkManager:
`sudo systemctl restart NetworkManager`

And spikes gone. You saved my day, great thanks.