lwfinger/rtw89 #38: Fedora 34 Workstation with Secure Boot: Installation issues and how to solve it

lukinoway, 2021-08-19:
The installation of this driver for Fedora 34 Workstation with secure boot is somehow little bit different compared with the README, but I was able to get it working.

So here is how I solved it:
1. checkout main branch, not the v5!
2. prepare certificate for self-signing
```sh
1) mkdir ~/mok && cd ~/mok

2) openssl req -new -x509 -newkey rsa:2048 -keyout MOK.priv -outform DER -out MOK.der -nodes -days 36500 -subj "/CN=Custom MOK/"

3) sudo mokutil --import MOK.der
```
3. restart computer
4. after restart open terminal in rtw89 dir
```sh
1) make

2) /usr/src/kernels/$(uname -r)/scripts/sign-file \
 sha256 \
 /home/<USER>/mok/MOK.priv \
 /home/<USER>/mok/MOK.der \
 rtw89core.ko
 
3)  /usr/src/kernels/$(uname -r)/scripts/sign-file \
 sha256 \
 /home/<USER>/mok/MOK.priv \
 /home/<USER>/mok/MOK.der \
 rtw89pci.ko

4) sudo make install
```
5. restart computer

For me this worked and WIFI is now working great!

marcusklaas, 2021-08-20:
Yessss, that's exactly the situation I was in as well. I like to think I was slowly getting there myself, but following your instructions worked perfectly the first time! Currently writing this over wifi on my new Fedora machine :-)

Big thank to you, @lukinoway!

mawi675lr81, 2021-08-29:
Hi, not much familiar with the git. By main branch, you mean just use 'git clone https://github.com/lwfinger/rtw89.git'?

Also, do we need to redo your steps everytime there's a kernel update?

lwfinger, 2021-08-29:
I suggest that you do some reading on the use of git. It is a very useful utility.

Yes, your command will get the main branch here.

When you get a new kernel, you should cd to the rtw89 directory and redo the 'make' and 'sudo make install' commands. Once you get a network, the you should do a 'git pull' to get any updates, and redo the make sequence if there were any updates.

closed 2021-08-29 by lwfinger

Truesh00ter, 2021-09-04:
@lukinoway your guide isn't fully complete.

What I missed was after step 3, that we need to manually initiate the mok import process by pressing a key during the boot process, importing the key and utilizing the used password when importing the MOK.der file. After I did this smooth sails ahead. Thanks!

ARSrabon, 2021-09-24:
> The installation of this driver for Fedora 34 Workstation with secure boot is somehow little bit different compared with the README, but I was able to get it working.
> 
> So here is how I solved it:
> 
> 1. checkout main branch, not the v5!
> 2. prepare certificate for self-signing
> 
> ```shell
> 1) mkdir ~/mok && cd ~/mok
> 
> 2) openssl req -new -x509 -newkey rsa:2048 -keyout MOK.priv -outform DER -out MOK.der -nodes -days 36500 -subj "/CN=Custom MOK/"
> 
> 3) sudo mokutil --import MOK.der
> ```
> 
> 1. restart computer
> 2. after restart open terminal in rtw89 dir
> 
> ```shell
> 1) make
> 
> 2) /usr/src/kernels/$(uname -r)/scripts/sign-file \
>  sha256 \
>  /home/<USER>/mok/MOK.priv \
>  /home/<USER>/mok/MOK.der \
>  rtw89core.ko
>  
> 3)  /usr/src/kernels/$(uname -r)/scripts/sign-file \
>  sha256 \
>  /home/<USER>/mok/MOK.priv \
>  /home/<USER>/mok/MOK.der \
>  rtw89pci.ko
> 
> 4) sudo make install
> ```
> 
> 1. restart computer
> 
> For me this worked and WIFI is now working great!

Hi,
Thank you for sharing your steps. This really helped me installing the driver and I have the secure boot enabled. I'm using the code from the v6 branch.

***Note: I'm adding my device information here, in case other users can find it via GitHub issue search.
Device: Lenovo Legion 5 Pro 16ACH
Distro: Linux Mint 20.02 
Kernel: 5.11.0-36-generic
secure boot: enabled

lwfinger, 2021-09-24:
Thank you for this. Note that the main and v5 branches have exactly the same code, thus your step #1 seems to be untrue.

lukinoway, 2021-09-25:
@lwfinger at the time I tried this out v5 was not working and therefore i mentioned it here.
Potentially it is working now, but haven't tested that.

When doing next kernel upgrade, will test v6.

Truesh00ter, 2021-10-05:
With Fedora 34 and kernel 5.14.9-200.fc34.x86_64 I'm having issues with the script.
Can´t seem to get WiFi back to working again.

Skipping BTF generation for /home/matthew/rtw89/rtw89core.ko due to unavailability of vmlinux
[..]
Skipping BTF generation for /home/matthew/rtw89/rtw89pci.ko due to unavailability of vmlinux

Previously I had to use above fix by @lukinoway on my Lenovo ThinkPad P14s on Fedora, but with the current kernel it seems to have stopped working.