lwfinger/rtw89 #40: Problem with make sign-install command

Leviathan13186, 2021-08-22:
Hey guys, sorry if this is a simple issue I'm super new to linux. I'm running Kubuntu and followed all the steps and on the last one I got this output ,"make: *** No rule to make target 'sign-install'.  Stop." Is there something I did wrong?

sh1n0b1, 2021-08-22:
I had the same problem. Here's how I resolve this:
1. Follow [this](https://ubuntu.com/blog/how-to-sign-things-for-secure-boot) create and entroll a self-sign cert/key to MOK (Make sure to reboot and enroll the key during boot up). Make sure the key got added by running "cat /proc/keys" and search for the common name of the cert you specified in your openssl.cnf file. 
2. run "make install" on the code, skip the "sign-install" command
3. Sign the two module.ko file (rtw89pci.ko and rtw89core.ko)
4. "sudo modprobe rtw89pci"

strangebreww, 2021-08-22:
That's because `sign-install` is missing in `v5` branch atm. Possible workaround is to clone the main branch instead.

lwfinger, 2021-08-22:
v6 will not build on older kernels either. I have been busy with another project.

Leviathan13186, 2021-08-22:
> I had the same problem. Here's how I resolve this:
> 
> 1. Follow [this](https://ubuntu.com/blog/how-to-sign-things-for-secure-boot) to create and enroll a self-sign cert/key to MOK (Make sure to reboot and enroll the key during boot up). Make sure the key got added by running "cat /proc/keys" and search for the common name of the cert you specified in your openssl.cnf file. 
> 
> 2. run "make install" on the code, skip the "sign-install" command
> 
> 3. Sign the two module.ko file (rtw89pci.ko and rtw89core.ko)
> 
> 4. "sudo modprobe rtw89pci"

Thank you I'll try this! So just copy paste the commands under kernels and bootloaders in that link right? Is the sudo mod probe the command that signs the two module.ko?

sh1n0b1, 2021-08-23:
> Thank you I'll try this! So just copy paste the commands under kernels and bootloaders in that link right? Is the sudo mod probe the command that signs the two module.ko?

No, the command to sign the module is: "kmodsign sha512 MOK.priv MOK.der module.ko"
Replace "module.ko" with the two module files. Follow the guide.

Leviathan13186, 2021-08-24:
Ok so i tried with the no "sign-install" and just did "install" and that worked. Probably should have tried that first before going on here. But thank you for helping at least!! thank you all !

lukinoway, 2021-08-29:
In case you are not running Ubuntu based system, in my case Fedora, check out #38.
Was struggling to find out how to fix it and wrote my "how to" over there.

closed 2023-10-02 by lwfinger