lwfinger/rtw89 #41: Random "disconnects"

manygave, 2021-08-23:
I'm not exactly a Linux expert so I'm not really sure what's going on here but at random I just can't connect to the internet so here's dmesg:
[dmesg.txt](https://github.com/lwfinger/rtw89/files/7035433/dmesg.txt)

felix-ulonska, 2021-09-01:
I have the same issue on Manjaro with 5.14 Kernel on a Lenovo Legion 5 Pro. If supplying additional logs for this issue are helpful, please reach out to me!

felix-ulonska, 2021-09-01:
The request file:
[dmesg.txt](https://github.com/lwfinger/rtw89/files/7094607/dmesg.txt)

lwfinger, 2021-09-02:
Both of you are seeing some kind of firmware error. This kind of error can occur because the BIOS does not support the kinds of accesses. Please create a file named /etc/module.d/50-rtw89.conf. In this file, create a line that says

options rtw89pci disable_aspm_l1=y

If that does not help, change the line to say

options rtw89pci disable_aspm_l1ss=y

If that still fails, try

options rtw89pci disable_aspm_l1=y disable_aspm_l1=y

Please let me know your results.

felix-ulonska, 2021-09-07:
I have tested this the last day with every configuration. Sadly it still seems to fail regularly. 

I have attached a dmesg.txt with options rtw89pci disable_aspm_l1=y disable_aspm_l1ss=y

Is there a way to validate that the options are loaded?

lwfinger, 2021-09-07:
Yes.  do 'sudo modprobe -rv rtw89pci' and 'sudo modprobe -v rtw89pci'. The second command should show the options.

I do not see your dmesg.txt.

Soulwest, 2021-09-10:
> Both of you are seeing some kind of firmware error. This kind of error can occur because the BIOS does not support the kinds of accesses. Please create a file named /etc/module.d/50-rtw89.conf. In this file, create a line that says
> 
> options rtw89pci disable_aspm_l1=y
> 
> If that does not help, change the line to say
> 
> options rtw89pci disable_aspm_l1ss=y
> 
> If that still fails, try
> 
> options rtw89pci disable_aspm_l1=y disable_aspm_l1ss=y
> 
> Please let me know your results.

This helps me fix disconnects, but now I have issues with connection speed same as in https://github.com/lwfinger/rtw89/issues/30

truefire7, 2021-09-10:
[dmest.txt](https://github.com/lwfinger/rtw89/files/7143860/dmest.txt)
I am having the same issue on a Yoga 6, Manjaro latest. I've tried using all the way up to the most recent kernels, just in case the support was native yet :)

Not seeing handshake errors in dmesg, but I am noticing similarities in connection stability with distance as the other user in [29](https://github.com/lwfinger/rtw89/issues/29).

Output from sudo modprobe -rv rtw89pci

rmmod rtw89pci
rmmod rtw89core
rmmod mac80211
rmmod libarc4
rmmod cfg80211

Options list with sudo modprobe -v rtw89pci:

insmod /lib/modules/5.14.0-0-MANJARO/kernel/drivers/net/wireless/realtek/rtw89/rtw89core.ko 
insmod /lib/modules/5.14.0-0-MANJARO/kernel/drivers/net/wireless/realtek/rtw89/rtw89pci.ko 

I have not yet tried the options .conf, I'm not familiar enough with the output of dmsg to know if my issue is the same. Let me know if I should try that.

Wyris, 2021-09-10:
I am able to connect to my WiFi network but DNS doesn't seem to work. I can ping IPs and even access my router's admin page but anything involving domains doesn't work.
```[32913.100889] wlp4s0: authenticate with ______
[32913.281061] wlp4s0: send auth to ______ (try 1/3)
[32913.284263] wlp4s0: authenticated
[32913.284743] wlp4s0: associate with ______ (try 1/3)
[32913.292837] wlp4s0: RX AssocResp from ______ (capab=0xc11 status=0 aid=1)
[32913.401510] wlp4s0: associated
[32913.522992] IPv6: ADDRCONF(NETDEV_CHANGE): wlp4s0: link becomes ready
```

lwfinger, 2021-09-10:
DNS is not handled by the driver. If you can ping other addresses, then the driver is working.

What distro and what network management software are you using?

Wyris, 2021-09-10:
> DNS is not handled by the driver. If you can ping other addresses, then the driver is working.

I am aware. It's just unusual.
I am using Manjaro with 5.10.61 and NetworkManager.

lwfinger, 2021-09-10:
Run the command 'dig google.com' and post the output.

Does DNS work with other network devices?

Wyris, 2021-09-10:
Interestingly enough, dig is able to resolve hostnames but pacman and the browser aren't.

lwfinger, 2021-09-10:
I would still like to see the output of the dig command. Any differences between it and that of some other network device?

Wyris, 2021-09-10:
Sorry, I should have done it earlier.
```
; <<>> DiG 9.16.20 <<>> google.com
;; global options: +cmd
;; Got answer:
;; ->>HEADER<<- opcode: QUERY, status: NOERROR, id: 48942
;; flags: qr rd ra; QUERY: 1, ANSWER: 1, AUTHORITY: 0, ADDITIONAL: 1

;; OPT PSEUDOSECTION:
; EDNS: version: 0, flags:; udp: 512
;; QUESTION SECTION:
;google.com.                    IN      A

;; ANSWER SECTION:
google.com.             84      IN      A       142.250.185.238

;; Query time: 169 msec
;; SERVER: 9.9.9.9#53(9.9.9.9)
;; WHEN: Fri Sep 10 18:49:55 CEST 2021
;; MSG SIZE  rcvd: 55
```
No, the resolution works fine. It must be some problem with NetworkManager.
Steps to reproduce:
1. Disconnect (from the wired connection (which I normally use when at home)
2. Connect to my wireless network (sometimes fails but succeeds eventually [journalctl.txt](https://github.com/lwfinger/rtw89/files/7145308/journalctl.txt))
3. Try to open any website/access any web resource by domain name

lwfinger, 2021-09-10:
I was not aware of that Quad9 DNS server (9.9.9.9). Thanks for the info.

Wyris, 2021-09-13:
After replacing the Realtek card with Intel AX200 I do not have the described problem anymore.

Soulwest, 2021-09-15:
> This helps me fix disconnects.

Unfortunately, disconnects come back. 
After some time WIFI disconnects and doesn't want to connect again. Even after turn the WIFI on/off and after reloading the rtw89pci module.
I've tried to connect to several WIFIs but could connect only to the nearest one (my phone with hotspot). It looks like the module can't connect to far WiFi points with a low signal level (< -60dbm).

When I tried to connect in [NetworkManager](https://github.com/lwfinger/rtw89/files/7167375/NetworkManager.txt) logs I see a timeout error 
and a lot of these errors in dmesg logs
`[394685.141856] rtw89_pci 0000:04:00.0: timed out to flush queues`

Soulwest, 2021-09-27:
> After replacing the Realtek card with Intel AX200 I do not have the described problem anymore.

The same problem and the same solution helps. Bye-bye Realtek.

closed 2021-11-06 by manygave