lwfinger/rtw89 #45: Unable to use airplay deauth

Popimopi, 2021-08-29:
Kali Linux
HI, I updated with this driver and my WiFi started working perfectly but I am unable to deauth devices using Airplay. I can successfully go to monitor mode and it can capture a handshake but cant deauth devices.

lwfinger, 2021-08-30:
Are any error messages logged? How about in the dmesg output?

Popimopi, 2021-08-31:
> Are any error messages logged? How about in the dmesg output?
Hi, Thanks for the reply.
No error messages logged. It is sending deauth packets but the devices are not being disconnected. I have used it on several connections as well as on several devices. Aireplay-ng seems to capture the WPA handshake once I manually reconnect my devices.

dmseg output-
─# dmesg
[    0.000000] Linux version 5.10.0-kali9-amd64 (devel@kali.org) (gcc-10 (Debian 10.2.1-6) 10.2.1 20210110, GNU ld (GNU Binutils for Debian) 2.35.2) #1 SMP Debian 5.10.46-4kali1 (2021-08-09)
[    0.000000] Command line: BOOT_IMAGE=/live/vmlinuz-5.10.0-kali9-amd64 boot=live persistence components quiet splash noeject findiso= persistence
[    0.000000] x86/fpu: Supporting XSAVE feature 0x001: 'x87 floating point registers'
[    0.000000] x86/fpu: Supporting XSAVE feature 0x002: 'SSE registers'
[    0.000000] x86/fpu: Supporting XSAVE feature 0x004: 'AVX registers'
[    0.000000] x86/fpu: Supporting XSAVE feature 0x200: 'Protection Keys User registers'
[    0.000000] x86/fpu: xstate_offset[2]:  576, xstate_sizes[2]:  256
[    0.000000] x86/fpu: xstate_offset[9]:  832, xstate_sizes[9]:    8
[    0.000000] x86/fpu: Enabled xstate features 0x207, context size is 840 bytes, using 'compacted' format.
[    0.000000] BIOS-provided physical RAM map:
[    0.000000] BIOS-e820: [mem 0x0000000000000000-0x000000000009ffff] usable
[    0.000000] BIOS-e820: [mem 0x00000000000a0000-0x00000000000fffff] reserved
[    0.000000] BIOS-e820: [mem 0x0000000000100000-0x0000000009eeffff] usable
[    0.000000] BIOS-e820: [mem 0x0000000009ef0000-0x000000000a000fff] reserved
[    0.000000] BIOS-e820: [mem 0x000000000a001000-0x000000000a1fffff] usable
[    0.000000] BIOS-e820: [mem 0x000000000a200000-0x000000000a20efff] ACPI NVS
[    0.000000] BIOS-e820: [mem 0x000000000a20f000-0x00000000b6ec7fff] usable
[    0.000000] BIOS-e820: [mem 0x00000000b6ec8000-0x00000000b7045fff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000b7046000-0x00000000b70c3fff] ACPI data
[    0.000000] BIOS-e820: [mem 0x00000000b70c4000-0x00000000b72bcfff] ACPI NVS
[    0.000000] BIOS-e820: [mem 0x00000000b72bd000-0x00000000bcb83fff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000bcb84000-0x00000000bcbfdfff] type 20
[    0.000000] BIOS-e820: [mem 0x00000000bcbfe000-0x00000000bdffffff] usable
[    0.000000] BIOS-e820: [mem 0x00000000be000000-0x00000000bfffffff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000f0000000-0x00000000f7ffffff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000fd000000-0x00000000fdffffff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000fe700000-0x00000000fe700fff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000feb80000-0x00000000fec01fff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000fec10000-0x00000000fec10fff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000fed00000-0x00000000fed00fff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000fed40000-0x00000000fed44fff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000fed80000-0x00000000fed8ffff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000fedc4000-0x00000000fedc9fff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000fedcc000-0x00000000fedcefff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000fedd5000-0x00000000fedd5fff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000ff000000-0x00000000ffffffff] reserved
[    0.000000] BIOS-e820: [mem 0x0000000100000000-0x000000041e2fffff] usable
[    0.000000] BIOS-e820: [mem 0x000000041e300000-0x000000043fffffff] reserved
[    0.000000] NX (Execute Disable) protection: active
[    0.000000] efi: EFI v2.70 by American Megatrends
[    0.000000] efi: ACPI=0xb70c3000 ACPI 2.0=0xb70c3014 TPMFinalLog=0xb7272000 SMBIOS=0xbca0e000 SMBIOS 3.0=0xbca0d000 MEMATTR=0xb34ff698 ESRT=0xb4086f98 
[    0.000000] secureboot: Secure boot could not be determined (mode 0)
[    0.000000] SMBIOS 3.3.0 present.
[    0.000000] DMI: HP Victus by HP Laptop 16-e0xxx/88EE, BIOS F.09 07/16/2021
[    0.000000] tsc: Fast TSC calibration using PIT
[    0.000000] tsc: Detected 3193.968 MHz processor
[    0.000131] e820: update [mem 0x00000000-0x00000fff] usable ==> reserved
[    0.000132] e820: remove [mem 0x000a0000-0x000fffff] usable
[    0.000137] last_pfn = 0x41e300 max_arch_pfn = 0x400000000
[    0.000140] MTRR default type: uncachable
[    0.000140] MTRR fixed ranges enabled:
[    0.000141]   00000-9FFFF write-back
[    0.000141]   A0000-DFFFF uncachable
[    0.000141]   E0000-FFFFF write-protect
[    0.000142] MTRR variable ranges enabled:
[    0.000142]   0 base 000000000000 mask FFFF80000000 write-back
[    0.000143]   1 base 000080000000 mask FFFFC0000000 write-back
[    0.000143]   2 disabled
[    0.000144]   3 disabled
[    0.000144]   4 disabled
[    0.000144]   5 disabled
[    0.000144]   6 disabled
[    0.000145]   7 disabled
[    0.000145] TOM2: 0000000440000000 aka 17408M
[... 1056 lines cut ...]
[   32.568469] snd_hda_codec_realtek hdaudioC2D0:    mono: mono_out=0x0
[   32.568471] snd_hda_codec_realtek hdaudioC2D0:    inputs:
[   32.568474] snd_hda_codec_realtek hdaudioC2D0:      Mic=0x19
[   32.570110] Bluetooth: hci0: RTL: examining hci_ver=0b hci_rev=000a lmp_ver=0b lmp_subver=8852
[   32.570130] Bluetooth: hci0: RTL: unknown IC info, lmp subver 8852, hci rev 000a, hci ver 000b
[   32.570133] Bluetooth: hci0: RTL: assuming no firmware upload needed
[   32.622047] input: HD-Audio Generic Mic as /devices/pci0000:00/0000:00:08.1/0000:06:00.6/sound/card2/input31
[... cut ...]
[   41.581001] r8169 0000:02:00.0: firmware: direct-loading firmware rtl_nic/rtl8168h-2.fw
[   41.609265] Generic FE-GE Realtek PHY r8169-200:00: attached PHY driver [Generic FE-GE Realtek PHY] (mii_bus:phy_addr=r8169-200:00, irq=IGNORE)
[   41.756941] amdgpu 0000:06:00.0: vgaarb: changed VGA decodes: olddecodes=io+mem,decodes=none:owns=none
[   41.807334] r8169 0000:02:00.0 eth0: Link is Down
[   44.144654] r8169 0000:02:00.0 eth0: Link is Up - 100Mbps/Full - flow control rx/tx
[   44.144670] IPv6: ADDRCONF(NETDEV_CHANGE): eth0: link becomes ready

FruitBoi, 2022-11-10:
Also having an issue with packet injection. Have an 8852be which goes into monitor mode fine, but upon running the aireplay injection test hard freezes the system.

lwfinger, 2022-11-10:
A "system freeze" is not really much of a diagnostic. Is the caps lock light flashing, or is the CPU running hard? If the former, do you have another Linux system that could serve a a netconsole?

I am not the one to debug this kind of problem. Use the instructions in the last paragraph of README.md to submit email that the developers read. Note that an imprecise description of the failure will be ignored there. They are less responsive than I am.

If you tell me exactly what you did, I will try to duplicate the issue.

Anzar-Makrani, 2023-02-25:
I have same problem too.(rtl8852ae)
running `aireplay --test wlo1 `freeze the screen **including light on caps lock** (if  lights on it wont turn off when pressed and vice versia).

Here are the thing i did on my hp victus
1. used airmon-ng and put the interface in monitor mode 

- it did successful switch to monitor mid but doesn't switch back to Managed
   - tried switching back with iwxonfig o/p: device or resource is busy
   - with ifconfig no output but it didnt switch back to managed mode 
   - i tried powering off device with both iwconfig and ifconfig(ifdown) didn't power it down

2. ran wifite 

-  wps attacks doesn't seems to work with both reaver and bully this include wps pint attack,null pin attack and pixie-dust
    - with reaver o/p: reaver processes stop with exit code 1
    - with bully deauth is always on timeout)
 - deauth attack doesn't work
    - something's wrong with deauth packets it didn't deauth any client
    - although when i manualy disconnect the client and reconnected it to the ap wifite did captured the handshake and start cracking it
    - pmkid attack; cant say anything its very difficult to test

for some reason the card have hard time finding hidden networks 
- tried 5-6 times didnt find any (in wifite) but when i specified a channel to scan it did.
- meanwhile airodump does find hidden networks with no problems

ph4ntom-rev, 2026-11-01:
Hello everyone,

I investigated why packet injection / deauth frames (`aireplay-ng`) and raw TX consistently fail or trigger kernel panics on RTL8852AE/BE with the stock `rtw89` driver.

By reverse-engineering the RISC-V firmware scheduling state machine and MAC register behavior in IDA Pro, we found four distinct host-driver root causes rather than an irreversible firmware limitation:
1. **Uninitialized MACID Scheduler Context**: The firmware keeps the transmit queue paused for unassociated monitor interfaces unless default CMAC/DMAC tables are pushed and registered via `rtw89_fw_h2c_join_info()`. Using `RTW89_SELF_ROLE_AP` with `RTW89_NET_TYPE_NO_LINK` initializes the MACID without triggering client beacon-loss timeouts or broadcasting AP beacons.
2. **Band 0 MAC Promiscuous Filtering (`0xCE20` - `0xCE2C`)**: Hardware registers `R_AX_RX_FLTR_OPT`, `R_AX_CTRL_FLTR`, and `R_AX_DATA_FLTR` drop Control frames (ACK/RTS/CTS) and cross-BSS Data by default unless unlocked with `0x55555555`.
3. **`NULL` `vif` Dereference on Injection**: `mac80211` passes `info->control.vif = NULL` on injected Radiotap frames, which triggered an unconditional `vif_to_rtwvif(vif)` NULL pointer dereference in `rtw89_ops_tx()` and `rtw89_core_tx_write()`.
4. **Invalid Rate Index**: When Radiotap headers omit an explicit rate, `mac80211` sets `rates[0].idx = -1`, which was cast to `(u8)255`, writing an invalid hardware rate and leaving `desc_info->en_wd_info` unset.

We developed an open-source host-side patchset that enables full monitor mode and packet injection on RTL8852AE using 100% stock, unmodified `linux-firmware` images (compatible with Linux 6.1 through 7.x+ and `lwfinger/rtw89`):
👉 https://github.com/ph4ntom-rev/rtw89-monitor-injection

Hope this helps anyone looking to get `airmon-ng` and `aireplay-ng` working on modern Realtek 802.11ax laptops!

itsParassharma, 2026-11-01:
> Hello everyone,
>
> I investigated why packet injection / deauth frames (`aireplay-ng`) and raw TX consistently fail or trigger kernel panics on RTL8852AE/BE with the stock `rtw89` driver.
>
> By [...]

Thats pretty good. Thanks for sharing. Finally a fix after 5 years. I guess AI succesfully bug fixed it. I might still have the antenna somewhere so will try it.

closed 2026-11-01 by itsParassharma