lwfinger/rtw89 #48: wifi installs find, bluetooth remains non-functioning - debian 11

kreativkod, 2021-09-06:
Thanks a zillion for the 8852AE wifi driver! 
I just got a new Lenovo laptop and your rtw89 package found the wificard. The kernel package 5.10.0-8-amd only had a driver for rtl8852a, not for rtl8852ae.

But I remain confused. Bluetooth does not seem to function (cannot initiate pairing operations, and my smartphone does not find my computer). Was it supposed to become workable with your package?

Do I need to change from the btusb driver that seems to load at boot?

Network:   Device-1: Realtek driver: rtw89_pci 
           IF: wlo1 state: up mac: 14:5a:fc:1e:b3:4f 
Bluetooth: Device-1: Realtek Bluetooth Radio type: USB driver: btusb 
           Report: ID: hci0 state: up running pscan iscan bt-v: 3.0 address: 14:5A:FC:1E:B3:50

lwfinger, 2021-09-06:
My package is the wifi driver only. Drivers btusb and btrtl are correct for the BT part. Check the output of dmesg to make sure you have the correct firmware, otherwise I have no suggestions.

As stated many times - I do NOT do Bluetooth!!

kreativkod, 2021-09-06:
I understand. Thanks for taking the time to answer.

christianbaun, 2021-09-27:
I have the same issue with Debian 11. WIFI works well, but bluetooth not at all. Did anyone get it running?

lwfinger, 2021-09-27:
Well, you did not provide any information. Using v6 of rtw89 and the 5.14 kernel, I have gotten BT to work.

What does lsusb have to say about your BT?

christianbaun, 2021-09-27:
It is a ThinkPad T14s AMD Gen1 running Debian 11 (stable) with the shipped Kernel 5.10.

```
$ uname -srvmo
Linux 5.10.0-8-amd64 #1 SMP Debian 5.10.46-4 (2021-08-03) x86_64 GNU/Linux
```

I did this:

```
git clone https://github.com/lwfinger/rtw89.git -b v5
cd rtw89
make
sudo make sign-install
```

After a reboot...
 
```
# inxi --bluetooth -xxx
Bluetooth: Device-1: Realtek Bluetooth Radio type: USB driver: btusb v: 0.8 bus ID: 6-4:3 chip ID: 0bda:4852 class ID: e001 
           serial: 00e04c000001 
           Report: ID: hci0 state: up running bt-v: 3.0 lmp-v: 5.2 sub-v: 8852 hci-v: 5.2 rev: a address: 14:5A:FC:25:84:70 

```
```
# dmesg | grep -i Blue
[    1.964351] usb 6-4: Product: Bluetooth Radio
[    3.038192] thinkpad_acpi: rfkill switch tpacpi_bluetooth_sw: radio is unblocked
[    3.513666] Bluetooth: Core ver 2.22
[    3.513692] Bluetooth: HCI device and connection manager initialized
[    3.513698] Bluetooth: HCI socket layer initialized
[    3.513699] Bluetooth: L2CAP socket layer initialized
[    3.513702] Bluetooth: SCO socket layer initialized
[    3.520343] Bluetooth: hci0: RTL: examining hci_ver=0b hci_rev=000a lmp_ver=0b lmp_subver=8852
[    3.520347] Bluetooth: hci0: RTL: unknown IC info, lmp subver 8852, hci rev 000a, hci ver 000b
[    3.520348] Bluetooth: hci0: RTL: assuming no firmware upload needed
[    4.095245] Bluetooth: BNEP (Ethernet Emulation) ver 1.3
[    4.095248] Bluetooth: BNEP filters: protocol multicast
[    4.095252] Bluetooth: BNEP socket layer initialized
[    7.093501] Bluetooth: RFCOMM TTY layer initialized
[    7.093509] Bluetooth: RFCOMM socket layer initialized
[    7.093517] Bluetooth: RFCOMM ver 1.11
```

lwfinger, 2021-09-27:
Where the ***** is the output of lsusb that I asked for? I do not know why I continue to bother with answering your questions. You never seem to read what I writ3e,

If your device is recognized as a generic BT device, then the special code needed for the Realrek device will not be loaded.

christianbaun, 2021-09-27:
It is the same information, provided by inxi.

```
$ lsusb | grep Blue
Bus 006 Device 003: ID 0bda:4852 Realtek Semiconductor Corp. Bluetooth Radio
```

lwfinger, 2021-09-27:
That device is not in the kernel tables. In addition, my previous update for devices with ID 0x04ca:0x4006 have not yet propagated through the system. You will need to build your own kernel. Install the source and edit drivers/bluetooth/btusb.c.

Find the code that says
        /* Realtek 8852AE Bluetooth devices */
        { USB_DEVICE(0x0bda, 0xc852), .driver_info = BTUSB_REALTEK |
                                                     BTUSB_WIDEBAND_SPEECH },

Duplicate the last two lines and replace 0xc852 with 0x4852.

christianbaun, 2021-09-28:
> ... You will need to build your own kernel. Install the source and edit drivers/bluetooth/btusb.c.

I decided to pick another option. I ordered a new Intel AX200 Wireless Adapter WI-FI 6 2230 yesterday for less than 20€ after your post. It arrived today. Replacing the Realtek RTL8852AE with the Intel NIC required less than 5 minutes, and after rebooting, WIFI and Bluetooth immediately worked well. For me, that's better than investing hours into trying to get the Realtek NIC running propperly.

lwfinger, 2021-09-28:
Your choice.