lwfinger/rtw89 #50: wifi stops working after kernel update

zublos, 2021-09-14:
Hi,

The module first worked fine for kernel 5.11.0-27 but after I updated to kernel 5.11.0-34 it stops working.

After the I executed the 4 lines code it just outputs "Install rtw89 SUCCESS"
But when I try to unload and load the module this error shows up: "modprobe: ERROR: could not insert 'rtw89pci': Exec format error"

"dmesg -t" outputs: "rtw89core: disagrees about version of symbol module_layout"

Linux Mint 20.2 on a new HP Laptop

I'm not very experienced with linux, so I appreciate every help
Thanks!

[dmesg.txt](https://github.com/lwfinger/rtw89/files/7160219/dmesg.txt)

manygave, 2021-09-14:
Hi mate you're supposed to "reinstall" the driver I run fedora but here's what I do 

> rm rtw89

> git clone https://github.com/lwfinger/rtw89

> cd rtw89

> make

> sudo make install 

This works for me but essentially remove the directory and reinstall 

Btw I probably could of said this a bit better

lwfinger, 2021-09-14:
You do not need to remove the directory. The following sequence workgs for me:
cd ~/rtw89
git pull              <- optional - do not use if no Internet.
make
sudo make install
sudo modprobe -v rtw89pci

Your problem was that the source used to build the modules did not match the running kernel. When the kernel changes, you cannot prebuild before the reboot. Build after the reboot.