lwfinger/rtw89 #51: v6

domenkozar, 2021-09-15:
https://lwn.net/Articles/866924/

capfredf, 2021-09-15:
there is a [branch](https://github.com/lwfinger/rtw89/tree/v6) for v6

domenkozar, 2021-09-15:
Nice! Did anyone tested it yet?

lwfinger, 2021-09-15:
Are youo implying that I would post untested code without saying so? Of course it is tested, at least with the later kernels. There may be build problems that I missed, but I did try kernels back to v5.4.

domenkozar, 2021-09-15:
Not implying at all! Just asking for other folks if they tried it too.

Anyway it works, but it doesn't fix my speed issues.

BenBE, 2021-09-15:
Trying to build Debian packages like for the v5 failed for me with v6 yielding a warning about a "binary only" package (can have a closer look if you need the exact error message from DKMS when trying to install the resulting deb file).

Incubant, 2021-09-15:
In my case the v6 branch does not compile. and default (v5) does not fix my missing wifi any more

Is there something i have to consider.
`[linux@linux-82ln rtw89]$ uname -srm
Linux 5.10.61-1-MANJARO x86_64

[linux@linux-82ln rtw89]$ make
make -C /lib/modules/5.10.61-1-MANJARO/build M=/home/linux/rtw89 modules
make[1]: Entering directory '/usr/lib/modules/5.10.61-1-MANJARO/build'
  CC [M]  /home/linux/rtw89/debug.o
/home/linux/rtw89/debug.c: In function ‘rtw89_debug_priv_txpwr_table_get’:
/home/linux/rtw89/debug.c:564:9: error: implicit declaration of function ‘tw89_print_sar’ [-Werror=implicit-function-declaration]
  564 |         tw89_print_sar(m, rtwdev);
      |         ^~~~~~~~~~~~~~
cc1: some warnings being treated as errors
make[2]: *** [scripts/Makefile.build:280: /home/linux/rtw89/debug.o] Error 1
make[1]: *** [Makefile:1822: /home/linux/rtw89] Error 2
make[1]: Leaving directory '/usr/lib/modules/5.10.61-1-MANJARO/build'
make: *** [Makefile:54: all] Error 2`

lwfinger, 2021-09-15:
The basic sar codes in the kernel were not defined until kernel 5.11, thus the code where your error occurred is inside an 
#ifdef LINUX_VERSION_CODE >= KERNEL_VERSION(5, 11, 0). As you have kernel 5.10.16, the code should not be reached.

There was a mistake in debug.c. Do a 'git pull' and try again.

Incubant, 2021-09-15:
Maybe there is another problem.

/home/linux/rtw89/phy.c:996:15: error: implicit declaration of function ‘rtw89_query_sar’ [-Werror=implicit-function-declaration]
  996 |         sar = rtw89_query_sar(rtwdev);
      |               ^~~~~~~~~~~~~~~
cc1: some warnings being treated as errors

lwfinger, 2021-09-15:
I missed two other places. Pull and try again.

Incubant, 2021-09-15:
i tried (with not much experience in programming c) to just surround the rtw89_query_sar and not the return function and it compiled and reenabled my wifi **woohoo**.

With your changes there are the following exceptions:

make -C /lib/modules/5.10.61-1-MANJARO/build M=/home/linux/rtw89 modules
make[1]: Entering directory '/usr/lib/modules/5.10.61-1-MANJARO/build'
  CC [M]  /home/linux/rtw89/phy.o
/home/linux/rtw89/phy.c: In function ‘rtw89_phy_read_txpwr_limit’:
/home/linux/rtw89/phy.c:981:21: warning: unused variable ‘sar’ [-Wunused-variable]
  981 |         s8 lmt = 0, sar;
      |                     ^~~
/home/linux/rtw89/phy.c: In function ‘rtw89_phy_read_txpwr_limit_ru’:
/home/linux/rtw89/phy.c:1129:16: error: ‘lmt’ undeclared (first use in this function)
 1129 |         return lmt;
      |                ^~~
/home/linux/rtw89/phy.c:1129:16: note: each undeclared identifier is reported only once for each function it appears in
/home/linux/rtw89/phy.c:1109:24: warning: unused variable ‘sar’ [-Wunused-variable]
 1109 |         s8 lmt_ru = 0, sar;
      |                        ^~~
/home/linux/rtw89/phy.c:1131:1: error: control reaches end of non-void function [-Werror=return-type]
 1131 | }
      | ^
cc1: some warnings being treated as errors

lwfinger, 2021-09-15:
I think I fixed those problems. Try another pull when you get a chance.

Incubant, 2021-09-15:
Its running without any warnings and exception. Thank you very much for reenabling my wiki since 3 weeks. Thank you very much.

lwfinger, 2021-09-15:
@domenkozar: I guess you have a low signal. It has been difficult for me to move my computer as the rtw89 is mounted on an external extender, but I did it today. Connecting in the 5 GHz band to an AC1900 access point, at a distance of 1 m I get a signal of -44dBm and down/up rates of 518/507 Mbps connecting to a LibreSpeed server on my local network. I then moved from my office to the kitchen about 5 m away, but with some walls in the way. That got a signal of -70 dBm and speeds of 29.7/37.2 Mbps.

Refresh my memory as to your signal and rates.

Incubant, 2021-09-16:
Sorry for the offtopic question, but I am a Linux newb, at the moment I have to install the rtw89 after every Manjaro package update. Is there a trick to avoid that. Very thankfully

lwfinger, 2021-09-16:
The short answer is NO. Unlike Windows, external drivers must be built using exactly the same compiler and kernel headers as were used to build the kernel. If you want to automate the process, implement dkms. I do not advise how to do this, but there are lots of articles.

closed 2021-09-16 by lwfinger