lwfinger/rtw89 #52: Bug report against linux

fthx, 2021-09-16:
Hi,

Your repo saved wy wifi, thanks for that!

But that's not a clean situation, so I did report:
https://bugs.launchpad.net/ubuntu/+source/linux/+bug/1943817

I don't understand what's wrong with default setup/install. I installed Ubuntu 21.10 dev and the driver is present in rtw89/ .
But no kernel module seems to be present. I did test 5.13 & 5.14 kernels here.

So this kernel module is simply missing or the issue is more complex?

lissyx, 2021-09-16:
> Hi,
>
> Your repo saved wy wifi, thanks for that!
>
> But that's not a clean situation, so I did report:
> https://bugs.launchpad.net/ubuntu/+source/linux/+bug/1943817
>
> I don't understand what's wrong with [...]

I believe from https://patchwork.kernel.org/project/linux-wireless/list/?series=534633&state=* that the driver is just not yet accepted upstream.

lissyx, 2021-09-16:
FTR, reading your launchpad bug, you might want to build the debian package out of the repo `debuild -b -us -uc`, this will produce you a dkms compatible driver and limit the pain over kernel upgrades.

fthx, 2021-09-16:
+= *
https://patchwork.kernel.org/project/linux-wireless/list/?series=534633&state=*
:-)
Thanks for that, I'll have a look.

fthx, 2021-09-16:
Aaaah ok, the star is not included by GitHub!!

lwfinger, 2021-09-16:
As noted above, the driver is still in review and is not accepted upstream. This repo is my means of supplying the driver in advance so you do not have to look at non-functioning hardware.

fthx, 2021-09-16:
Thanks.
Sorry for noise, I come from Intel wifi world. Not perfect at all, but with kernel modules. ;-)
Hope Ubuntu kernel team will do something.

Thanks @lissyx for the debian package tip.

closed 2021-09-16 by fthx