lwfinger/rtw89 #55: Wifi keeps disconnecting

C0D3-M4513R, 2021-10-03:
My Wifi keeps disconnecting. It does it pretty frequently. 
This log is with 30 minutes uptime. So far my wifi hasn't yet disconnected iirc.
I think, that these messages sound not good, so I just wanna have someone look over them, that knows some stuff.
[lspci.txt](https://github.com/lwfinger/rtw89/files/7274523/lspci.txt)
[test.txt](https://github.com/lwfinger/rtw89/files/7274517/test.txt)

lwfinger, 2021-10-04:
The lspci text told me nothing. The dmesg output shows some firmware errors. Have you tried the options for rtw89pci?

You can implement them by creating a file /etc/modprobe.d/50-rtw89.conf, and adding a line that says:

 options rtw89pci disable_aspm_l1=y

If that does not help, change the line to
 options rtw89pci disable_aspm_l1ss=y

If that still does not help, change the line to

options rtw89pci disable_aspm_l1ss=y disable_aspm_l1ss=y

If one of these helps, the problem is with your BIOS as it is not meeting specs.

C0D3-M4513R, 2021-10-04:
I will try these options. If the bios isn't meeting spec. I'm not surprised tbh. The wifi card is inside a lenovo laptop, which is intended for windows only.
Bios update utility and everything else windows only.
Many hw functions are toggled with some random pci call.
I have previously also done a bios update, that fixed something (iirc it was with suspend/hibernate 
/resume)
I have also seem some dmesg messages saying, that the bios is not doing what it expected.
The Bios options are also nonexistent. You can't even configure the boot order (but you can install custom secure boot keys).

C0D3-M4513R, 2021-10-04:
So after setting the first/three options, the fw errors have disappeared for now.
Now I am still curiously seeing some things:
```
[ 6303.660543] rtw89_pci 0000:03:00.0: timed out to flush queues
[ 6303.866373] rtw89_pci 0000:03:00.0: timed out to flush queues
[ 6304.277176] rtw89_pci 0000:03:00.0: timed out to flush queues
[ 6338.992785] rtw89_pci 0000:03:00.0: timed out to flush queues
```
Also the wifi interface Disconnects, due to unspecified reasons. (Probably the wifi card is failing to jump from ap's inside a meshed network.)
`bssid_part` there is always the same.
```
[  522.600473] wlo1: deauthenticated from bssid_part:3c (Reason: 1=UNSPECIFIED)
[  532.036432] wlo1: authenticate with bssid_part:12
[  532.245370] wlo1: send auth to bssid_part:12 (try 1/3)
[  532.246964] wlo1: authenticated
[  532.250595] wlo1: associate with bssid_part:12 (try 1/3)
[  532.252840] wlo1: RX AssocResp from bssid_part:12 (capab=0x1511 status=0 aid=1)
[  532.460752] wlo1: associated
[  536.299931] wlo1: deauthenticated from bssid_part:12 (Reason: 15=4WAY_HANDSHAKE_TIMEOUT)
```

C0D3-M4513R, 2021-10-08:
OK, so I'm not using a meshed wifi anymore, for other reasons.
I noticed that my wifi connection doesn't drop altogether anymore. I just cannot access anything.
Not even the router. Disconnecting and reconnecting makes everything works again?

lwfinger, 2021-10-09:
If you have access to a 5G router, try it. There will be less crowding.  A signal level of -45 dBm should not be a problem.

closed 2021-10-14 by C0D3-M4513R

C0D3-M4513R, 2022-03-06:
@raphink For me those logs look really normal. To me it looks, like the router may sometimes just not respond to auth requests? The fact, that it tries to connect means, that your computer is sending out a auth signal, but maybe the router just doesn't recive or respond to that auth signal. Try using a different ap maybe? It doesn't look like a problem on that computer from those logs.