lwfinger/rtw89 #57: rtw89pci NOT working on Lenovo Thinkpad L15 Ubuntu 20.04

gamicolo, 2021-10-04:
Hi,

I need some help to get the driver working.

I've installed the drive following the steps detailed on the github repo. The first installation was done with the SecureBoot option, and the message when I try to load the driver was:

modprobe: FATAL: Module rtw89pci not found in directory /lib/modules/5.10.0-1049-oem

After disable the SecureBoot option, when I try to load the driver its says:

modprobe: ERROR: could not insert 'rtw89pci': Exec format error

[lshw_network.txt](https://github.com/lwfinger/rtw89/files/7280790/lshw_network.txt)
[lspci.txt](https://github.com/lwfinger/rtw89/files/7280791/lspci.txt)
[lsusb.txt](https://github.com/lwfinger/rtw89/files/7280792/lsusb.txt)
[rfkill.txt](https://github.com/lwfinger/rtw89/files/7280793/rfkill.txt)
[uname.txt](https://github.com/lwfinger/rtw89/files/7280794/uname.txt)
[dmesg.txt](https://github.com/lwfinger/rtw89/files/7280788/dmesg.txt)
.

lwfinger, 2021-10-04:
In your dmesg.txt, did you notice the lines that say "rtw89core: disagrees about version of symbol module_layout"?

That message arises when your kernel include directory points to a different version and/or compiler than the running kernel.

Look at the output of the running kernel with 'uname -r'.
Then look at the first line logged after a 'make'.

Your versions match with 5.10.0-1049-oem". The dmesg.txt that you posted shows that the kernel was built with "gcc (Ubuntu 9.3.0-17ubuntu1~20.04) 9.3.0" The output of 'gcc --version' should match. If it does, then I have no idea why your module load is failing.

hlawan, 2021-10-05:
I had the same error message 

`modprobe: ERROR: could not insert 'rtw89pci': Exec format error`

on an L14 running Ubuntu 20.04.3 using secure boot (Kernel and gcc versions like yours). `gcc --version  also returned `Ubuntu 9.3.0-17ubuntu1~20.04`.

I got the error after I...
 -  creating a certificate for secure boot like described here: https://ubuntu.com/blog/how-to-sign-things-for-secure-boot
 -  running `make`
 - signing rtw89core.ko rtw89pci.ko with the new keys
 - running `sudo make install`
 - running `sudo modprobe rtw89pci`

What finally worked for me was:

- checkout the main branch
- run `make`
- run `sudo bash -c 'MOK_KEY_DIR=/PATH/TO/MY/NEW/KEYS make sign-install` (like described in the readme for distros other than ubuntu)
- run `sudo modprobe rtw89pci`
 
HTH

gamicolo, 2021-10-06:
Hi,

The solution that hlawan suggested seems to work. Nevertheless I'm little bit confused about the step to follow. What I did was:

1. enable the secure boot option
2. follow the steps detailed in https://ubuntu.com/blog/how-to-sign-things-for-secure-boot (here I get confused: the *.ko files need to be sign (according to the step "Let’s sign things")? If that's the case first we need to have the *.ko files running make?)
3. checkout the main branch
4. run make
5. run sudo bash -c 'MOK_KEY_DIR=/PATH/TO/MY/NEW/KEYS make sign-install' 
6. run sudo modprobe rtw89pci

The main question is if we need to sign the .ko files or not, I did it because I already have the .ko files from a previous branch (that rename before checkout a new main branch).

Another is why the installation without Secure Boot is not working. I check what lwfinger mentioned but I don't know what to do with it

uname -r
5.10.0-1049-oem

first line of make
make -C /lib/modules/5.10.0-1049-oem/build M=/home/gamicolo/development/git/rtw89 modules

Both seems to point to the same kernel version.

lwfinger, 2021-10-06:
The .ko files are what need signing.The process has two parts: (1) Sign the files with the key, and (2) Enter the key in the database. Your first sequence with secure boot skipped step 2. Your second set of steps picked that up. Step 2 needes to be done only once for a given key. Step 1 must be done each time the .ko files need to be remade, i.e. every time you run "make", and before "sudo make install".

I do not know what went wrong with the non-secure boot, but the fact that your secure boot steps worked means that the problem is solved. The addition of keys does not magically solve any other problems. It just lets the kernel load it when secure boot is enabled.

closed 2021-10-06 by lwfinger

Wilfredfury, 2021-11-07:
IdeaPad 5 Pro 14ACN6
`lspci -v | grep -i network`
```
01:00.0 Network controller: Realtek Semiconductor Co., Ltd. Device 8852
```
according to repository you're supposed to do
```
When your kernel changes, then you need to do the following:
cd ~/rtw89
git pull
make
sudo make install
```
which doesn't help and trying `sudo modprobe rtw89pci` results in the
```
modprobe: ERROR: could not insert 'rtw89pci': Exec format error
```
with `dmesg -t`
```
rtw89core: disagrees about version of symbol module_layout
```
doing
```
cd ~/rtw89
git pull
sudo make uninstall
sudo make clean
make
sudo make install
```
fixed this for me (I removed older kernel 37 first before I found the uninstall and clean commands for make)

by executing `modprobe --dump-modversions rtw89core.ko | grep module_layout`
`0x3364d393	module_layout`
and comparing it to your newly installed kernel module (5.11.0-38-generic corresponds to uname -r)
```
cd /lib/modules/5.11.0-38-generic/build
cat Module.symvers | grep module_layout
```
```
0x972f93ab	module_layout	vmlinux	EXPORT_SYMBOL
```
you can see that the symbols indeed don't match and you're having the module from old install.
Haven't found any 14ACN6 specific guides yet, so I hope this helps somebody in the future.

lwfinger, 2021-11-07:
You CANNOT build the module after installing a new kernel, but BEFORE you reboot. Reboot first!! In fact after the reboot, the make process will see that the headers have changed and rebuild all affected routines, thus 'make clean' is rarely needed, just 'make' and 'sudo make install'.