lwfinger/rtw89 #60: new linux user stuck after installing rtw89 firmware. still no wifi for 8852ae on 20.04 5.11 - help appreciated

speakernikesa, 2021-10-14:
here is some of what i have got so far. 

nikesa@nikesa-Yogajji-6-13:~/rtw89$ sudo make install
make -C /lib/modules/5.11.0-37-generic/build M=/home/nikesa/rtw89 modules
make[1]: Entering directory '/usr/src/linux-headers-5.11.0-37-generic'
make[1]: Leaving directory '/usr/src/linux-headers-5.11.0-37-generic'
cp *.bin /lib/firmware/rtw89/.
Install rtw89 SUCCESS
nikesa@nikesa-Yogajji-6-13:~/rtw89$ sudo modprobe rtw89pci
modprobe: ERROR: could not insert 'rtw89pci': Operation not permitted
nikesa@nikesa-Yogajji-6-13:~/rtw89$ cd ~/rtw89
nikesa@nikesa-Yogajji-6-13:~/rtw89$ git pull
Already up to date.
nikesa@nikesa-Yogajji-6-13:~/rtw89$

lshw -C 
 *-pci:1
             description: PCI bridge
             product: Renoir PCIe GPP Bridge [1022:1634]
             vendor: Advanced Micro Devices, Inc. [AMD] [1022]
             physical id: 2.4
             bus info: pci@0000:00:02.4
             version: 00
             width: 32 bits
             clock: 33MHz
             capabilities: pci normal_decode bus_master cap_list
             configuration: driver=pcieport
             resources: irq:34 ioport:2000(size=4096) memory:d0500000-d05fffff
           *-network UNCLAIMED
                description: Network controller
                product: Realtek Semiconductor Co., Ltd. [10EC:8852]
                vendor: Realtek Semiconductor Co., Ltd. [10EC]
                physical id: 0
                bus info: pci@0000:02:00.0
                version: 00
                width: 64 bits
                clock: 33MHz
                capabilities: cap_list
                configuration: latency=0
                resources: ioport:2000(size=256) memory:d0500000-d05fffff

When I try to load the module:

nikesa@nikesa-Yogajji-6-13:~$ cd ~/rtw89
nikesa@nikesa-Yogajji-6-13:~/rtw89$ sudo modprobe rtw89pci
modprobe: ERROR: could not insert 'rtw89pci': Operation not permitted
nikesa@nikesa-Yogajji-6-13:~/rtw89$

speakernikesa, 2021-10-14:
nikesa@nikesa-Yogajji-6-13:~/rtw89$ cd ~/rtw89
nikesa@nikesa-Yogajji-6-13:~/rtw89$ git pull
Already up to date.
nikesa@nikesa-Yogajji-6-13:~/rtw89$ make
make -C /lib/modules/5.11.0-37-generic/build M=/home/nikesa/rtw89 modules
make[1]: Entering directory '/usr/src/linux-headers-5.11.0-37-generic'
make[1]: Leaving directory '/usr/src/linux-headers-5.11.0-37-generic'
nikesa@nikesa-Yogajji-6-13:~/rtw89$ sudo make install
[sudo] password for nikesa: 
make -C /lib/modules/5.11.0-37-generic/build M=/home/nikesa/rtw89 modules
make[1]: Entering directory '/usr/src/linux-headers-5.11.0-37-generic'
make[1]: Leaving directory '/usr/src/linux-headers-5.11.0-37-generic'
cp *.bin /lib/firmware/rtw89/.
Install rtw89 SUCCESS
nikesa@nikesa-Yogajji-6-13:~/rtw89$

fthx, 2021-11-07:
https://github.com/lwfinger/rtw89/issues/61#issuecomment-962506179

speakernikesa, 2021-11-07:
> #61 (comment)

Just installed 5.15.1 and still no wifi showing up

lwfinger, 2021-11-08:
You did see the "after" in my statement. That means 5.16. With some distros, it has been backported to 5.15, or 5.14, but not on yours. You need to clone the git repo, and make the driver in the way described here.

fthx, 2021-11-08:
In my previous message, I point to Ubuntu's kernel, which does have a rtw89 backport since 5.15.0.8-8.

petereth, 2021-11-26:
hi, how could I find the 
The V7 version of rtw89 driver 
https://patchwork.kernel.org/project/linux-wireless/list/?series=559649&state=*

how to install on Lenovo Legion 5 Pro?

thx

lwfinger, 2021-11-27:
git clone https://github.com/lwfinger/rtw89.git -b v7
cd rtw89
make
sudo make install
sudo modprobe -v rtw89pci

petereth, 2021-11-27:
> 

thank you very much done but after:

sudo modprobe -v rtw89pci

this error
[image]

how to fix?
thx

lwfinger, 2021-11-27:
The reason it failed is listed with more detail in the log output of the dmesg command. If you have secure boot enabled, then follow the prescription in https://github.com/lwfinger/rtw89/issues/74.

petereth, 2021-12-03:
> The reason it failed is listed with more detail in the log output of the dmesg command. If you have secure boot enabled, then follow the prescription in #74.

I make this but after make and this two lines
/usr/src/linux-headers-$(uname -r)/scripts/sign-file sha256 /home/$(echo $USER)/mok/MOK.priv /home/$(echo $USER)/mok/MOK.der rtw89core.ko
/usr/src/linux-headers-$(uname -r)/scripts/sign-file sha256 /home/$(echo $USER)/mok/MOK.priv /home/$(echo $USER)/mok/MOK.der rtw89pci.ko

this happend
root@marlies-Legion-5-15ACH6H:/home/marlies# cd rtw89/
root@marlies-Legion-5-15ACH6H:/home/marlies/rtw89# make
make -C /lib/modules/5.13.0-22-generic/build M=/home/marlies/rtw89 modules
make[1]: Verzeichnis „/usr/src/linux-headers-5.13.0-22-generic“ wird betreten
  CC [M]  /home/marlies/rtw89/core.o
  CC [M]  /home/marlies/rtw89/debug.o
  CC [M]  /home/marlies/rtw89/mac80211.o
  CC [M]  /home/marlies/rtw89/mac.o
  CC [M]  /home/marlies/rtw89/phy.o
  CC [M]  /home/marlies/rtw89/fw.o
  CC [M]  /home/marlies/rtw89/rtw8852a.o
  CC [M]  /home/marlies/rtw89/rtw8852a_table.o
  CC [M]  /home/marlies/rtw89/rtw8852a_rfk.o
  CC [M]  /home/marlies/rtw89/rtw8852a_rfk_table.o
  CC [M]  /home/marlies/rtw89/cam.o
  CC [M]  /home/marlies/rtw89/efuse.o
  CC [M]  /home/marlies/rtw89/regd.o
  CC [M]  /home/marlies/rtw89/coex.o
  CC [M]  /home/marlies/rtw89/ps.o
  CC [M]  /home/marlies/rtw89/sar.o
  CC [M]  /home/marlies/rtw89/ser.o
  LD [M]  /home/marlies/rtw89/rtw89core.o
  CC [M]  /home/marlies/rtw89/pci.o
  LD [M]  /home/marlies/rtw89/rtw89pci.o
  MODPOST /home/marlies/rtw89/Module.symvers
  CC [M]  /home/marlies/rtw89/rtw89core.mod.o
  LD [M]  /home/marlies/rtw89/rtw89core.ko
  BTF [M] /home/marlies/rtw89/rtw89core.ko
Skipping BTF generation for /home/marlies/rtw89/rtw89core.ko due to unavailability of vmlinux
  CC [M]  /home/marlies/rtw89/rtw89pci.mod.o
  LD [M]  /home/marlies/rtw89/rtw89pci.ko
  BTF [M] /home/marlies/rtw89/rtw89pci.ko
Skipping BTF generation for /home/marlies/rtw89/rtw89pci.ko due to unavailability of vmlinux
make[1]: Verzeichnis „/usr/src/linux-headers-5.13.0-22-generic“ wird verlassen
root@marlies-Legion-5-15ACH6H:/home/marlies/rtw89# /usr/src/linux-headers-$(uname -r)/scripts/sign-file sha256 /home/$(echo $USER)/mok/MOK.priv /home/$(echo $USER)/mok/MOK.der rtw89core.ko
At main.c:160:
- SSL error:02001002:system library:fopen:No such file or directory: ../crypto/bio/bss_file.c:69
- SSL error:2006D080:BIO routines:BIO_new_file:no such file: ../crypto/bio/bss_file.c:76
sign-file: /home/root/mok/MOK.priv: No such file or directory
root@marlies-Legion-5-15ACH6H:/home/marlies/rtw89# /usr/src/linux-headers-$(uname -r)/scripts/sign-file sha256 /home/$(echo $USER)/mok/MOK.priv /home/$(echo $USER)/mok/MOK.der rtw89pci.ko
At main.c:160:
- SSL error:02001002:system library:fopen:No such file or directory: ../crypto/bio/bss_file.c:69
- SSL error:2006D080:BIO routines:BIO_new_file:no such file: ../crypto/bio/bss_file.c:76
sign-file: /home/root/mok/MOK.priv: No such file or directory
root@marlies-Legion-5-15ACH6H:/home/marlies/rtw89#

petereth, 2021-12-03:
I try this root@marlies-Legion-5-15ACH6H:/home/marlies/rtw89# sudo make install
make -C /lib/modules/5.13.0-22-generic/build M=/home/marlies/rtw89 modules
make[1]: Verzeichnis „/usr/src/linux-headers-5.13.0-22-generic“ wird betreten
make[1]: Verzeichnis „/usr/src/linux-headers-5.13.0-22-generic“ wird verlassen
Install rtw89 SUCCESS
root@marlies-Legion-5-15ACH6H:/home/marlies/rtw89#

closed 2022-12-14 by lwfinger