lwfinger/rtw89 #69: Unable to create wifi-hotspot as adapter does not support AP (master) mode

shivamnaik39, 2021-11-04:
this was the tool i was using....
`sudo lnxrouter --ap wlo1 MyAccessPoint -p MyPassPhrase`
`ERROR: Your adapter does not support AP (master) mode`

the output of `iw list` also clearly states that this does not support AP mode but on windows I am able to create wifi-hotspot.
so is there any way I can achieve this on linux, do I need to install some special dependency or different version of particular dependency?  I am currently using **Arch Linux with kernel 5.14.14** tried it on lts kernel, it doesn' t work.  btw normal wifi and bluetooth works very well with the help of this driver.

w407022008, 2021-12-11:
I am using a RTL8852AE wireless card and I am experiencing the same problem. Has a solution been found please?

lwfinger, 2021-12-11:
That feature has not yet been added as most people do not want to turn their $1000 laptop into a $50 router, thus it has very low priority.

If you really need it, buy a $10 wifi dongle with an RT5370 chip in it and base your router on it.

w407022008, 2021-12-11:
Thx any way, I have another wifi dongle with rtl8812bu..  I will have a test with it.

mailinglists35, 2022-08-26:
> very low priority.

So this means when you have nothing to and in the right mood you would be able to give this a chance?

I don't have a laptop but a mini pc and I would like to use the mini pcie slot inside it instead of keeping in use and hot an usb port with an usb adapter that works poorer than a pcie chip. Also came here after discovering that the one I wanted doesn't support AP mode on 5GHz (intel ax200)

lwfinger, 2022-08-26:
You are new here so I will educate you. I did not write the driver. I have no knowledge of the internals of the chip. Even if I wanted to, I would have little idea on what is needed to implement AP mode. My expertise is in the code between the chip and the OS.

If you want a new feature added, follow the instructions in the last paragraph of README.md.

closed 2022-08-26 by lwfinger

mailinglists35, 2022-08-27:
thanks @lwfinger for the explanation
I was under the wrong assumption, apologies if this has caused any nuisance.

lwfinger, 2022-08-27:
No problem. I do try to fix the problems where I have some ability, but adding features is not something I try.