lwfinger/rtw89 #72: T14 AMD Gen 2 5850U random reboots

nick-st, 2021-11-09:
Hi there, 
Thanks for the great work and effort for this driver. I know I'm supposed to submit possible issue through linux-wireless@vger.kernel.org, but for some reason this email doesn't work. 

I experienced a couple of sudden system reboots for no apparent reason. Previously I was using Ubuntu 20.04 and thought it is some version issue. Currently I used 21.10 with kernel 5.13.0-20-generic. The link below contains the system log and it seems like it is caused by the wifi module. 
https://github.com/nick-st/T14AMD5850U/blob/master/RandomRebootLog2.txt

Just for clarity I'm using the newest version.

lwfinger, 2021-11-09:
The reason you cannot post to linux-wireless@vger.kernel.org is that you are trying to send html-formatted messages. Only PLAIN TEXT mail is permitted. If you use Thunderbird, that is a setting.

I suspect that your BIOS is not up to specs. A lot of Lenovo models have that problem. To test, as root create a file named /etc/modprobe.d/50-rtw89.conf and add the following line:
options rtw89pci disable_aspm_l1=y disable_aspm_l1ss=y

Next run 'sudo modprobe -rv rtw89pci' and 'sudo modprobe -v rtw89pci'. If you did it right, you will see those two options listed at the end of the rtw89pci line.

nick-st, 2021-11-09:
Thanks for the prompt reply. These options did appear as described. Regarding your BIOS suggestion - the reboot happened 3 times and each time there was different BIOS version -1.09, 1.11, 1.12. 
$ sudo dmidecode -s bios-version
R1MET42W (1.12 )
Thanks

mwilck, 2021-12-12:
@nick-st, I think @lwfinger was asking whether your problem went away after adding the options he proposed.

nick-st, 2021-12-13:
Hi @lwfinger and @mwilck, apologies, probably I misunderstood. Unfortunately the problem still persists. I talked to someone from linux-wireless@vger.kernel.org and they suggested an additional parameter:

/etc/modprobe.d/50-rtw89.conf
options rtw89pci disable_aspm_l1=y disable_aspm_l1ss=y disable_clkreq=y

Also they advised me to setup netconsole since the log is truncated. To do this, I need an additional PC with Ethernet, which I don't have right now since I'm away from home. The PC rebooted once or twice over the last month (after I applied the options above). 
Thanks,
Nick

lwfinger, 2021-12-13:
Without knowing why it reboots, there is nothing any of us can do. By the way, the person you communicated with on linux-wireless is the actual author of the driver. If he has no further suggestions, there are none until you get that console dump.

white-gecko, 2022-01-09:
@nick-st could you please link the conversation on the kernel list here?

nick-st, 2022-01-09:
@white-gecko I'm not sure I understand which conversation you mean? Please give me more details. 
By the way as a general update I set up the netconsole, but there is no update since I haven't experienced a reboot over the last month or so. I'll keep this issue open for now.

white-gecko, 2022-01-09:
I was referring to you mail to the kernel mailing list. I think I found the correct one in the archive:
- https://lore.kernel.org/linux-wireless/15846b1b-5cad-796d-b562-1ab31743fd82@gmail.com/
- https://lore.kernel.org/linux-wireless/16486a1c4ec348318fd35c0d6ca6a454@realtek.com/

Please keep us updated on the outcome of your netconsole.

lwfinger, 2022-01-09:
If you are using the module parameters recommended in the second mail exchange listed above, AND the reboots stopped, then I think you found the problem. By the way, those reboots are actually caused by your BIOS not behaving properly.

closed 2022-01-09 by lwfinger

white-gecko, 2022-01-09:
Thanks for the clarification.
Is there any BIOS Update/specific version required or any setting that can be made in the BIOS?

lwfinger, 2022-01-09:
There are no BIOS settings that would help this problem. Any updates would have to be issued by your laptop vendor; however, they are not particularly reactive to Linux problems. If the Windows driver shows the same diffuculty, then there might be a fix.

eshimoniak, 2022-01-25:
One thing I've noticed is that the reboots always seem to happen for me during video calls (either Zoom or Discord), so perhaps it's something to do with sustained, high bandwidth usage?

Also, what are the chances that a new WiFi card (of a different model, like an AX200) would fix this issue? Is it something with this driver specifically that doesn't play nice with the BIOS, or something about the Linux network stack in general?

lwfinger, 2022-01-25:
Your guess is as good as mine. I use my rtw89 to do backups, which is about as high a bandwidth as one could expect. I see no crashes.

Until you can specify what event is actually causing the kernel BUG and the reboot, there is not much I can do. If you have a second computer available, then netconsole would give you the required information. I seem to recall that a Windows computer can be used, although it is easier to set up with Linux.

eshimoniak, 2022-01-25:
I actually do happen to have a second Linux machine available. What's involved in setting up netconsole?

white-gecko, 2022-01-25:
This provides some nice advice on how to setup a netconsole: https://wiki.ubuntu.com/Kernel/Netconsole

lwfinger, 2022-01-25:
That one is a little intense, but it does cover all the bases. If you have any problems with it, there are other web articles on the subject. Search for "netconsole setup". The main thing is that your kernel must have built netconsole. 'sudo modprobe -v netconsole' will verify that.