lwfinger/rtw89 #73: Very frequent driver lockups only resolvable by restarting wifi interface

jonded94, 2021-11-11:
Hi!

At first, let me thank you for your work making my new Thinkpad T14s usable on mobile work.

Unfortunately, I experience very frequent (i.e. sometimes multiple times per minute) hard driver lockups which are only resolvable by restarting the entire wifi interface.

I'm using:

```
$ uname -a
Linux XXXX-ThinkPad-T14s-Gen-1 5.13.0-21-generic #21-Ubuntu SMP Tue Oct 19 08:59:28 UTC 2021 x86_64 x86_64 x86_64 GNU/Linux
$ lsb_release -a
No LSB modules are available.
Distributor ID:	Ubuntu
Description:	Ubuntu 21.10
Release:	21.10
Codename:	impish
```

I'm on the current version of the driver:

```
$ cd rtw89 && git rev-parse HEAD
c3a461aa70086832c1d9c6f58d1477b3ab9cc151
```

I already created a file under `/etc/modprobe.d/` with the options mentioned in the README and some issues already:

```
$ cat /etc/modprobe.d/rtw89pci.conf 
options rtw89pci disable_aspm_l1=y
options rtw89pci disable_aspm_l1ss=y
```

For some reason, this does not really have an effect on the hangups?

This is the output of `/var/log/syslog/ | grep rtw89` whenever I experience a hangup:

```
Nov 11 13:20:00 jonas-ThinkPad-T14s-Gen-1 kernel: [16356.241094] rtw89_pci 0000:03:00.0: FW status = 0x5f001140
Nov 11 13:20:00 jonas-ThinkPad-T14s-Gen-1 kernel: [16356.241104] rtw89_pci 0000:03:00.0: FW BADADDR = 0x0
Nov 11 13:20:00 jonas-ThinkPad-T14s-Gen-1 kernel: [16356.241109] rtw89_pci 0000:03:00.0: FW EPC/RA = 0xb8978f81
Nov 11 13:20:00 jonas-ThinkPad-T14s-Gen-1 kernel: [16356.241113] rtw89_pci 0000:03:00.0: FW MISC = 0xb8990c7b
Nov 11 13:20:00 jonas-ThinkPad-T14s-Gen-1 kernel: [16356.241117] rtw89_pci 0000:03:00.0: R_AX_HALT_C2H = 0x1000
Nov 11 13:20:00 jonas-ThinkPad-T14s-Gen-1 kernel: [16356.241121] rtw89_pci 0000:03:00.0: R_AX_SER_DBG_INFO = 0xf1010010
Nov 11 13:20:00 jonas-ThinkPad-T14s-Gen-1 kernel: [16356.241128] rtw89_pci 0000:03:00.0: [ERR]fw PC = 0xb8991ac7
Nov 11 13:20:00 jonas-ThinkPad-T14s-Gen-1 kernel: [16356.241402] rtw89_pci 0000:03:00.0: [ERR]fw PC = 0xb8991bd1
Nov 11 13:20:00 jonas-ThinkPad-T14s-Gen-1 kernel: [16356.241418] rtw89_pci 0000:03:00.0: [ERR]fw PC = 0xb8991ac5
Nov 11 13:20:00 jonas-ThinkPad-T14s-Gen-1 kernel: [16356.241433] rtw89_pci 0000:03:00.0: [ERR]fw PC = 0xb8991bb9
Nov 11 13:20:00 jonas-ThinkPad-T14s-Gen-1 kernel: [16356.241449] rtw89_pci 0000:03:00.0: [ERR]fw PC = 0xb8991bd5
Nov 11 13:20:00 jonas-ThinkPad-T14s-Gen-1 kernel: [16356.241464] rtw89_pci 0000:03:00.0: [ERR]fw PC = 0xb8991bc3
Nov 11 13:20:00 jonas-ThinkPad-T14s-Gen-1 kernel: [16356.241479] rtw89_pci 0000:03:00.0: [ERR]fw PC = 0xb8991bc7
Nov 11 13:20:00 jonas-ThinkPad-T14s-Gen-1 kernel: [16356.241494] rtw89_pci 0000:03:00.0: [ERR]fw PC = 0xb8991ae7
Nov 11 13:20:00 jonas-ThinkPad-T14s-Gen-1 kernel: [16356.241509] rtw89_pci 0000:03:00.0: [ERR]fw PC = 0xb8991acd
Nov 11 13:20:00 jonas-ThinkPad-T14s-Gen-1 kernel: [16356.241524] rtw89_pci 0000:03:00.0: [ERR]fw PC = 0xb8991bbd
Nov 11 13:20:00 jonas-ThinkPad-T14s-Gen-1 kernel: [16356.241540] rtw89_pci 0000:03:00.0: [ERR]fw PC = 0xb8991ba7
Nov 11 13:20:00 jonas-ThinkPad-T14s-Gen-1 kernel: [16356.241555] rtw89_pci 0000:03:00.0: [ERR]fw PC = 0xb8991ad5
Nov 11 13:20:00 jonas-ThinkPad-T14s-Gen-1 kernel: [16356.241570] rtw89_pci 0000:03:00.0: [ERR]fw PC = 0xb8991ac1
Nov 11 13:20:00 jonas-ThinkPad-T14s-Gen-1 kernel: [16356.241585] rtw89_pci 0000:03:00.0: [ERR]fw PC = 0xb8991ab1
Nov 11 13:20:00 jonas-ThinkPad-T14s-Gen-1 kernel: [16356.241600] rtw89_pci 0000:03:00.0: [ERR]fw PC = 0xb8991bb1
Nov 11 13:20:00 jonas-ThinkPad-T14s-Gen-1 kernel: [16356.241614] rtw89_pci 0000:03:00.0: --->
Nov 11 13:20:00 jonas-ThinkPad-T14s-Gen-1 kernel: [16356.241618] rtw89_pci 0000:03:00.0: R_AX_SER_DBG_INFO =0xf1010010
Nov 11 13:20:00 jonas-ThinkPad-T14s-Gen-1 kernel: [16356.241622] rtw89_pci 0000:03:00.0: R_AX_CMAC_ERR_ISR =0x00000002
Nov 11 13:20:00 jonas-ThinkPad-T14s-Gen-1 kernel: [16356.241627] rtw89_pci 0000:03:00.0: R_AX_DMAC_ERR_ISR =0x00000040
Nov 11 13:20:00 jonas-ThinkPad-T14s-Gen-1 kernel: [16356.241631] rtw89_pci 0000:03:00.0: R_AX_WDE_ERR_FLAG_CFG =0x00000000 
Nov 11 13:20:00 jonas-ThinkPad-T14s-Gen-1 kernel: [16356.241635] rtw89_pci 0000:03:00.0: R_AX_PLE_ERR_FLAG_CFG =0x8e020000
Nov 11 13:20:00 jonas-ThinkPad-T14s-Gen-1 kernel: [16356.241640] rtw89_pci 0000:03:00.0: R_AX_WDE_ERR_IMR=0xffffffff 
Nov 11 13:20:00 jonas-ThinkPad-T14s-Gen-1 kernel: [16356.241644] rtw89_pci 0000:03:00.0: R_AX_WDE_ERR_ISR=0x00000000
Nov 11 13:20:00 jonas-ThinkPad-T14s-Gen-1 kernel: [16356.241648] rtw89_pci 0000:03:00.0: R_AX_PLE_ERR_IMR=0xffffffdf 
Nov 11 13:20:00 jonas-ThinkPad-T14s-Gen-1 kernel: [16356.241652] rtw89_pci 0000:03:00.0: R_AX_PLE_ERR_FLAG_ISR=0x00000004
[... 181 lines cut ...]
Nov 11 13:20:47 jonas-ThinkPad-T14s-Gen-1 kernel: [16403.381972] rtw89_pci 0000:03:00.0: timed out to flush queues
Nov 11 13:20:47 jonas-ThinkPad-T14s-Gen-1 kernel: [16403.591943] rtw89_pci 0000:03:00.0: timed out to flush queues
Nov 11 13:20:47 jonas-ThinkPad-T14s-Gen-1 kernel: [16404.063802] rtw89_pci 0000:03:00.0: timed out to flush queues
Nov 11 13:20:48 jonas-ThinkPad-T14s-Gen-1 kernel: [16404.264496] rtw89_pci 0000:03:00.0: timed out to flush queues
Nov 11 13:20:48 jonas-ThinkPad-T14s-Gen-1 kernel: [16404.788209] rtw89_pci 0000:03:00.0: timed out to flush queues
Nov 11 13:20:48 jonas-ThinkPad-T14s-Gen-1 kernel: [16404.995754] rtw89_pci 0000:03:00.0: timed out to flush queues
Nov 11 13:20:49 jonas-ThinkPad-T14s-Gen-1 kernel: [16405.475911] rtw89_pci 0000:03:00.0: timed out to flush queues
Nov 11 13:20:49 jonas-ThinkPad-T14s-Gen-1 kernel: [16405.683718] rtw89_pci 0000:03:00.0: timed out to flush queues
Nov 11 13:20:49 jonas-ThinkPad-T14s-Gen-1 kernel: [16406.207753] rtw89_pci 0000:03:00.0: timed out to flush queues
Nov 11 13:20:50 jonas-ThinkPad-T14s-Gen-1 kernel: [16406.409895] rtw89_pci 0000:03:00.0: timed out to flush queues
Nov 11 13:20:50 jonas-ThinkPad-T14s-Gen-1 kernel: [16406.952577] rtw89_pci 0000:03:00.0: timed out to flush queues
Nov 11 13:20:50 jonas-ThinkPad-T14s-Gen-1 kernel: [16407.155980] rtw89_pci 0000:03:00.0: timed out to flush queues
Nov 11 13:20:51 jonas-ThinkPad-T14s-Gen-1 kernel: [16407.691751] rtw89_pci 0000:03:00.0: timed out to flush queues
Nov 11 13:20:51 jonas-ThinkPad-T14s-Gen-1 kernel: [16407.893789] rtw89_pci 0000:03:00.0: timed out to flush queues
Nov 11 13:20:52 jonas-ThinkPad-T14s-Gen-1 kernel: [16408.415751] rtw89_pci 0000:03:00.0: timed out to flush queues
Nov 11 13:20:52 jonas-ThinkPad-T14s-Gen-1 kernel: [16408.622250] rtw89_pci 0000:03:00.0: timed out to flush queues
Nov 11 13:20:52 jonas-ThinkPad-T14s-Gen-1 kernel: [16409.145792] rtw89_pci 0000:03:00.0: timed out to flush queues
Nov 11 13:20:53 jonas-ThinkPad-T14s-Gen-1 kernel: [16409.352391] rtw89_pci 0000:03:00.0: timed out to flush queues
Nov 11 13:20:53 jonas-ThinkPad-T14s-Gen-1 kernel: [16409.870168] rtw89_pci 0000:03:00.0: timed out to flush queues
Nov 11 13:20:53 jonas-ThinkPad-T14s-Gen-1 kernel: [16410.072720] rtw89_pci 0000:03:00.0: timed out to flush queues
Nov 11 13:20:54 jonas-ThinkPad-T14s-Gen-1 kernel: [16410.596024] rtw89_pci 0000:03:00.0: timed out to flush queues
Nov 11 13:20:54 jonas-ThinkPad-T14s-Gen-1 kernel: [16410.802020] rtw89_pci 0000:03:00.0: timed out to flush queues
Nov 11 13:20:55 jonas-ThinkPad-T14s-Gen-1 kernel: [16411.316075] rtw89_pci 0000:03:00.0: timed out to flush queues
Nov 11 13:20:55 jonas-ThinkPad-T14s-Gen-1 kernel: [16411.519697] rtw89_pci 0000:03:00.0: timed out to flush queues
```

gnu4ever, 2021-11-25:
Same problem here.

damien-talos, 2021-11-25:
Same issue here, logs show the same pattern as above.
```
λ uname -a
Linux pop-os 5.13.0-7620-generic #20~1634827117~21.04~874b071-Ubuntu SMP Fri Oct 29 15:06:55 UTC  x86_64 x86_64 x86_64 GNU/Linux
λ lsb_release -a
No LSB modules are available.
Distributor ID:	Pop
Description:	Pop!_OS 21.04
Release:	21.04
Codename:	hirsute
```

raphink, 2022-03-02:
On Ubuntu (and derivatives), the module is called `rtw89_pci` instead of `rtw89pci`. So `/etc/modprobe.d/50-rtw89.conf` needs to read:

```
options rtw89_pci disable_aspm_l1=y
options rtw89_pci disable_aspm_l1ss=y
```

You can check that the values where taken into account by looking into the module parameters:

```
❯ for i in /sys/module/rtw89_pci/parameters/disable_*; do echo $i; cat $i; done
/sys/module/rtw89_pci/parameters/disable_aspm_l1
Y
/sys/module/rtw89_pci/parameters/disable_aspm_l1ss
Y
/sys/module/rtw89_pci/parameters/disable_clkreq
N
```

lwfinger, 2022-03-02:
Names such as rtw89_pci indicate that the driver comes from the in-kernel version. Names such as rtw89pci are specifically for this repo so that I can tell the difference.

closed 2022-03-02 by lwfinger

raphink, 2022-03-02:
That said, even after disabling all 3 options (with the underscore since I'm on Ubuntu), I still get disconnections with logs like:

<details>
  <summary>`dmesg -w` during disconnection</summary>
  
```
[  697.819690] rtw89_pci 0000:03:00.0: timed out to flush queues
[  698.547520] wlp3s0: authenticate with 38:43:7d:87:11:87
[  698.772879] wlp3s0: send auth to 38:43:7d:87:11:87 (try 1/3)
[  698.777794] wlp3s0: authenticated
[  698.778834] wlp3s0: associate with 38:43:7d:87:11:87 (try 1/3)
[  698.795229] wlp3s0: RX AssocResp from 38:43:7d:87:11:87 (capab=0x1411 status=0 aid=4)
[  698.906763] wlp3s0: associated
[  698.968718] wlp3s0: Limiting TX power to 20 (20 - 0) dBm as advertised by 38:43:7d:87:11:87
[  903.382222] rtw89_pci 0000:03:00.0: FW status = 0xb8001228
[  903.382229] rtw89_pci 0000:03:00.0: FW BADADDR = 0x0
[  903.382232] rtw89_pci 0000:03:00.0: FW EPC/RA = 0xb897be7b
[  903.382235] rtw89_pci 0000:03:00.0: FW MISC = 0xb898faa3
[  903.382238] rtw89_pci 0000:03:00.0: R_AX_HALT_C2H = 0x30000011
[  903.382240] rtw89_pci 0000:03:00.0: R_AX_SER_DBG_INFO = 0xf0000001
[  903.382246] rtw89_pci 0000:03:00.0: [ERR]fw PC = 0xb893a697
[  903.382262] rtw89_pci 0000:03:00.0: [ERR]fw PC = 0xb89c325b
[  903.382276] rtw89_pci 0000:03:00.0: [ERR]fw PC = 0xb8939a8d
[  903.382289] rtw89_pci 0000:03:00.0: [ERR]fw PC = 0xb897ce3d
[  903.382303] rtw89_pci 0000:03:00.0: [ERR]fw PC = 0xb89a0cb1
[  903.382316] rtw89_pci 0000:03:00.0: [ERR]fw PC = 0xb89a0cb9
[  903.382329] rtw89_pci 0000:03:00.0: [ERR]fw PC = 0xb899041d
[  903.382342] rtw89_pci 0000:03:00.0: [ERR]fw PC = 0xb897bd95
[  903.382356] rtw89_pci 0000:03:00.0: [ERR]fw PC = 0xb89a0cbb
[  903.382369] rtw89_pci 0000:03:00.0: [ERR]fw PC = 0xb89c212d
[  903.382382] rtw89_pci 0000:03:00.0: [ERR]fw PC = 0xb897bb73
[  903.382396] rtw89_pci 0000:03:00.0: [ERR]fw PC = 0xb89b7fbb
[  903.382409] rtw89_pci 0000:03:00.0: [ERR]fw PC = 0xb897bb73
[  903.382422] rtw89_pci 0000:03:00.0: [ERR]fw PC = 0xb897be37
[  903.382435] rtw89_pci 0000:03:00.0: [ERR]fw PC = 0xb89a0d21
[  903.382447] rtw89_pci 0000:03:00.0: ser event = 0x30000011
[  903.401239] rtw89_pci 0000:03:00.0: FW status = 0xb8001228
[  903.401269] rtw89_pci 0000:03:00.0: FW BADADDR = 0x0
[  903.401273] rtw89_pci 0000:03:00.0: FW EPC/RA = 0xb897be7b
[  903.401276] rtw89_pci 0000:03:00.0: FW MISC = 0xb898faa1
[  903.401279] rtw89_pci 0000:03:00.0: R_AX_HALT_C2H = 0x30000011
[  903.401282] rtw89_pci 0000:03:00.0: R_AX_SER_DBG_INFO = 0xf0000081
[  903.401292] rtw89_pci 0000:03:00.0: [ERR]fw PC = 0xb8935221
[  903.401318] rtw89_pci 0000:03:00.0: [ERR]fw PC = 0xb89b3c17
[  903.401332] rtw89_pci 0000:03:00.0: [ERR]fw PC = 0xb89b3c1b
[  903.401345] rtw89_pci 0000:03:00.0: [ERR]fw PC = 0xb89b3c13
[  903.401360] rtw89_pci 0000:03:00.0: [ERR]fw PC = 0xb8915953
[  903.401373] rtw89_pci 0000:03:00.0: [ERR]fw PC = 0xb89159ad
[  903.401386] rtw89_pci 0000:03:00.0: [ERR]fw PC = 0xb89b3bbd
[  903.401555] rtw89_pci 0000:03:00.0: [ERR]fw PC = 0xb89b3c13
[  903.401582] rtw89_pci 0000:03:00.0: [ERR]fw PC = 0xb89b3c15
[  903.401596] rtw89_pci 0000:03:00.0: [ERR]fw PC = 0xb8915953
[  903.401609] rtw89_pci 0000:03:00.0: [ERR]fw PC = 0xb89159ad
[  903.401621] rtw89_pci 0000:03:00.0: [ERR]fw PC = 0xb89159b3
[  903.401634] rtw89_pci 0000:03:00.0: [ERR]fw PC = 0xb89b3c05
[  903.401648] rtw89_pci 0000:03:00.0: [ERR]fw PC = 0xb89b3c17
[  903.401662] rtw89_pci 0000:03:00.0: [ERR]fw PC = 0xb89b3c19
[  903.401677] rtw89_pci 0000:03:00.0: ser event = 0x30000011
[  903.401736] rtw89_pci 0000:03:00.0: FW status = 0xb8001228
[  903.401739] rtw89_pci 0000:03:00.0: FW BADADDR = 0x0
[  903.401742] rtw89_pci 0000:03:00.0: FW EPC/RA = 0xb897be7b
[  903.401746] rtw89_pci 0000:03:00.0: FW MISC = 0xb898faa1
[  903.401748] rtw89_pci 0000:03:00.0: R_AX_HALT_C2H = 0x30000011
[  903.401753] rtw89_pci 0000:03:00.0: R_AX_SER_DBG_INFO = 0xf0000081
[  903.401761] rtw89_pci 0000:03:00.0: [ERR]fw PC = 0xb8935247
[  903.401827] rtw89_pci 0000:03:00.0: [ERR]fw PC = 0xb89b3c15
[  903.401846] rtw89_pci 0000:03:00.0: [ERR]fw PC = 0xb89b3c15
[  903.401861] rtw89_pci 0000:03:00.0: [ERR]fw PC = 0xb89b3c19
[  903.401875] rtw89_pci 0000:03:00.0: [ERR]fw PC = 0xb89b3c1b
[  903.401889] rtw89_pci 0000:03:00.0: [ERR]fw PC = 0xb8915953
[... 649 lines cut ...]
[  952.225030] rtw89_pci 0000:03:00.0: firmware failed to ack for leaving ps mode
[  954.239942] rtw89_pci 0000:03:00.0: firmware failed to ack for leaving ps mode
```
</details>

<details>
  <summary>Stopping and restarting Wifi</summary>

```
[  955.672445] wlp3s0: deauthenticating from 38:43:7d:87:11:87 by local choice (Reason: 3=DEAUTH_LEAVING)
[  955.688296] rtw89_pci 0000:03:00.0: firmware failed to ack for leaving ps mode
[  956.095121] rtw89_pci 0000:03:00.0: timed out to flush queues
[  956.295479] rtw89_pci 0000:03:00.0: timed out to flush queues
[  957.295452] rtw89_pci 0000:03:00.0: c2h reg timeout
[  957.301550] rtw89_pci 0000:03:00.0: FW does not process h2c registers
[  968.317519] wlp3s0: authenticate with 38:43:7d:87:11:75
[  968.540353] wlp3s0: send auth to 38:43:7d:87:11:75 (try 1/3)
[  968.560495] wlp3s0: authenticated
[  968.560941] wlp3s0: associate with 38:43:7d:87:11:75 (try 1/3)
[  968.597691] wlp3s0: RX AssocResp from 38:43:7d:87:11:75 (capab=0x511 status=0 aid=5)
[  968.709050] wlp3s0: associated
[  968.709128] wlp3s0: Limiting TX power to 14 (17 - 3) dBm as advertised by 38:43:7d:87:11:75
[  968.785022] IPv6: ADDRCONF(NETDEV_CHANGE): wlp3s0: link becomes ready
```
</details>

raphink, 2022-03-02:
FTR, there is a bug report on Ubuntu at https://www.mail-archive.com/kernel-packages@lists.launchpad.net/msg471779.html