lwfinger/rtw89 #74: how to use sign make install ubuntu 21.10

XOMUS, 2021-11-12:
`mkdir ~/mok && cd ~/mok`
`openssl req -new -x509 -newkey rsa:2048 -keyout MOK.priv -outform DER -out MOK.der -nodes -days 36500 -subj "/CN=Custom MOK/"`
`sudo mokutil --import MOK.der`
`reboot`
blue screen>eroll mok>yes>keep the password you entered in sudo mokutil --import MOK.der>reboot
`cd rtw89`
`make`
`/usr/src/linux-headers-5.13.0-21-generic/scripts/sign-file sha256 /home/<USERNAME>/mok/MOK.priv /home/<USERNAME>/mok/MOK.der rtw89core.ko`
`/usr/src/linux-headers-5.13.0-21-generic/scripts/sign-file sha256 /home/<USERNAME>/mok/MOK.priv /home/<USERNAME>/mok/MOK.der rtw89pci.ko`
`sudo make install`
`reboot`

XOMUS, 2021-11-13:
use v6 or v7 breanch

petereth, 2021-12-06:
> `sudo apt-get update`
> `sudo apt-get install make gcc linux-headers-$(uname -r) build-essential git`
> `git clone https://github.com/lwfinger/rtw89.git`
> (or v6)
> `mkdir ~/mok && cd ~/mok`
> `openssl req -new -x509 -newkey rsa:2048 -keyout MOK.priv -outform DER -out MOK.der -nodes -days 36500 -subj "/CN=Custom MOK/"`
> `sudo mokutil --import MOK.der`
> `reboot`
> blue screen>enroll mok>yes>keep the password you entered in sudo mokutil --import MOK.der>reboot
> `cd rtw89`
> `make`
> `/usr/src/linux-headers-$(uname -r)/scripts/sign-file sha256 /home/$(echo $USER)/mok/MOK.priv /home/$(echo $USER)/mok/MOK.der rtw89core.ko`
> `/usr/src/linux-headers-$(uname -r)/scripts/sign-file sha256 /home/$(echo $USER)/mok/MOK.priv /home/$(echo $USER)/mok/MOK.der rtw89pci.ko`
> `sudo make install`
> `reboot`

I make this but after make and this two lines
/usr/src/linux-headers-$(uname -r)/scripts/sign-file sha256 /home/$(echo $USER)/mok/MOK.priv /home/$(echo $USER)/mok/MOK.der rtw89core.ko
/usr/src/linux-headers-$(uname -r)/scripts/sign-file sha256 /home/$(echo $USER)/mok/MOK.priv /home/$(echo $USER)/mok/MOK.der rtw89pci.ko

this happend
root@marlies-Legion-5-15ACH6H:/home/marlies# cd rtw89/
root@marlies-Legion-5-15ACH6H:/home/marlies/rtw89# make
make -C /lib/modules/5.13.0-22-generic/build M=/home/marlies/rtw89 modules
make[1]: Verzeichnis „/usr/src/linux-headers-5.13.0-22-generic“ wird betreten
CC [M] /home/marlies/rtw89/core.o
CC [M] /home/marlies/rtw89/debug.o
CC [M] /home/marlies/rtw89/mac80211.o
CC [M] /home/marlies/rtw89/mac.o
CC [M] /home/marlies/rtw89/phy.o
CC [M] /home/marlies/rtw89/fw.o
CC [M] /home/marlies/rtw89/rtw8852a.o
CC [M] /home/marlies/rtw89/rtw8852a_table.o
CC [M] /home/marlies/rtw89/rtw8852a_rfk.o
CC [M] /home/marlies/rtw89/rtw8852a_rfk_table.o
CC [M] /home/marlies/rtw89/cam.o
CC [M] /home/marlies/rtw89/efuse.o
CC [M] /home/marlies/rtw89/regd.o
CC [M] /home/marlies/rtw89/coex.o
CC [M] /home/marlies/rtw89/ps.o
CC [M] /home/marlies/rtw89/sar.o
CC [M] /home/marlies/rtw89/ser.o
LD [M] /home/marlies/rtw89/rtw89core.o
CC [M] /home/marlies/rtw89/pci.o
LD [M] /home/marlies/rtw89/rtw89pci.o
MODPOST /home/marlies/rtw89/Module.symvers
CC [M] /home/marlies/rtw89/rtw89core.mod.o
LD [M] /home/marlies/rtw89/rtw89core.ko
BTF [M] /home/marlies/rtw89/rtw89core.ko
Skipping BTF generation for /home/marlies/rtw89/rtw89core.ko due to unavailability of vmlinux
CC [M] /home/marlies/rtw89/rtw89pci.mod.o
LD [M] /home/marlies/rtw89/rtw89pci.ko
BTF [M] /home/marlies/rtw89/rtw89pci.ko
Skipping BTF generation for /home/marlies/rtw89/rtw89pci.ko due to unavailability of vmlinux
make[1]: Verzeichnis „/usr/src/linux-headers-5.13.0-22-generic“ wird verlassen
root@marlies-Legion-5-15ACH6H:/home/marlies/rtw89# /usr/src/linux-headers-$(uname -r)/scripts/sign-file sha256 /home/$(echo $USER)/mok/MOK.priv /home/$(echo $USER)/mok/MOK.der rtw89core.ko
At main.c:160:

SSL error:02001002:system library:fopen:No such file or directory: ../crypto/bio/bss_file.c:69
SSL error:2006D080:BIO routines:BIO_new_file:no such file: ../crypto/bio/bss_file.c:76
sign-file: /home/root/mok/MOK.priv: No such file or directory
root@marlies-Legion-5-15ACH6H:/home/marlies/rtw89# /usr/src/linux-headers-$(uname -r)/scripts/sign-file sha256 /home/$(echo $USER)/mok/MOK.priv /home/$(echo $USER)/mok/MOK.der rtw89pci.ko
At main.c:160:
SSL error:02001002:system library:fopen:No such file or directory: ../crypto/bio/bss_file.c:69
SSL error:2006D080:BIO routines:BIO_new_file:no such file: ../crypto/bio/bss_file.c:76
sign-file: /home/root/mok/MOK.priv: No such file or directory
root@marlies-Legion-5-15ACH6H:/home/marlies/rtw89#

XOMUS, 2021-12-06:
Insert commands one at a time

XOMUS, 2021-12-06:
Change $(uname -r) to your username

petereth, 2021-12-06:
> Change $(uname -r) to your username

when try, say command not found

root@marlies-xxx:/home/marlies/rtw89# /usr/src/linux-headers-$(marlies -r)/scripts/sign-file sha256 /home/$(echo §marlies)/mok/MOK.priv /home/$(echo $USER)/mok/MOK.der rtw89core.ko
marlies: command not found.
bash: /usr/src/linux-headers-/scripts/sign-file: File or directory not found
root@marlies-xxx:/home/marlies/rtw89# 

I changed to marlies

petereth, 2021-12-06:
> Write /your username/ in command
thank you for help

ok to you mean here as the username for: uname
sudo apt-get install make gcc linux-headers-$(uname -r) build-essential git

or here $USER for username
/usr/src/linux-headers-$(uname -r)/scripts/sign-file sha256 /home/$(echo $USER)/mok/MOK.priv /home/$(echo $USER)/mok/MOK.der rtw89core.ko

sorry for asking

XOMUS, 2021-12-06:
From you
 Open
how to use sign make install
#74
XOMUS opened this issue 24 days ago · 5 comments
Comments
 
@XOMUS XOMUS commented 24 days ago • 
edited 
'sudo apt-get update'
'sudo apt-get install make gcc linux-headers-$(uname -r) build-essential git'
'git clone https://github.com/lwfinger/rtw89.git'
(or v6)
'mkdir ~/mok && cd ~/mok' 
'openssl req -new -x509 -newkey rsa:2048 -keyout MOK.priv -outform DER -out MOK.der -nodes -days 36500 -subj "/CN=Custom MOK/"'
'sudo mokutil --import MOK.der'
'reboot'
blue screen>enroll mok>yes>keep the password you entered in sudo mokutil --import MOK.der>reboot
'cd rtw89'
'make'
'/usr/src/linux-headers-$(uname -r)/scripts/sign-file sha256 /home/username/mok/MOK.priv /home/usename/mok/MOK.der rtw89core.ko' 
'/usr/src/linux-headers-$(uname -r)/scripts/sign-file sha256 /home/username/mok/MOK.priv /home/username/mok/MOK.der rtw89pci.ko'
'sudo make install'
'reboot'

XOMUS, 2021-12-06:
Working?

petereth, 2021-12-06:
thank you I am not on the ubuntu laptop but as soon I am there I reply

thank you for help

> Working?

XOMUS, 2021-12-06:
Change (username) to your name

XOMUS, 2021-12-06:
Change $(echo $USER) to your username

XOMUS, 2021-12-06:
Don't touch $(UNAME-r)

XOMUS, 2021-12-06:
You use root seasion

XOMUS, 2021-12-06:
Maybe I'm dumb but I'm just new to Linux

petereth, 2021-12-06:
> Don't touch $(UNAME-r)

ok

petereth, 2021-12-06:
> You use root seasion

yes

XOMUS, 2021-12-06:
Don't yuse root seasion

XOMUS, 2021-12-06:
Use

XOMUS, 2021-12-06:
Use only sudo And it will be without all this that I wrote to you to work

petereth, 2021-12-06:
> Use only sudo And it will be without all this that I wrote to you to work

ok I should use not the root session and for every command I should use sudo
like
sudo git clone https://github.com/lwfinger/rtw89.git

XOMUS, 2021-12-06:
Hmmmmmmmmmmmm

XOMUS, 2021-12-06:
⏰⏰⏰⏰⏰⏰⏰⏰⏰⏰⏰⏰⏰⏰⏰⏰⏰⏰⏰⏰⏰⏰⏰⏰⏰⏰⏰⏰⏰⏰⏰⏰👍

XOMUS, 2021-12-06:
Working?

petereth, 2021-12-06:
> Working?
sorry back on the laptop in 1 or 2 hour 
thx

XOMUS, 2021-12-06:
👍or👎

XOMUS, 2021-12-06:
Okay

lwfinger, 2021-12-06:
Yoou should use sudo ONLY for the 'sudo make install' or 'sudo make sign-install'. EVERYTHING else should be done as a regular user. To clean up your mess, change directory to the one containing the rtw89 directory and 'sudo rm -rf rtw89'. That will remove all those files owned by root. Then do the following:

git clone https://github.com/lwfinger/rtw89.git
cd rtw89
make
sudo make sign-install

petereth, 2021-12-06:
after installing like this:
sudo make sign-install
and
mkdir ~/mok && cd ~/mok

here he said: mok exists 

 this never happen:
blue screen>enroll mok>yes>keep the password you entered in sudo mokutil --import MOK.der>reboot

lwfinger, 2021-12-06:
That was a bug in the Makefile. do a 'git pull' and 'sudo sign-install'.

petereth, 2021-12-06:
> That was a bug in the Makefile. do a 'git pull' and 'sudo sign-install'.

done

marlies@marlies-Legion-xxx:~/rtw89$ sudo make sign-install
make -C /lib/modules/5.13.0-22-generic/build M=/home/marlies/rtw89 modules
make[1]: Verzeichnis „/usr/src/linux-headers-5.13.0-22-generic“ wird betreten
make[1]: Verzeichnis „/usr/src/linux-headers-5.13.0-22-generic“ wird verlassen
mkdir -p ~/mok && pushd ~/mok
/bin/sh: 1: pushd: not found
make: *** [Makefile:95: sign] Fehler 127

lwfinger, 2021-12-06:
You need to install the package that implements pushd and popd.

petereth, 2021-12-06:
> You need to install the package that implements pushd and popd.

thank you - how to install?

lwfinger, 2021-12-06:
That is a matter for your distro. What one do you use?

petereth, 2021-12-06:
> That is a matter for your distro. What one do you use?

I use Ubuntu 22.10 latest

lwfinger, 2021-12-06:
It looks as if 'apt-cache search pushd popd' should find the packages to install with apt-get install.

flrnd, 2021-12-07:
Fedora 35
```
make: popd: No such file or directory
make: *** [Makefile:98: sign] Error 127
```
popd is already on the system. using sudo bla bla bla.

petereth, 2021-12-07:
> 

it looks like it is installed, so what should I do next to try?

thx

petereth, 2021-12-07:
should I try with this again?

'sudo apt-get update'
'sudo apt-get install make gcc linux-headers-$(uname -r) build-essential git'
'git clone https://github.com/lwfinger/rtw89.git'
(or v6)
'mkdir ~/mok && cd ~/mok'
'openssl req -new -x509 -newkey rsa:2048 -keyout MOK.priv -outform DER -out MOK.der -nodes -days 36500 -subj "/CN=Custom MOK/"'
'sudo mokutil --import MOK.der'
'reboot'

This is not working on the Lenovo LENOVO Legion 5, R5-5600H:

blue screen>enroll mok>yes>keep the password you entered in sudo mokutil --import MOK.der>reboot

cd rtw89
make
sudo make sign-install

'/usr/src/linux-headers-$(uname -r)/scripts/sign-file sha256 /home/username/mok/MOK.priv /home/usename/mok/MOK.der rtw89core.ko'
'/usr/src/linux-headers-$(uname -r)/scripts/sign-file sha256 /home/username/mok/MOK.priv /home/username/mok/MOK.der rtw89pci.ko'
'sudo make install'
'reboot'

lwfinger, 2021-12-07:
The Makefile was using /bin/bash as its shell; however, standard sh does not have popd built in. On my system, it seems to be included, thus it worked here.

I changed the make to use bash. Do a pull and try again.

petereth, 2021-12-07:
thx
> The Makefile was using /bin/bash as its shell; however, standard sh does not have popd built in. On my system, it seems to be included, thus it worked here.
> 
> I changed the make to use bash. Do a pull and try again.

when I make 
sudo make sign-install

I get

marlies@marlies-Legion-5-15ACH6H:~/rtw89$ sudo make sign-install
make -C /lib/modules/5.13.0-22-generic/build M=/home/marlies/rtw89 modules
make[1]: Verzeichnis „/usr/src/linux-headers-5.13.0-22-generic“ wird betreten
make[1]: Verzeichnis „/usr/src/linux-headers-5.13.0-22-generic“ wird verlassen
mkdir -p ~/mok && pushd ~/mok
~/mok /home/marlies/rtw89
openssl req -new -x509 -newkey rsa:2048 -keyout MOK.priv -outform DER -out MOK.der -nodes -days 36500 -subj "/CN=Custom MOK/"
Generating a RSA private key
...........+++++
.................+++++
writing new private key to 'MOK.priv'
-----
mokutil --import MOK.der
input password: 
input password again: 
popd
/bin/bash: Zeile 1: popd: Der Verzeichnisstapel ist leer.
make: *** [Makefile:98: sign] Fehler 1
marlies@marlies-Legion-5-15ACH6H:~/rtw89$

flrnd, 2021-12-07:
@petereth Sign it yourself:
```
sudo ./sign-file sha256 MOK.priv MOK.der rtw89core.ko
sudo ./sign-file sha256 MOK.priv MOK.der rtw89pci.ko
sudo make install

And reboot your system.
```
You should have the MOK.priv / der files inside rtw89 directory already.
Reboot, when prompt on your bios enrolle the key with the password you used on input.
Profit.

petereth, 2021-12-07:
> @petereth Sign it yourself:
>
> ```
> sudo ./sign-file sha256 MOK.priv MOK.der rtw89core.ko
> sudo ./sign-file sha256 MOK.priv MOK.der rtw89pci.ko
> sudo make install
>
> And reboot your system.
> ```
>
> You should [...]

```
marlies@marlies-Legion-5-15ACH6H:~/rtw89$ sudo ./sign-file sha256 MOK.priv MOK.der rtw89core.ko
[sudo] Passwort für marlies: 
marlies@marlies-Legion-5-15ACH6H:~/rtw89$ sudo ./sign-file sha256 MOK.priv MOK.der rtw89pci.ko
marlies@marlies-Legion-5-15ACH6H:~/rtw89$ sudo make install
make -C /lib/modules/5.13.0-22-generic/build M=/home/marlies/rtw89 modules
make[1]: Verzeichnis „/usr/src/linux-headers-5.13.0-22-generic“ wird betreten
make[1]: Verzeichnis „/usr/src/linux-headers-5.13.0-22-generic“ wird verlassen
Install rtw89 SUCCESS
marlies@marlies-Legion-5-15ACH6H:~/rtw89$ 
```

done

lwfinger, 2021-12-08:
@petereth: I have no idea why the pushd failes. That is why popd reports Der Verzeichnisstapel ist leer.

As far as I can tell, your distro is truly borked.

petereth, 2021-12-08:
sorry

> @petereth: I have no idea why the pushd failed. That is why popd reports Der Verzeichnisstapel ist leer.
> 
> As far as I can tell, your distro is truly borked.

I try find the wlan adapter but not here, could I to the steps in a different way because for the Ubuntu 21.10 ? thx

[image]

petereth, 2021-12-08:
i liste the command that are working and the others that are not working

'sudo apt-get update' - working
'sudo apt-get install make gcc linux-headers-$(uname -r) build-essential git' - working
'git clone https://github.com/lwfinger/rtw89.git' - working
'mkdir ~/mok && cd ~/mok' - working
'openssl req -new -x509 -newkey rsa:2048 -keyout MOK.priv -outform DER -out MOK.der -nodes -days 36500 -subj "/CN=Custom MOK/"' - working
'sudo mokutil --import MOK.der' - working
'reboot' - working 

not working: on the Lenovo LENOVO Legion 5, R5-5600H:
blue screen>enroll mok>yes>keep the password you entered in sudo mokutil --import MOK.der>reboot

'cd rtw89' - working
'make' - working
'sudo make sign-install' - not working correctly because of:
/bin/bash: Zeile 1: popd: Der Verzeichnisstapel ist leer.

hope we find a way

Thank you

flrnd, 2021-12-08:
@petereth `lsmod | grep rtw89`, if the output to that is empty, the module is not loaded. Then try `modprobe  rtw89pci`
Check dmesg output. 

If that doesn't bring the device, only thing I can say is that openSUSE folks added support out of the box to this card on Tumbleweed.

lwfinger, 2021-12-08:
I do not think @petereth could handle Tumbleweed; however, they also added it to openSUSE Leap 15.3. Download and install it with a Gnome desktop, and it will look and behave much like Ubuntu.

petereth, 2021-12-09:
> @petereth `lsmod | grep rtw89`, if the output to that is empty, the module is not loaded. Then try `modprobe rtw89pci` Check dmesg output.
> 
> If that doesn't bring the device, only thing I can say is that openSUSE folks added support out of the box to this card on Tumbleweed.

done

``` 
root@marlies-Legion-5-15ACH6H:/home/marlies# lsmod | grep rtw89
root@marlies-Legion-5-15ACH6H:/home/marlies# modprobe rtw89pci
modprobe: ERROR: could not insert 'rtw89pci': Operation not permitted
root@marlies-Legion-5-15ACH6H:/home/marlies# 
```

lwfinger, 2021-12-09:
What does dmesg say?

petereth, 2021-12-09:
> What does dmesg say?

sorry if I ask is the command only?
'dmesg'
or
'dmesg -wH'

thx

petereth, 2021-12-09:
> When I ask what dmesg has to say, I usually mean concerning the error. If I want the whole thing, I will ask for that.
> 
> The error is [ 2056.119919] Lockdown: modprobe: unsigned module loading is restricted; see man kernel_lockdown.7 [ 2077.625924] Lockdown: modprobe: unsigned module loading is restricted; see man kernel_lockdown.7
> 
> You are using secure boot and HAVE NOT signed the modules before installing them.

thank you for the info!
how to I sign it right?
thx

petereth, 2021-12-10:
> What do you think we have been trying to tell you!!

you tell me

sudo ./sign-file sha256 MOK.priv MOK.der rtw89core.ko
sudo ./sign-file sha256 MOK.priv MOK.der rtw89pci.ko
sudo make install

And reboot your system.
You should have the MOK.priv / der files inside rtw89 directory already.
Reboot, when prompt on your bios enrolle the key with the password you used on input.
Profit.

but I do not come into the bios enrolle

is there another way?

thx

petereth, 2021-12-10:
if I turn off: using secure boot is there a better way?
thx

closed 2021-12-10 by lwfinger

lwfinger, 2021-12-12:
The Makefile has been updated to fix the signing problems.

petereth, 2021-12-12:
> The Makefile has been updated to fix the signing problems.

thank you very much

now are these steps ok?

'sudo apt-get update'
'sudo apt-get install make gcc linux-headers-$(uname -r) build-essential git'
'git clone https://github.com/lwfinger/rtw89.git'
(or v6)
'mkdir ~/mok && cd ~/mok'
'openssl req -new -x509 -newkey rsa:2048 -keyout MOK.priv -outform DER -out MOK.der -nodes -days 36500 -subj "/CN=Custom MOK/"'
'sudo mokutil --import MOK.der'
'reboot'

lwfinger, 2021-12-12:
I do not know about the apt-get. That is distro specific; however the README.md has specific instructions for Ubuntu and Fedora users.

For the rest, you can clone the repo if you do not already have it. DO NOT USE v6 or v7. Branch main is the one to use!!!!!

After it is cloned, do 'make' followed by 'sudo make sign-install' if you want signed modules. For the unsigned case, use 'sudo make install'. Then reboot. If you are using signed modules, pay careful attention to the reboot. If you get the blue MOK screen, be sure to accept the key(s).

petereth, 2021-12-12:
> I do not know about the apt-get. That is distro specific; however the README.md has specific instructions for Ubuntu and Fedora users.
>
> For the rest, you can clone the repo if you do not already [...]

ok so for my Ubuntu 21.10 installation on the Lenovo Legion 5, R5-5600H

first I remove the old rtw89 folder : 
sudo rm -rf rtw89

after installing 

signed modules:
git clone https://github.com/lwfinger/rtw89.git
cd rtw89
make
sudo make sign-install

unsigned:
git clone https://github.com/lwfinger/rtw89.git
cd rtw89
make
sudo make install

and in signed version I have to find out why the keyboard from the Lenovo Legion 5, R5-5600H:
https://unix.stackexchange.com/questions/608217/keyboard-does-not-work-in-mokmanager-during-key-enrollment

hope it get it work

lwfinger, 2021-12-12:
YOU DO NOT HAVE TO DO BOTH SIGNED AND UNSIGNED MAKES. The signed version is ONLY if you are doing secure boot!!!!!!!!!!!!!!!!!!!!

petereth, 2021-12-12:
> YOU DO NOT HAVE TO DO BOTH SIGNED AND UNSIGNED MAKES. The signed version is ONLY if you are doing secure boot!!!!!!!!!!!!!!!!!!!!

yes I understand, I only put that here for remember if I bring the unsigned because of the keyboard not to work !

thank you for help

petereth, 2021-12-21:
thank you all for helping, without secure boot the installation works well!

really thank you
:)

[2 more replies]