lwfinger/rtw89 #75: [META] Status of upstreaming/backports

jmerdich, 2021-11-14:
Now that v7 of this patchset has [landed](https://git.kernel.org/pub/scm/linux/kernel/git/kvalo/wireless-drivers-next.git/commit/?id=e3ec7017f6a20d12ddd9fe23d345ebb7b8c104dd), this dkms setup will eventually no longer be necessary. Since this is the first place most people see, it seems prudent to tpoint to the right places here. So, here goes.

## Driver availability status:
 - [X] Nightly linux kernel
 - [ ] Mainline linux kernel (slated for 5.16)
 - [ ] Official distro support available for...
     - [X] Ubuntu 20.04 via "oem" kernel (`sudo apt install linux-oem-20.04d`)
     - [ ] Ubuntu 21.10+ via automatic kernel updates (available soon, see [ticket](https://bugs.launchpad.net/ubuntu/+source/linux/+bug/1945967) for details)
     - [ ] ...

Comments and direct updates from repo admins welcome, especially for status of backports in other distros.

gpdevsoft, 2021-11-22:
Hi, first of all I would like to thank @Iwfinger for his very helpful work, I am quite inexperienced with manual driver installation so I wanted to ask @jmerdich if to install OEM kernel in * buntu 20.04.3 distribution with sudo apt install linux -oem-20.04d Will I have Realtek 8852AE wireless network card drivers available?
if I understand correctly the OEM kernel for Ubuntu 20.04 is currently at version 5.10 but in my distribution (recently installed) I already have the kernel 5.11 and the drivers for that wireless network card are not there

Thank you for answer!

lwfinger, 2021-11-22:
You need kernel 5.16 to be sure to have.the driver. Some distros will either backport the driver to older kernels, or provide a package matching this repo, but not all will do that. I have no idea what Ubuntu is doing - I do not use it.

gpdevsoft, 2021-11-22:
ok, thank you for clarification, today i'have compile and install your latest version (V7) on xubuntu 20.04.3 LTS (now the latest stable kernel is 5.11.0-40), all seem to be fine, even with the version V5 i have never had disconnect or speed issues, about the signal strength I'm using wavemon (it's in the official repo of ubuntu) to test it and everything seems ok, using the 5GHz frequency the signal quality is reduced but it is a limit of this technology not of the driver or of the network card

jmerdich, 2021-11-22:
> if to install OEM kernel in * buntu 20.04.3 distribution with sudo apt install linux -oem-20.04d Will I have Realtek 8852AE wireless network card drivers available?

Yes. This was backported as far back as 5.10 on this special OEM variant of the kernel that Ubuntu supports. That said, this is *not* the place to report any issues you encounter there nor is the usual kernel mailing lists, you should use Ubuntu's tracker, probably on [this ticket](https://bugs.launchpad.net/ubuntu/+source/linux/+bug/1945967). Just tracking its existence here.

Linux 5.16 now has rc's available to try as well.

capfredf, 2021-11-28:
I think the latest Opensuse Tumbleweed comes with this driver too, though the kernel version is 5.15.2

lwfinger, 2021-11-28:
Yes, openSUSE has backported the driver to the TW kernel, and have made their equivalent of a PPA for Leap 15.2 and Leap 15.3. Both versions of Leap run a modified 5.3 kernel.

closed 2021-11-28 by lwfinger