lwfinger/rtw89 #76: Wifi stops working after automatic kernel update to 5.13.0-21-generic

naimenz, 2021-11-16:
Hi,
Thanks for the driver, I just got a Lenovo Legion S7 and I would've been stuck without it. Yesterday when I booted into Ubuntu, the wifi wasn't working. I tried running `sudo modprobe rtw89pci` again but I was getting an 'invalid argument' error. 

I found that my system had automatically upgraded from 5.13.0-19-generic to 5.13.0-21-generic. After a few missteps, I booted into the old kernel and the wifi worked again, and I could add and remove the driver with modprobe.

I don't know enough to understand what's going on, but if other people are experiencing the same problem, maybe try loading the old kernel from the "advanced options" in grub.

lwfinger, 2021-11-17:
AFTER you boot into the new kernel, then you need to run 'make' and 'sudo make install'. The driver must be built using exactly the same headers as used to build the kernel. There is no way around that. Investigate dkms if you want the build to be done automatically.

Gusyatnikova, 2021-11-17:
Hello! It seems a have the same problem. When i try run the make command i get Error 2. Can you? please? help me with it?

nata@pop-os:~/wifi deals/rtw89(v5)$ make
make -C /lib/modules/5.13.0-7620-generic/build M=/home/nata/wifi deals/rtw89 modules
make[1]: Entering directory '/usr/src/linux-headers-5.13.0-7620-generic'
make[1]: *** No rule to make target 'deals/rtw89'.  Stop.
make[1]: Leaving directory '/usr/src/linux-headers-5.13.0-7620-generic'
make: *** [Makefile:52: all] Error 2

naimenz, 2021-11-17:
Here's what happens when I boot into the new kernel and do that.
```
ian@ian-Legion:~$ cd rtw89
ian@ian-Legion:~/rtw89$ make
make -C /lib/modules/5.13.0-21-generic/build M=/home/ian/rtw89 modules
make[1]: Entering directory '/usr/src/linux-headers-5.13.0-21-generic'
make[1]: Leaving directory '/usr/src/linux-headers-5.13.0-21-generic'
ian@ian-Legion:~/rtw89$ sudo make install
make -C /lib/modules/5.13.0-21-generic/build M=/home/ian/rtw89 modules
make[1]: Entering directory '/usr/src/linux-headers-5.13.0-21-generic'
make[1]: Leaving directory '/usr/src/linux-headers-5.13.0-21-generic'
cp *.bin /lib/firmware/rtw89/.
Install rtw89 SUCCESS
ian@ian-Legion:~/rtw89$ sudo modprobe rtw89pci
modprobe: ERROR: could not insert 'rtw89pci': Invalid argument
ian@ian-Legion:~/rtw89$
```

naimenz, 2021-11-17:
Here are the two parts of dmesg that mention rtw89. The full thing is extremely long.
Do the two parts correspond to boot and then when I run `sudo modprobe rtw89pci`, do you think?
```
[Wed Nov 17 12:52:22 2021] pstore: Using crash dump compression: deflate
[Wed Nov 17 12:52:22 2021] pstore: Registered efi as persistent store backend
[Wed Nov 17 12:52:22 2021] cfg80211: Loading compiled-in X.509 certificates for regulatory database
[Wed Nov 17 12:52:22 2021] cfg80211: Loaded X.509 cert 'sforshee: 00b28ddf47aef9cea7'
[Wed Nov 17 12:52:22 2021] AVX2 version of gcm_enc/dec engaged.
[Wed Nov 17 12:52:22 2021] AES CTR mode by8 optimization enabled
[Wed Nov 17 12:52:22 2021] loop7: detected capacity change from 0 to 307976
[Wed Nov 17 12:52:22 2021] rtw89core: loading out-of-tree module taints kernel.
[Wed Nov 17 12:52:22 2021] ACPI Warning: \_SB.PCI0.GPP0.PEGP._DSM: Argument #4 type mismatch - Found [Buffer], ACPI requires [Package] (20210331/nsarguments-61)
[Wed Nov 17 12:52:22 2021] rtw89core: module verification failed: signature and/or required key missing - tainting kernel
[Wed Nov 17 12:52:22 2021] pci 0000:01:00.0: optimus capabilities: enabled, status dynamic power, hda bios codec supported
[Wed Nov 17 12:52:22 2021] VGA switcheroo: detected Optimus DSM method \_SB_.PCI0.GPP0.PEGP handle
[Wed Nov 17 12:52:22 2021] rtw89core: disagrees about version of symbol __netif_napi_del
[Wed Nov 17 12:52:22 2021] rtw89core: Unknown symbol __netif_napi_del (err -22)
[Wed Nov 17 12:52:22 2021] nouveau 0000:01:00.0: enabling device (0002 -> 0003)
[Wed Nov 17 12:52:22 2021] rtw89core: disagrees about version of symbol netif_napi_add
[Wed Nov 17 12:52:22 2021] rtw89core: Unknown symbol netif_napi_add (err -22)
[Wed Nov 17 12:52:22 2021] rtw89core: disagrees about version of symbol napi_enable
[Wed Nov 17 12:52:22 2021] rtw89core: Unknown symbol napi_enable (err -22)
[Wed Nov 17 12:52:22 2021] rtw89core: disagrees about version of symbol __dev_kfree_skb_any
[Wed Nov 17 12:52:22 2021] rtw89core: Unknown symbol __dev_kfree_skb_any (err -22)
[Wed Nov 17 12:52:22 2021] rtw89core: disagrees about version of symbol napi_disable
[Wed Nov 17 12:52:22 2021] rtw89core: Unknown symbol napi_disable (err -22)
[Wed Nov 17 12:52:22 2021] rtw89core: disagrees about version of symbol init_dummy_netdev
[Wed Nov 17 12:52:22 2021] rtw89core: Unknown symbol init_dummy_netdev (err -22)
[Wed Nov 17 12:52:22 2021] nouveau 0000:01:00.0: unknown chipset (b76000a1)
[Wed Nov 17 12:52:22 2021] EXT4-fs (nvme1n1p2): mounted filesystem with ordered data mode. Opts: (null). Quota mode: none.
[Wed Nov 17 12:52:22 2021] EXT4-fs (nvme1n1p4): mounted filesystem with ordered data mode. Opts: (null). Quota mode: none.
[Wed Nov 17 12:52:22 2021] Adding 8000508k swap on /dev/nvme1n1p3.  Priority:-2 extents:1 across:8000508k SSFS
[Wed Nov 17 12:52:22 2021] ideapad_acpi VPC2004:00: Keyboard backlight control not available
[Wed Nov 17 12:52:22 2021] hid-generic 0003:048D:C967.0002: input,hiddev0,hidraw1: USB HID v1.10 Keyboard [ITE Tech. Inc. ITE Device(8296)] on usb-0000:06:00.4-4/input0
[Wed Nov 17 12:52:22 2021] usbcore: registered new interface driver usbhid
[Wed Nov 17 12:52:22 2021] usbhid: USB HID core driver
[Wed Nov 17 12:52:22 2021] loop8: detected capacity change from 0 to 311936
[Wed Nov 17 12:52:22 2021] ideapad_acpi VPC2004:00: DYTC interface is not available
```
This next one comes right at the end
```
[Wed Nov 17 12:52:25 2021] rfkill: input handler disabled
[Wed Nov 17 12:52:29 2021] Bluetooth: RFCOMM TTY layer initialized
[Wed Nov 17 12:52:29 2021] Bluetooth: RFCOMM socket layer initialized
[Wed Nov 17 12:52:29 2021] Bluetooth: RFCOMM ver 1.11
[Wed Nov 17 12:52:29 2021] rfkill: input handler enabled
[Wed Nov 17 12:52:30 2021] rfkill: input handler disabled
[Wed Nov 17 12:53:03 2021] rtw89core: disagrees about version of symbol __netif_napi_del
[Wed Nov 17 12:53:03 2021] rtw89core: Unknown symbol __netif_napi_del (err -22)
[Wed Nov 17 12:53:03 2021] rtw89core: disagrees about version of symbol netif_napi_add
[Wed Nov 17 12:53:03 2021] rtw89core: Unknown symbol netif_napi_add (err -22)
[Wed Nov 17 12:53:03 2021] rtw89core: disagrees about version of symbol napi_enable
[Wed Nov 17 12:53:03 2021] rtw89core: Unknown symbol napi_enable (err -22)
[Wed Nov 17 12:53:03 2021] rtw89core: disagrees about version of symbol __dev_kfree_skb_any
[Wed Nov 17 12:53:03 2021] rtw89core: Unknown symbol __dev_kfree_skb_any (err -22)
[Wed Nov 17 12:53:03 2021] rtw89core: disagrees about version of symbol napi_disable
[Wed Nov 17 12:53:03 2021] rtw89core: Unknown symbol napi_disable (err -22)
[Wed Nov 17 12:53:03 2021] rtw89core: disagrees about version of symbol init_dummy_netdev
[Wed Nov 17 12:53:03 2021] rtw89core: Unknown symbol init_dummy_netdev (err -22)
```

lwfinger, 2021-11-18:
Yes, the first is from the boot and the second is from the modprobe.

The source is from /usr/src/linux-headers-5.13.0-21-generic"
What does 'uname -r' say? The two MUST be identical.

naimenz, 2021-11-18:
The two are the same. 
```
ian@ian-Legion:~$ uname -a
Linux ian-Legion 5.13.0-21-generic #21-Ubuntu SMP Tue Oct 19 08:59:28 UTC 2021 x86_64 x86_64 x86_64 GNU/Linux
```
Do you have any suspicions of what might be causing it? I can look into it more and try to rule out common problems.

lwfinger, 2021-11-18:
After you boot, do 'make clean', 'make', and 'sudo make install', followed by 'sudo modprobe -v rtw89pci'. Does that help?

naimenz, 2021-11-18:
Success! Thanks so much for your help. Did I just mess up the install somehow?

anatak, 2021-11-18:
I had similar problem after updating to 5.11.0-40-generic . `make clean` did help. Maybe it should be mention in readme after updating kernel for others "noobs" like me :)

lwfinger, 2021-11-18:
Yes, that should probably be mentioned; however, I have no idea what changes Ubuntu makes in the headers that fools make. For me, any change in the kernel automatically triggers a build of the objects that are affected, as well as the .ko files.

shveikus, 2021-11-26:
Faced with the same issue, resolved for me by placing bin files into `/usr/lib/firmware/rtw89/` folder.
After that Kernel update doesn't affect the driver visibility. 
Seems it will be useful to update Makefile.

lwfinger, 2021-11-26:
Use branch v7. That installs the firmware files.

Having the firmware files be present has nothing to do with driver availability following a kernel update. You either need to install DKMS or you need to build manually.

closed 2021-12-02 by lwfinger