lwfinger/rtw89 #77: The bluetooth is not fixed

aisclark91, 2021-11-17:
This solution works perfectly to solve the wifi issue but doesn't solve the Bluetooth driver absence.

lwfinger, 2021-11-17:
What BT device does lsusb show?

aisclark91, 2021-11-17:
`asclark@asclark:~$ lsusb
Bus 004 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 003 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 003: ID 0bda:385a Realtek Semiconductor Corp. Bluetooth Radio
Bus 001 Device 002: ID 30c9:0035 DLEQNA1TGFC5KC HP TrueVision HD Camera
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
`

lwfinger, 2021-11-17:
That device is not (yet) in the kernel tables. I will prepare a kernel patch and send it upstream; however, I will need the output from 'cat /sys/kernel/debug/usb/devices' that pertains to this device.

Can you patch and build your own kernel? If so, I will post a simple patch to get you on the air before the full patch propagates through the stable kernels.

aisclark91, 2021-11-17:
```
asclark@asclark:~$ sudo cat /sys/kernel/debug/usb/devices
[sudo] password for asclark:            

T:  Bus=01 Lev=00 Prnt=00 Port=00 Cnt=00 Dev#=  1 Spd=480  MxCh= 4
B:  Alloc=  0/800 us ( 0%), #Int=  0, #Iso=  0
D:  Ver= 2.00 Cls=09(hub  ) Sub=00 Prot=01 MxPS=64 #Cfgs=  1
P:  Vendor=1d6b ProdID=0002 Rev= 5.04
S:  Manufacturer=Linux 5.4.0-90-generic xhci-hcd
S:  Product=xHCI Host Controller
S:  SerialNumber=0000:03:00.3
C:* #Ifs= 1 Cfg#= 1 Atr=e0 MxPwr=  0mA
I:* If#= 0 Alt= 0 #EPs= 1 Cls=09(hub  ) Sub=00 Prot=00 Driver=hub
E:  Ad=81(I) Atr=03(Int.) MxPS=   4 Ivl=256ms

T:  Bus=01 Lev=01 Prnt=01 Port=02 Cnt=01 Dev#=  2 Spd=480  MxCh= 0
D:  Ver= 2.01 Cls=ef(misc ) Sub=02 Prot=01 MxPS=64 #Cfgs=  1
P:  Vendor=30c9 ProdID=0035 Rev= 0.03
S:  Manufacturer=DLEQNA1TGFC5KC
S:  Product=HP TrueVision HD Camera
S:  SerialNumber=0001
C:* #Ifs= 3 Cfg#= 1 Atr=80 MxPwr=500mA
A:  FirstIf#= 0 IfCount= 2 Cls=0e(video) Sub=03 Prot=00
A:  FirstIf#= 2 IfCount= 1 Cls=fe(app. ) Sub=01 Prot=00
I:* If#= 0 Alt= 0 #EPs= 1 Cls=0e(video) Sub=01 Prot=00 Driver=uvcvideo
E:  Ad=83(I) Atr=03(Int.) MxPS=  32 Ivl=4ms
I:* If#= 1 Alt= 0 #EPs= 0 Cls=0e(video) Sub=02 Prot=00 Driver=uvcvideo
I:  If#= 1 Alt= 1 #EPs= 1 Cls=0e(video) Sub=02 Prot=00 Driver=uvcvideo
E:  Ad=81(I) Atr=05(Isoc) MxPS= 128 Ivl=125us
I:  If#= 1 Alt= 2 #EPs= 1 Cls=0e(video) Sub=02 Prot=00 Driver=uvcvideo
E:  Ad=81(I) Atr=05(Isoc) MxPS= 512 Ivl=125us
I:  If#= 1 Alt= 3 #EPs= 1 Cls=0e(video) Sub=02 Prot=00 Driver=uvcvideo
E:  Ad=81(I) Atr=05(Isoc) MxPS=1024 Ivl=125us
I:  If#= 1 Alt= 4 #EPs= 1 Cls=0e(video) Sub=02 Prot=00 Driver=uvcvideo
E:  Ad=81(I) Atr=05(Isoc) MxPS=1536 Ivl=125us
I:  If#= 1 Alt= 5 #EPs= 1 Cls=0e(video) Sub=02 Prot=00 Driver=uvcvideo
E:  Ad=81(I) Atr=05(Isoc) MxPS=2048 Ivl=125us
I:  If#= 1 Alt= 6 #EPs= 1 Cls=0e(video) Sub=02 Prot=00 Driver=uvcvideo
E:  Ad=81(I) Atr=05(Isoc) MxPS=2688 Ivl=125us
I:  If#= 1 Alt= 7 #EPs= 1 Cls=0e(video) Sub=02 Prot=00 Driver=uvcvideo
E:  Ad=81(I) Atr=05(Isoc) MxPS=3060 Ivl=125us
I:* If#= 2 Alt= 0 #EPs= 0 Cls=fe(app. ) Sub=01 Prot=01 Driver=(none)

T:  Bus=01 Lev=01 Prnt=01 Port=03 Cnt=02 Dev#=  3 Spd=12   MxCh= 0
D:  Ver= 1.00 Cls=e0(wlcon) Sub=01 Prot=01 MxPS=64 #Cfgs=  1
P:  Vendor=0bda ProdID=385a Rev= 0.00
S:  Manufacturer=Realtek
S:  Product=Bluetooth Radio
S:  SerialNumber=00e04c000001
C:* #Ifs= 2 Cfg#= 1 Atr=e0 MxPwr=500mA
I:* If#= 0 Alt= 0 #EPs= 3 Cls=e0(wlcon) Sub=01 Prot=01 Driver=btusb
E:  Ad=81(I) Atr=03(Int.) MxPS=  16 Ivl=1ms
E:  Ad=02(O) Atr=02(Bulk) MxPS=  64 Ivl=0ms
E:  Ad=82(I) Atr=02(Bulk) MxPS=  64 Ivl=0ms
I:* If#= 1 Alt= 0 #EPs= 2 Cls=e0(wlcon) Sub=01 Prot=01 Driver=btusb
E:  Ad=03(O) Atr=01(Isoc) MxPS=   0 Ivl=1ms
E:  Ad=83(I) Atr=01(Isoc) MxPS=   0 Ivl=1ms
I:  If#= 1 Alt= 1 #EPs= 2 Cls=e0(wlcon) Sub=01 Prot=01 Driver=btusb
E:  Ad=03(O) Atr=01(Isoc) MxPS=   9 Ivl=1ms
E:  Ad=83(I) Atr=01(Isoc) MxPS=   9 Ivl=1ms
I:  If#= 1 Alt= 2 #EPs= 2 Cls=e0(wlcon) Sub=01 Prot=01 Driver=btusb
E:  Ad=03(O) Atr=01(Isoc) MxPS=  17 Ivl=1ms
E:  Ad=83(I) Atr=01(Isoc) MxPS=  17 Ivl=1ms
I:  If#= 1 Alt= 3 #EPs= 2 Cls=e0(wlcon) Sub=01 Prot=01 Driver=btusb
E:  Ad=03(O) Atr=01(Isoc) MxPS=  25 Ivl=1ms
E:  Ad=83(I) Atr=01(Isoc) MxPS=  25 Ivl=1ms
I:  If#= 1 Alt= 4 #EPs= 2 Cls=e0(wlcon) Sub=01 Prot=01 Driver=btusb
E:  Ad=03(O) Atr=01(Isoc) MxPS=  33 Ivl=1ms
E:  Ad=83(I) Atr=01(Isoc) MxPS=  33 Ivl=1ms
I:  If#= 1 Alt= 5 #EPs= 2 Cls=e0(wlcon) Sub=01 Prot=01 Driver=btusb
E:  Ad=03(O) Atr=01(Isoc) MxPS=  49 Ivl=1ms
E:  Ad=83(I) Atr=01(Isoc) MxPS=  49 Ivl=1ms

T:  Bus=02 Lev=00 Prnt=00 Port=00 Cnt=00 Dev#=  1 Spd=10000 MxCh= 2
B:  Alloc=  0/800 us ( 0%), #Int=  0, #Iso=  0
D:  Ver= 3.10 Cls=09(hub  ) Sub=00 Prot=03 MxPS= 9 #Cfgs=  1
P:  Vendor=1d6b ProdID=0003 Rev= 5.04
S:  Manufacturer=Linux 5.4.0-90-generic xhci-hcd
S:  Product=xHCI Host Controller
S:  SerialNumber=0000:03:00.3
C:* #Ifs= 1 Cfg#= 1 Atr=e0 MxPwr=  0mA
I:* If#= 0 Alt= 0 #EPs= 1 Cls=09(hub  ) Sub=00 Prot=00 Driver=hub
E:  Ad=81(I) Atr=03(Int.) MxPS=   4 Ivl=256ms

T:  Bus=03 Lev=00 Prnt=00 Port=00 Cnt=00 Dev#=  1 Spd=480  MxCh= 4
B:  Alloc=  0/800 us ( 0%), #Int=  0, #Iso=  0
D:  Ver= 2.00 Cls=09(hub  ) Sub=00 Prot=01 MxPS=64 #Cfgs=  1
P:  Vendor=1d6b ProdID=0002 Rev= 5.04
S:  Manufacturer=Linux 5.4.0-90-generic xhci-hcd
S:  Product=xHCI Host Controller
S:  SerialNumber=0000:03:00.4
C:* #Ifs= 1 Cfg#= 1 Atr=e0 MxPwr=  0mA
I:* If#= 0 Alt= 0 #EPs= 1 Cls=09(hub  ) Sub=00 Prot=00 Driver=hub
E:  Ad=81(I) Atr=03(Int.) MxPS=   4 Ivl=256ms

T:  Bus=04 Lev=00 Prnt=00 Port=00 Cnt=00 Dev#=  1 Spd=10000 MxCh= 2
B:  Alloc=  0/800 us ( 0%), #Int=  0, #Iso=  0
D:  Ver= 3.10 Cls=09(hub  ) Sub=00 Prot=03 MxPS= 9 #Cfgs=  1
P:  Vendor=1d6b ProdID=0003 Rev= 5.04
S:  Manufacturer=Linux 5.4.0-90-generic xhci-hcd
S:  Product=xHCI Host Controller
S:  SerialNumber=0000:03:00.4
C:* #Ifs= 1 Cfg#= 1 Atr=e0 MxPwr=  0mA
I:* If#= 0 Alt= 0 #EPs= 1 Cls=09(hub  ) Sub=00 Prot=00 Driver=hub
E:  Ad=81(I) Atr=03(Int.) MxPS=   4 Ivl=256ms
```
And yes I can patch the files, as long the makefile targets are not different.

lwfinger, 2021-11-17:
The patch is attached.
[patch_for_0bda_385a.txt](https://github.com/lwfinger/rtw89/files/7556955/patch_for_0bda_385a.txt)
:
Thanks for the other info. If you want credit in the kernel log for reporting the problem and testing the fix, then send your full name in an E-mail to larry.finger@lwfinger.net. If you would prefer to keep your anonymity, that part is optional.

aisclark91, 2021-11-17:
Thanks so much for your quick response! I have a question, when I installed the driver I used the v5 branch. I wonder if the change should be done over main or v5?

lwfinger, 2021-11-17:
The change is not to the wifi driver in this repo. That was why I asked about building your own kernel as the patch needs to be done to the kernel source. Bluetooth and wifi drivers are essentially separate other than they need to coexist on the 2.4 GHz radio.

On another point, you should be using the v7 branch of the wifi driver. It is the most up to date.

aisclark91, 2021-11-17:
Yes, that is why I erased my comment!, I am modifying it. Thanks

aisclark91, 2021-11-17:
ok I build my kernel again, and well it still don't work but is still recognized:

```
T:  Bus=01 Lev=00 Prnt=00 Port=00 Cnt=00 Dev#=  1 Spd=480  MxCh= 4
B:  Alloc=  0/800 us ( 0%), #Int=  0, #Iso=  0
D:  Ver= 2.00 Cls=09(hub  ) Sub=00 Prot=01 MxPS=64 #Cfgs=  1
P:  Vendor=1d6b ProdID=0002 Rev= 5.04
S:  Manufacturer=Linux 5.4.0-90-generic xhci-hcd
S:  Product=xHCI Host Controller
S:  SerialNumber=0000:03:00.3
C:* #Ifs= 1 Cfg#= 1 Atr=e0 MxPwr=  0mA
I:* If#= 0 Alt= 0 #EPs= 1 Cls=09(hub  ) Sub=00 Prot=00 Driver=hub
E:  Ad=81(I) Atr=03(Int.) MxPS=   4 Ivl=256ms

T:  Bus=01 Lev=01 Prnt=01 Port=02 Cnt=01 Dev#=  2 Spd=480  MxCh= 0
D:  Ver= 2.01 Cls=ef(misc ) Sub=02 Prot=01 MxPS=64 #Cfgs=  1
P:  Vendor=30c9 ProdID=0035 Rev= 0.03
S:  Manufacturer=DLEQNA1TGFC5KC
S:  Product=HP TrueVision HD Camera
S:  SerialNumber=0001
C:* #Ifs= 3 Cfg#= 1 Atr=80 MxPwr=500mA
A:  FirstIf#= 0 IfCount= 2 Cls=0e(video) Sub=03 Prot=00
A:  FirstIf#= 2 IfCount= 1 Cls=fe(app. ) Sub=01 Prot=00
I:* If#= 0 Alt= 0 #EPs= 1 Cls=0e(video) Sub=01 Prot=00 Driver=uvcvideo
E:  Ad=83(I) Atr=03(Int.) MxPS=  32 Ivl=4ms
I:* If#= 1 Alt= 0 #EPs= 0 Cls=0e(video) Sub=02 Prot=00 Driver=uvcvideo
I:  If#= 1 Alt= 1 #EPs= 1 Cls=0e(video) Sub=02 Prot=00 Driver=uvcvideo
E:  Ad=81(I) Atr=05(Isoc) MxPS= 128 Ivl=125us
I:  If#= 1 Alt= 2 #EPs= 1 Cls=0e(video) Sub=02 Prot=00 Driver=uvcvideo
E:  Ad=81(I) Atr=05(Isoc) MxPS= 512 Ivl=125us
I:  If#= 1 Alt= 3 #EPs= 1 Cls=0e(video) Sub=02 Prot=00 Driver=uvcvideo
E:  Ad=81(I) Atr=05(Isoc) MxPS=1024 Ivl=125us
I:  If#= 1 Alt= 4 #EPs= 1 Cls=0e(video) Sub=02 Prot=00 Driver=uvcvideo
E:  Ad=81(I) Atr=05(Isoc) MxPS=1536 Ivl=125us
I:  If#= 1 Alt= 5 #EPs= 1 Cls=0e(video) Sub=02 Prot=00 Driver=uvcvideo
E:  Ad=81(I) Atr=05(Isoc) MxPS=2048 Ivl=125us
I:  If#= 1 Alt= 6 #EPs= 1 Cls=0e(video) Sub=02 Prot=00 Driver=uvcvideo
E:  Ad=81(I) Atr=05(Isoc) MxPS=2688 Ivl=125us
I:  If#= 1 Alt= 7 #EPs= 1 Cls=0e(video) Sub=02 Prot=00 Driver=uvcvideo
E:  Ad=81(I) Atr=05(Isoc) MxPS=3060 Ivl=125us
I:* If#= 2 Alt= 0 #EPs= 0 Cls=fe(app. ) Sub=01 Prot=01 Driver=(none)

T:  Bus=01 Lev=01 Prnt=01 Port=03 Cnt=02 Dev#=  3 Spd=12   MxCh= 0
D:  Ver= 1.00 Cls=e0(wlcon) Sub=01 Prot=01 MxPS=64 #Cfgs=  1
P:  Vendor=0bda ProdID=385a Rev= 0.00
S:  Manufacturer=Realtek
S:  Product=Bluetooth Radio
S:  SerialNumber=00e04c000001
C:* #Ifs= 2 Cfg#= 1 Atr=e0 MxPwr=500mA
I:* If#= 0 Alt= 0 #EPs= 3 Cls=e0(wlcon) Sub=01 Prot=01 Driver=btusb
E:  Ad=81(I) Atr=03(Int.) MxPS=  16 Ivl=1ms
E:  Ad=02(O) Atr=02(Bulk) MxPS=  64 Ivl=0ms
E:  Ad=82(I) Atr=02(Bulk) MxPS=  64 Ivl=0ms
I:* If#= 1 Alt= 0 #EPs= 2 Cls=e0(wlcon) Sub=01 Prot=01 Driver=btusb
E:  Ad=03(O) Atr=01(Isoc) MxPS=   0 Ivl=1ms
E:  Ad=83(I) Atr=01(Isoc) MxPS=   0 Ivl=1ms
I:  If#= 1 Alt= 1 #EPs= 2 Cls=e0(wlcon) Sub=01 Prot=01 Driver=btusb
E:  Ad=03(O) Atr=01(Isoc) MxPS=   9 Ivl=1ms
E:  Ad=83(I) Atr=01(Isoc) MxPS=   9 Ivl=1ms
I:  If#= 1 Alt= 2 #EPs= 2 Cls=e0(wlcon) Sub=01 Prot=01 Driver=btusb
E:  Ad=03(O) Atr=01(Isoc) MxPS=  17 Ivl=1ms
E:  Ad=83(I) Atr=01(Isoc) MxPS=  17 Ivl=1ms
I:  If#= 1 Alt= 3 #EPs= 2 Cls=e0(wlcon) Sub=01 Prot=01 Driver=btusb
E:  Ad=03(O) Atr=01(Isoc) MxPS=  25 Ivl=1ms
E:  Ad=83(I) Atr=01(Isoc) MxPS=  25 Ivl=1ms
I:  If#= 1 Alt= 4 #EPs= 2 Cls=e0(wlcon) Sub=01 Prot=01 Driver=btusb
E:  Ad=03(O) Atr=01(Isoc) MxPS=  33 Ivl=1ms
E:  Ad=83(I) Atr=01(Isoc) MxPS=  33 Ivl=1ms
I:  If#= 1 Alt= 5 #EPs= 2 Cls=e0(wlcon) Sub=01 Prot=01 Driver=btusb
E:  Ad=03(O) Atr=01(Isoc) MxPS=  49 Ivl=1ms
E:  Ad=83(I) Atr=01(Isoc) MxPS=  49 Ivl=1ms

T:  Bus=02 Lev=00 Prnt=00 Port=00 Cnt=00 Dev#=  1 Spd=10000 MxCh= 2
B:  Alloc=  0/800 us ( 0%), #Int=  0, #Iso=  0
D:  Ver= 3.10 Cls=09(hub  ) Sub=00 Prot=03 MxPS= 9 #Cfgs=  1
P:  Vendor=1d6b ProdID=0003 Rev= 5.04
S:  Manufacturer=Linux 5.4.0-90-generic xhci-hcd
S:  Product=xHCI Host Controller
S:  SerialNumber=0000:03:00.3
C:* #Ifs= 1 Cfg#= 1 Atr=e0 MxPwr=  0mA
I:* If#= 0 Alt= 0 #EPs= 1 Cls=09(hub  ) Sub=00 Prot=00 Driver=hub
E:  Ad=81(I) Atr=03(Int.) MxPS=   4 Ivl=256ms

T:  Bus=03 Lev=00 Prnt=00 Port=00 Cnt=00 Dev#=  1 Spd=480  MxCh= 4
B:  Alloc=  0/800 us ( 0%), #Int=  0, #Iso=  0
D:  Ver= 2.00 Cls=09(hub  ) Sub=00 Prot=01 MxPS=64 #Cfgs=  1
P:  Vendor=1d6b ProdID=0002 Rev= 5.04
S:  Manufacturer=Linux 5.4.0-90-generic xhci-hcd
S:  Product=xHCI Host Controller
S:  SerialNumber=0000:03:00.4
C:* #Ifs= 1 Cfg#= 1 Atr=e0 MxPwr=  0mA
I:* If#= 0 Alt= 0 #EPs= 1 Cls=09(hub  ) Sub=00 Prot=00 Driver=hub
E:  Ad=81(I) Atr=03(Int.) MxPS=   4 Ivl=256ms

T:  Bus=04 Lev=00 Prnt=00 Port=00 Cnt=00 Dev#=  1 Spd=10000 MxCh= 2
B:  Alloc=  0/800 us ( 0%), #Int=  0, #Iso=  0
D:  Ver= 3.10 Cls=09(hub  ) Sub=00 Prot=03 MxPS= 9 #Cfgs=  1
P:  Vendor=1d6b ProdID=0003 Rev= 5.04
S:  Manufacturer=Linux 5.4.0-90-generic xhci-hcd
S:  Product=xHCI Host Controller
S:  SerialNumber=0000:03:00.4
C:* #Ifs= 1 Cfg#= 1 Atr=e0 MxPwr=  0mA
I:* If#= 0 Alt= 0 #EPs= 1 Cls=09(hub  ) Sub=00 Prot=00 Driver=hub
E:  Ad=81(I) Atr=03(Int.) MxPS=   4 Ivl=256ms
asclark@asclark:~$ 
```

lwfinger, 2021-11-18:
What does the dmesg log have to say about the Bluetooth?

In addition, post the output of 'lsmod | grep bt'.

aisclark91, 2021-11-18:
The dmesg log gives me this: 
```
[    0.000000] Linux version 5.4.0-90-generic (root@asclark) (gcc version 9.3.0 (Ubuntu 9.3.0-17ubuntu1~20.04)) #101+test SMP Thu Nov 18 17:09:34 EST 2021 (Ubuntu 5.4.0-90.101+test-generic 5.4.148)
[    0.000000] Command line: BOOT_IMAGE=/boot/vmlinuz-5.4.0-90-generic root=UUID=e421c8d9-60eb-49d8-8827-62606ff3ea9e ro quiet splash
[    0.000000] KERNEL supported cpus:
[    0.000000]   Intel GenuineIntel
[    0.000000]   AMD AuthenticAMD
[    0.000000]   Hygon HygonGenuine
[    0.000000]   Centaur CentaurHauls
[    0.000000]   zhaoxin   Shanghai  
[    0.000000] x86/fpu: Supporting XSAVE feature 0x001: 'x87 floating point registers'
[    0.000000] x86/fpu: Supporting XSAVE feature 0x002: 'SSE registers'
[    0.000000] x86/fpu: Supporting XSAVE feature 0x004: 'AVX registers'
[    0.000000] x86/fpu: xstate_offset[2]:  576, xstate_sizes[2]:  256
[    0.000000] x86/fpu: Enabled xstate features 0x7, context size is 832 bytes, using 'compacted' format.
[    0.000000] BIOS-provided physical RAM map:
[    0.000000] BIOS-e820: [mem 0x0000000000000000-0x000000000009ffff] usable
[    0.000000] BIOS-e820: [mem 0x00000000000a0000-0x00000000000fffff] reserved
[    0.000000] BIOS-e820: [mem 0x0000000000100000-0x0000000009eeffff] usable
[    0.000000] BIOS-e820: [mem 0x0000000009ef0000-0x000000000a000fff] reserved
[    0.000000] BIOS-e820: [mem 0x000000000a001000-0x000000000a1fffff] usable
[    0.000000] BIOS-e820: [mem 0x000000000a200000-0x000000000a20bfff] ACPI NVS
[    0.000000] BIOS-e820: [mem 0x000000000a20c000-0x00000000c5f52fff] usable
[    0.000000] BIOS-e820: [mem 0x00000000c5f53000-0x00000000c60a8fff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000c60a9000-0x00000000c611cfff] ACPI data
[    0.000000] BIOS-e820: [mem 0x00000000c611d000-0x00000000c76d1fff] ACPI NVS
[    0.000000] BIOS-e820: [mem 0x00000000c76d2000-0x00000000ccffdfff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000ccffe000-0x00000000cdffffff] usable
[    0.000000] BIOS-e820: [mem 0x00000000ce000000-0x00000000cfffffff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000f0000000-0x00000000f7ffffff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000fd000000-0x00000000fdffffff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000fe700000-0x00000000fe700fff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000feb80000-0x00000000fec01fff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000fec10000-0x00000000fec10fff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000fed00000-0x00000000fed00fff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000fed40000-0x00000000fed44fff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000fed80000-0x00000000fed8ffff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000fedc4000-0x00000000fedc9fff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000fedcc000-0x00000000fedcefff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000fedd5000-0x00000000fedd5fff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000ff000000-0x00000000ffffffff] reserved
[    0.000000] BIOS-e820: [mem 0x0000000100000000-0x000000020f33ffff] usable
[    0.000000] BIOS-e820: [mem 0x000000020f340000-0x000000022fffffff] reserved
[    0.000000] NX (Execute Disable) protection: active
[    0.000000] e820: update [mem 0xc2b1b018-0xc2b28857] usable ==> usable
[    0.000000] e820: update [mem 0xc2b1b018-0xc2b28857] usable ==> usable
[    0.000000] extended physical RAM map:
[    0.000000] reserve setup_data: [mem 0x0000000000000000-0x000000000009ffff] usable
[    0.000000] reserve setup_data: [mem 0x00000000000a0000-0x00000000000fffff] reserved
[    0.000000] reserve setup_data: [mem 0x0000000000100000-0x0000000009eeffff] usable
[    0.000000] reserve setup_data: [mem 0x0000000009ef0000-0x000000000a000fff] reserved
[    0.000000] reserve setup_data: [mem 0x000000000a001000-0x000000000a1fffff] usable
[    0.000000] reserve setup_data: [mem 0x000000000a200000-0x000000000a20bfff] ACPI NVS
[    0.000000] reserve setup_data: [mem 0x000000000a20c000-0x00000000c2b1b017] usable
[    0.000000] reserve setup_data: [mem 0x00000000c2b1b018-0x00000000c2b28857] usable
[    0.000000] reserve setup_data: [mem 0x00000000c2b28858-0x00000000c5f52fff] usable
[    0.000000] reserve setup_data: [mem 0x00000000c5f53000-0x00000000c60a8fff] reserved
[    0.000000] reserve setup_data: [mem 0x00000000c60a9000-0x00000000c611cfff] ACPI data
[    0.000000] reserve setup_data: [mem 0x00000000c611d000-0x00000000c76d1fff] ACPI NVS
[    0.000000] reserve setup_data: [mem 0x00000000c76d2000-0x00000000ccffdfff] reserved
[    0.000000] reserve setup_data: [mem 0x00000000ccffe000-0x00000000cdffffff] usable
[    0.000000] reserve setup_data: [mem 0x00000000ce000000-0x00000000cfffffff] reserved
[    0.000000] reserve setup_data: [mem 0x00000000f0000000-0x00000000f7ffffff] reserved
[    0.000000] reserve setup_data: [mem 0x00000000fd000000-0x00000000fdffffff] reserved
[    0.000000] reserve setup_data: [mem 0x00000000fe700000-0x00000000fe700fff] reserved
[    0.000000] reserve setup_data: [mem 0x00000000feb80000-0x00000000fec01fff] reserved
[    0.000000] reserve setup_data: [mem 0x00000000fec10000-0x00000000fec10fff] reserved
[    0.000000] reserve setup_data: [mem 0x00000000fed00000-0x00000000fed00fff] reserved
[    0.000000] reserve setup_data: [mem 0x00000000fed40000-0x00000000fed44fff] reserved
[    0.000000] reserve setup_data: [mem 0x00000000fed80000-0x00000000fed8ffff] reserved
[... 984 lines cut ...]
[    5.859589] Bluetooth: RFCOMM TTY layer initialized
[... cut ...]
[    9.202442] IPv6: ADDRCONF(NETDEV_CHANGE): wlo1: link becomes ready
```
lsmod | grep gives:
```
btusb                  57344  0
btrtl                  24576  1 btusb
btbcm                  16384  1 btusb
btintel                24576  1 btusb
bluetooth             552960  41 btrtl,btintel,btbcm,bnep,btusb,rfcomm
btrfs                1261568  0
xor                    24576  1 btrfs
zstd_compress         167936  1 btrfs
raid6_pq              114688  1 btrfs
libcrc32c              16384  1 btrfs
```

aisclark91, 2021-11-18:
Also I have tried to install the v7 driver branch and I got an error (In the old and new kernel):
```
asclark@asclark:~/Desktop/rtw89$ make
make -C /lib/modules/5.4.0-90-generic/build M=/home/asclark/Desktop/rtw89 modules
make[1]: Entering directory '/usr/src/linux-headers-5.4.0-90-generic'
  CC [M]  /home/asclark/Desktop/rtw89/core.o
In file included from /home/asclark/Desktop/rtw89/coex.h:8,
                 from /home/asclark/Desktop/rtw89/core.c:6:
/home/asclark/Desktop/rtw89/core.h:3428:5: warning: "CONFIG_SUSE_VERSION" is not defined, evaluates to 0 [-Wundef]
 3428 | #if CONFIG_SUSE_VERSION && CONFIG_SUSE_PATCHLEVEL && CONFIG_SUSE_VERSION == 15 && CONFIG_SUSE_PATCHLEVEL == 3
      |     ^~~~~~~~~~~~~~~~~~~
/home/asclark/Desktop/rtw89/core.h:3448:31: error: redefinition of ‘ieee80211_tx_dequeue_ni’
 3448 | static inline struct sk_buff *ieee80211_tx_dequeue_ni(struct ieee80211_hw *hw,
      |                               ^~~~~~~~~~~~~~~~~~~~~~~
In file included from /home/asclark/Desktop/rtw89/core.h:13,
                 from /home/asclark/Desktop/rtw89/coex.h:8,
                 from /home/asclark/Desktop/rtw89/core.c:6:
./include/net/mac80211.h:6276:31: note: previous definition of ‘ieee80211_tx_dequeue_ni’ was here
 6276 | static inline struct sk_buff *ieee80211_tx_dequeue_ni(struct ieee80211_hw *hw,
      |                               ^~~~~~~~~~~~~~~~~~~~~~~
make[2]: *** [scripts/Makefile.build:270: /home/asclark/Desktop/rtw89/core.o] Error 1
make[1]: *** [Makefile:1762: /home/asclark/Desktop/rtw89] Error 2
make[1]: Leaving directory '/usr/src/linux-headers-5.4.0-90-generic'
make: *** [Makefile:54: all] Error 2
```

lwfinger, 2021-11-19:
In your log, I see the line "[    4.297750] Bluetooth: hci0: RTL: assuming no firmware upload needed". That means that the necessaary firmware is not loaded. In directory /lib/firmware/rtl_bt, you should have files rtl8852au_config.bin and rtl8852au_fw.bin. 

I found and corrected an error in the Makefile - it was storing those firmware files in the wrong directory. Do a 'git pull' and a 'sudo make install' and it should work.

lwfinger, 2021-11-19:
The warning was my fault and a fix is pushed. I do not understand the error. For kernel 5.4, the erring code should be compiled, but to my knowledge that API was never modified. It was just deleted. It is possible that this was a side effect of the warning. Pull and try again.

aisclark91, 2021-11-19:
I had this problem to build in the v7 branch, but it builds in main and v5.
```
asclark@asclark:~/Desktop/rtw89$ make
make -C /lib/modules/5.4.0-90-generic/build M=/home/asclark/Desktop/rtw89 modules
make[1]: Entering directory '/usr/src/linux-headers-5.4.0-90-generic'
  CC [M]  /home/asclark/Desktop/rtw89/core.o
  CC [M]  /home/asclark/Desktop/rtw89/debug.o
  CC [M]  /home/asclark/Desktop/rtw89/mac80211.o
  CC [M]  /home/asclark/Desktop/rtw89/mac.o
/home/asclark/Desktop/rtw89/mac.c: In function ‘rtw89_mac_sub_pwr_seq’:
/home/asclark/Desktop/rtw89/mac.c:872:5: error: implicit declaration of function ‘fsleep’; did you mean ‘ssleep’? [-Werror=implicit-function-declaration]
  872 |     fsleep(cur_cfg->addr * 1000);
      |     ^~~~~~
      |     ssleep
cc1: some warnings being treated as errors
make[2]: *** [scripts/Makefile.build:270: /home/asclark/Desktop/rtw89/mac.o] Error 1
make[1]: *** [Makefile:1762: /home/asclark/Desktop/rtw89] Error 2
make[1]: Leaving directory '/usr/src/linux-headers-5.4.0-90-generic'
make: *** [Makefile:54: all] Error 2
```
Unfortunately the Bluetooth still doesn't work, but the firmware is now on:
```
[    0.000000] Linux version 5.4.0-90-generic (root@asclark) (gcc version 9.3.0 (Ubuntu 9.3.0-17ubuntu1~20.04)) #101+test SMP Thu Nov 18 17:09:34 EST 2021 (Ubuntu 5.4.0-90.101+test-generic 5.4.148)
[    0.000000] Command line: BOOT_IMAGE=/boot/vmlinuz-5.4.0-90-generic root=UUID=e421c8d9-60eb-49d8-8827-62606ff3ea9e ro quiet splash
[    0.000000] KERNEL supported cpus:
[    0.000000]   Intel GenuineIntel
[    0.000000]   AMD AuthenticAMD
[    0.000000]   Hygon HygonGenuine
[    0.000000]   Centaur CentaurHauls
[    0.000000]   zhaoxin   Shanghai  
[    0.000000] x86/fpu: Supporting XSAVE feature 0x001: 'x87 floating point registers'
[    0.000000] x86/fpu: Supporting XSAVE feature 0x002: 'SSE registers'
[    0.000000] x86/fpu: Supporting XSAVE feature 0x004: 'AVX registers'
[    0.000000] x86/fpu: xstate_offset[2]:  576, xstate_sizes[2]:  256
[    0.000000] x86/fpu: Enabled xstate features 0x7, context size is 832 bytes, using 'compacted' format.
[    0.000000] BIOS-provided physical RAM map:
[    0.000000] BIOS-e820: [mem 0x0000000000000000-0x000000000009ffff] usable
[    0.000000] BIOS-e820: [mem 0x00000000000a0000-0x00000000000fffff] reserved
[    0.000000] BIOS-e820: [mem 0x0000000000100000-0x0000000009eeffff] usable
[    0.000000] BIOS-e820: [mem 0x0000000009ef0000-0x000000000a000fff] reserved
[    0.000000] BIOS-e820: [mem 0x000000000a001000-0x000000000a1fffff] usable
[    0.000000] BIOS-e820: [mem 0x000000000a200000-0x000000000a20bfff] ACPI NVS
[    0.000000] BIOS-e820: [mem 0x000000000a20c000-0x00000000c5f52fff] usable
[    0.000000] BIOS-e820: [mem 0x00000000c5f53000-0x00000000c60a8fff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000c60a9000-0x00000000c611cfff] ACPI data
[    0.000000] BIOS-e820: [mem 0x00000000c611d000-0x00000000c76d1fff] ACPI NVS
[    0.000000] BIOS-e820: [mem 0x00000000c76d2000-0x00000000ccffdfff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000ccffe000-0x00000000cdffffff] usable
[    0.000000] BIOS-e820: [mem 0x00000000ce000000-0x00000000cfffffff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000f0000000-0x00000000f7ffffff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000fd000000-0x00000000fdffffff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000fe700000-0x00000000fe700fff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000feb80000-0x00000000fec01fff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000fec10000-0x00000000fec10fff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000fed00000-0x00000000fed00fff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000fed40000-0x00000000fed44fff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000fed80000-0x00000000fed8ffff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000fedc4000-0x00000000fedc9fff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000fedcc000-0x00000000fedcefff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000fedd5000-0x00000000fedd5fff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000ff000000-0x00000000ffffffff] reserved
[    0.000000] BIOS-e820: [mem 0x0000000100000000-0x000000020f33ffff] usable
[    0.000000] BIOS-e820: [mem 0x000000020f340000-0x000000022fffffff] reserved
[    0.000000] NX (Execute Disable) protection: active
[    0.000000] e820: update [mem 0xc2b1b018-0xc2b28857] usable ==> usable
[    0.000000] e820: update [mem 0xc2b1b018-0xc2b28857] usable ==> usable
[    0.000000] extended physical RAM map:
[    0.000000] reserve setup_data: [mem 0x0000000000000000-0x000000000009ffff] usable
[    0.000000] reserve setup_data: [mem 0x00000000000a0000-0x00000000000fffff] reserved
[    0.000000] reserve setup_data: [mem 0x0000000000100000-0x0000000009eeffff] usable
[... 1002 lines cut ...]
                Either enable ECC checking or force module loading by setting 'ecc_enable_override'.
                (Note that use of the override may cause unknown side effects.)
[    5.288455] EDAC amd64: Node 0: DRAM ECC disabled.
[    5.288459] EDAC amd64: ECC disabled in the BIOS or no ECC capability, module will not load.
                Either enable ECC checking or force module loading by setting 'ecc_enable_override'.
                (Note that use of the override may cause unknown side effects.)
[    5.352384] EDAC amd64: Node 0: DRAM ECC disabled.
[    5.352387] EDAC amd64: ECC disabled in the BIOS or no ECC capability, module will not load.
[... cut ...]
[    9.174978] wlo1: Limiting TX power to 30 (30 - 0) dBm as advertised by 48:5d:36:4d:5f:d6
[    9.196342] IPv6: ADDRCONF(NETDEV_CHANGE): wlo1: link becomes ready
[   10.863392] systemd-journald[397]: File /var/log/journal/7a0781c5129d4f848f901e53f906eaf1/user-1000.journal corrupted or uncleanly shut down, renaming and replacing.
[  455.926802] debugfs: File 'le_min_key_size' in directory 'hci0' already present!
[  455.926811] debugfs: File 'le_max_key_size' in directory 'hci0' already present!
```

lwfinger, 2021-11-19:
Power down and try again. Those BT modules behave differently once they have been accessed once since power up.

If that does not help, I will not be able to help any more. As I have stated many times. I do not do Bluetooth!

closed 2021-11-19 by lwfinger

shadiakiki1986, 2021-12-02:
fwiw, my `lsusb` gives:

```
# Bus 001 Device 002: ID 0bda:4852 Realtek Semiconductor Corp. Bluetooth Radio
```
I modified the patch file above to have `4852` instead of `385a` and am compiling the linux kernel. Will update here if this works for me or not when done