lwfinger/rtw89 #78: Error on installation: No such file or directory ../crypt/bio/bss_file.c:69

mattiasJohnson, 2021-11-19:
I'm trying to install this on my Lenovo Legion 7 running Ubuntu 20.04 and on running `sudo make sign-install` I am getting the following error:

```bash
mattias@mattias-Legion-S7-15ACH6:~/Downloads/programs-from-source/rtw89$ sudo make sign-install
make -C /lib/modules/5.11.0-40-generic/build M=/home/mattias/Downloads/programs-from-source/rtw89 modules
make[1]: Entering directory '/usr/src/linux-headers-5.11.0-40-generic'
make[1]: Leaving directory '/usr/src/linux-headers-5.11.0-40-generic'
kmodsign sha512 /var/lib/shim-signed/mok/MOK.priv /var/lib/shim-signed/mok/MOK.der rtw89core.ko
At main.c:248:
- SSL error:02001002:system library:fopen:No such file or directory: ../crypto/bio/bss_file.c:69
- SSL error:2006D080:BIO routines:BIO_new_file:no such file: ../crypto/bio/bss_file.c:76
kmodsign: /var/lib/shim-signed/mok/MOK.priv: No such file or directory
make: *** [Makefile:91: sign] Error 1
mattias@mattias-Legion-S7-15ACH6:~/Download
```

Is there something I can do? I am using branch v7 btw since I saw in another issue that that was the newest one.

XOMUS, 2021-11-19:
Use https://github.com/lwfinger/rtw89/issues/74

mattiasJohnson, 2021-11-19:
Thanks a lot, that solved it!

closed 2021-11-19 by mattiasJohnson