lwfinger/rtw89 #79: Skipping BTF generation due to unavailability of vmlinux

metju312, 2021-11-19:
`msadlo@msadlo-pc:~/Documents/rtw89$ make
make -C /lib/modules/5.13.0-21-generic/build M=/home/msadlo/Documents/rtw89 modules
make[1]: Entering directory '/usr/src/linux-headers-5.13.0-21-generic'
  CC [M]  /home/msadlo/Documents/rtw89/core.o
  CC [M]  /home/msadlo/Documents/rtw89/debug.o
  CC [M]  /home/msadlo/Documents/rtw89/mac80211.o
  CC [M]  /home/msadlo/Documents/rtw89/mac.o
  CC [M]  /home/msadlo/Documents/rtw89/phy.o
  CC [M]  /home/msadlo/Documents/rtw89/fw.o
  CC [M]  /home/msadlo/Documents/rtw89/rtw8852a.o
  CC [M]  /home/msadlo/Documents/rtw89/rtw8852a_table.o
  CC [M]  /home/msadlo/Documents/rtw89/rtw8852a_rfk.o
  CC [M]  /home/msadlo/Documents/rtw89/rtw8852a_rfk_table.o
  CC [M]  /home/msadlo/Documents/rtw89/cam.o
  CC [M]  /home/msadlo/Documents/rtw89/efuse.o
  CC [M]  /home/msadlo/Documents/rtw89/regd.o
  CC [M]  /home/msadlo/Documents/rtw89/coex.o
  CC [M]  /home/msadlo/Documents/rtw89/ps.o
  CC [M]  /home/msadlo/Documents/rtw89/sar.o
  CC [M]  /home/msadlo/Documents/rtw89/ser.o
  CC [M]  /home/msadlo/Documents/rtw89/util.o
  LD [M]  /home/msadlo/Documents/rtw89/rtw89core.o
  CC [M]  /home/msadlo/Documents/rtw89/pci.o
  LD [M]  /home/msadlo/Documents/rtw89/rtw89pci.o
  MODPOST /home/msadlo/Documents/rtw89/Module.symvers
  CC [M]  /home/msadlo/Documents/rtw89/rtw89core.mod.o
  LD [M]  /home/msadlo/Documents/rtw89/rtw89core.ko
  BTF [M] /home/msadlo/Documents/rtw89/rtw89core.ko
Skipping BTF generation for /home/msadlo/Documents/rtw89/rtw89core.ko due to unavailability of vmlinux
  CC [M]  /home/msadlo/Documents/rtw89/rtw89pci.mod.o
  LD [M]  /home/msadlo/Documents/rtw89/rtw89pci.ko
  BTF [M] /home/msadlo/Documents/rtw89/rtw89pci.ko
Skipping BTF generation for /home/msadlo/Documents/rtw89/rtw89pci.ko due to unavailability of vmlinux
make[1]: Leaving directory '/usr/src/linux-headers-5.13.0-21-generic'`

`msadlo@msadlo-pc:~/Documents/rtw89$ sudo make install
make -C /lib/modules/5.13.0-21-generic/build M=/home/msadlo/Documents/rtw89 modules
make[1]: Entering directory '/usr/src/linux-headers-5.13.0-21-generic'
make[1]: Leaving directory '/usr/src/linux-headers-5.13.0-21-generic'
cp: cannot stat 'rtl8852au*.bin': No such file or directory
make: *** [Makefile:61: install] Error 1
`

XOMUS, 2021-11-19:
Use https://github.com/lwfinger/rtw89/issues/74

Chnossos, 2021-11-19:
I get the exact same error as OP on a fresh Ubuntu 20.04 installation.

```bash
$ uname -mr; lsb_release -d
5.11.0-40-generic x86_64
Description:	Ubuntu 20.04.3 LTS
```

Chnossos, 2021-11-19:
> To make sign install work, use #74

@metju312 Following the post @XOMUS linked worked for me!

metju312, 2021-11-19:
Kernel 5.16 on Ubuntu 21.10 have this driver!

https://sypalo.com/how-to-upgrade-ubuntu

Worked for me on Lenovo Legion 5.

closed 2021-11-19 by metju312