lwfinger/rtw89 #80: make: *** [Makefile:61: install] Error 1

SkylinesBuilder, 2021-11-19:
When I try to do `sudo make install`, I get the following error:

`make -C /lib/modules/5.13.0-21-generic/build M=/home/guy/rtw89 modules`
`make[1]: Entering directory '/usr/src/linux-headers-5.13.0-21-generic`
`make[1]: Leaving directory '/usr/src/linux-headers-5.13.0-21-generic`
`cp: cannot stat 'rtw8852au*.bin": No such file or directory`
`make: *** [Makefile:61: install] Error 1`

I have followed the installation instructions, and it does do the `make` command sucesfully, but get's hung up on this problem.
I am somewhat new to Linux, but hope to be able to solve this problem. Thanks in advance.

XOMUS, 2021-11-19:
To make sign install work, use https://github.com/lwfinger/rtw89/issues/74

XOMUS, 2021-11-19:
Try this https://github.com/lwfinger/rtw89/issues/74

SkylinesBuilder, 2021-11-19:
Thank you very much, that did indeed fix the issue.

closed 2021-11-19 by SkylinesBuilder