lwfinger/rtw89 #81: building in the v7 branch

aisclark91, 2021-11-20:
While building in the v7 branch I have the following error:

```
/home/asclark/Desktop/rtw89/phy.c:2886:28: error: ‘struct ieee80211_bss_conf’ has no member named ‘he_bss_color’; did you mean ‘bss_color’?
 2886 |  bss_color = vif->bss_conf.he_bss_color.color;
      |                            ^~~~~~~~~~~~
      |                            bss_color
make[2]: *** [scripts/Makefile.build:270: /home/asclark/Desktop/rtw89/phy.o] Error 1
make[1]: *** [Makefile:1762: /home/asclark/Desktop/rtw89] Error 2
make[1]: Leaving directory '/usr/src/linux-headers-5.4.0-90-generic'
make: *** [Makefile:54: all] Error 2
```

aisclark91, 2021-11-20:
Also I know the BT issue 77 is closed, but I wanted to tell you that I modified btrlt.c where the table_id was missing. I incorporated the missing part and I got the following:

```[    4.299789] Bluetooth: hci0: RTL: loading rtl_bt/rtl8852au_fw.bin
[    4.301309] Bluetooth: hci0: RTL: loading rtl_bt/rtl8852au_config.bin
[    4.301444] rtw89core: module verification failed: signature and/or required key missing - tainting kernel
[    4.301452] Bluetooth: hci0: RTL: assuming no firmware upload needed
[    4.308858] rtw89_pci 0000:01:00.0: enabling device (0000 -> 0003)
[    4.314791] rtw89_pci 0000:01:00.0: Firmware version 0.13.24.0, cmd version 0, type 1
[    4.314794] rtw89_pci 0000:01:00.0: Firmware version 0.13.24.0, cmd version 0, type 3
[    4.335807] rtw89_pci 0000:01:00.0: chip rfe_type is 1
[    4.342979] rtw89_pci 0000:01:00.0 wlo1: renamed from wlan0
```
It seems `rtw89core` asks for a signed kernel...

lwfinger, 2021-11-20:
NO, that is your system, not rtw89. If you are using secure boot, then you need to sign the modules, then read https://github.com/lwfinger/rtw89/issues/74.

aisclark91, 2021-11-20:
mmm... I am not using Secure Boot, I have disabled it when I installed Linux. And indeed, I modified btusb.c and btrtl.c.  Well, for now, I just want to build the v7 version to check an improvement but the issue above persists.

lwfinger, 2021-11-20:
There is no issue with wifi in the above. It has created wl01, your wifi device.

This driver does not handle the BT device. That is done by btusb and btrtl, but only btusb needs to be modified. Perhaps you should post your patch or patches.

aisclark91, 2021-11-20:
Ok, I solve my bt and wifi problems. The changes I have made to my kernel (5.40-90.101 focal/Linux Mint) are the following:

In btusb.c: Add the following table inside the struct:
```
@@ static const struct usb_device_id blacklist_table[] = {

 	/* Realtek 8852AE Bluetooth devices */
 	{ USB_DEVICE(0x0bda, 0xc852), .driver_info = BTUSB_REALTEK },
	{ USB_DEVICE(0x0bda, 0x385a), .driver_info = BTUSB_REALTEK },
        { USB_DEVICE(0x0bda, 0x4852), .driver_info = BTUSB_REALTEK },
 	{ USB_DEVICE(0x04c5, 0x165c), .driver_info = BTUSB_REALTEK },
```

In btrtl.c: Make all the changes posted here,
[https://patchwork.ozlabs.org/project/ubuntu-kernel/patch/20210415035559.25743-3-aaron.ma@canonical.com/](url)

aisclark91, 2021-11-20:
Thanks so much for your help!

lwfinger, 2021-11-20:
I did not realize that your kernel was that old. The changes in that patchwork.ozlabs.org article are already in the mainline kernel.

In the blacklist struct, I have
        /* Realtek 8852AE Bluetooth devices */
        { USB_DEVICE(0x0bda, 0xc852), .driver_info = BTUSB_REALTEK |
                                                     BTUSB_WIDEBAND_SPEECH },
        { USB_DEVICE(0x0bda, 0x385a), .driver_info = BTUSB_REALTEK |
                                                     BTUSB_WIDEBAND_SPEECH },
        { USB_DEVICE(0x0bda, 0x4852), .driver_info = BTUSB_REALTEK |
                                                     BTUSB_WIDEBAND_SPEECH },
        { USB_DEVICE(0x04c5, 0x165c), .driver_info = BTUSB_REALTEK |
                                                     BTUSB_WIDEBAND_SPEECH },
        { USB_DEVICE(0x04ca, 0x4006), .driver_info = BTUSB_REALTEK |
                                                     BTUSB_WIDEBAND_SPEECH },

closed 2021-11-20 by lwfinger