lwfinger/rtw89 #83: Completely new on ubuntu

kanker2, 2021-11-26:
I tried to install this driver for my hp pavilon with ubuntu 20.04  LTS with a realtek 8852AE adapter, and when i try the commands for installation in all distros 
git clone https://github.com/lwfinger/rtw89.git -b v5
cd rtw89
make
sudo make install
i get the following after executing 'sudo make install':
make -C /lib/modules/5.11.0-27-generic/build M=/home/kanker/rtw89 modules
make[1]: Entering directory '/usr/src/linux-headers-5.11.0-27-generic'
make[1]: Leaving directory '/usr/src/linux-headers-5.11.0-27-generic'
cp: cannot stat 'rtl8852au*.bin': No such file or directory
make: *** [Makefile:61: install] Error 1
Can i get helped pls? i cant find nothing else about the drivers for this adapter. Thanks

hozblok, 2021-11-26:
@kanker2 
try this: https://github.com/lwfinger/rtw89/issues/74

lwfinger, 2021-11-26:
You should be running v7, not v5. Do the following from your rtw89 source directory:

git checkout origin/v7 -b v7
make
sudo make install

The v5 branch is obsolete and will soon be deleted, along with v6. Once I get the time, I plan to transfer the v7 code to main and delete v7 as well, but I will give plenty of warning.

danielheadbang, 2021-12-15:
Thanks, this worked perfectly on my side, was having the same issue, didn't actually check there was a newer branch

kanker2, 2022-02-08:
So much thanks now it works :)

closed 2022-02-08 by lwfinger