lwfinger/rtw89 #84: update problem and update kernel

ninuxi, 2021-11-30:
Hi
ubuntu 20.04 lts, kernel 5.11: work but after update and upgrade disappear.
update to kernel 5.15: disappear. thanks for the help!
```
cd ~/rtw89
make clean
git pull
```
make 
```
nto@nto:~/rtw89$ make
make -C /lib/modules/5.15.5-051505-generic/build M=/home/nto/rtw89 modules
make[1]: ingresso nella directory «/usr/src/linux-headers-5.15.5-051505-generic»
warning: the compiler differs from the one used to build the kernel
  The kernel was built by: gcc (Ubuntu 11.2.0-12ubuntu1) 11.2.0
  You are using:           gcc (Ubuntu 9.3.0-17ubuntu1~20.04) 9.3.0
  CC [M]  /home/nto/rtw89/core.o
In file included from /home/nto/rtw89/coex.h:8,
                 from /home/nto/rtw89/core.c:6:
/home/nto/rtw89/core.h: In function ‘rtw89_read32_mask’:
/home/nto/rtw89/core.h:2999:9: error: implicit declaration of function ‘field_get’; did you mean ‘of_led_get’? [-Werror=implicit-function-declaration]
 2999 |  return field_get(mask, rtw89_read32(rtwdev, addr));
      |         ^~~~~~~~~
      |         of_led_get
/home/nto/rtw89/core.h: In function ‘rtw89_write32_mask’:
/home/nto/rtw89/core.h:3023:25: error: implicit declaration of function ‘field_prep’ [-Werror=implicit-function-declaration]
 3023 |  set = (orig & ~mask) | field_prep(mask, data);
      |                         ^~~~~~~~~~
cc1: some warnings being treated as errors
make[2]: *** [scripts/Makefile.build:277: /home/nto/rtw89/core.o] Errore 1
make[1]: *** [Makefile:1874: /home/nto/rtw89] Errore 2
make[1]: uscita dalla directory «/usr/src/linux-headers-5.15.5-051505-generic»
make: *** [Makefile:54: all] Errore 2
```

Robyan02, 2021-11-30:
Same thing happened to me - Fedora 35 linux5.15

ninuxi, 2021-11-30:
resolved, bad resolved
whit new installation of ubuntu 21.10 (kernel 5.13), wifi work fine without your library

cordx, 2021-11-30:
i am getting the same make errors on mint 20.2 with kernel 5.4 not after an update. i was just trying to build for the first time. when i run a search over the repo for field_get and field_prep all other instances are all caps (FIELD_GET and [FIELD_PREP](url=https://github.com/lwfinger/rtw89/search?p=1&q=field_prep)).  i can see FIELD_GET and FIELD_PREP defined in /usr/src/linu-headers-$(uname -r)/include/linux/bitfield.h. i changed the instances in /rtw89/core.h to all caps and got past the first set of errors only to be met by a whole new batch that complained (among other things) about the all caps versions.

```
user@computer ~/D/rtw89 (main) [2]> make
make -C /lib/modules/5.4.0-90-generic/build M=/home/user/Downloads/rtw89 modules
make[1]: Entering directory '/usr/src/linux-headers-5.4.0-90-generic'
  CC [M]  /home/user/Downloads/rtw89/core.o
  CC [M]  /home/user/Downloads/rtw89/debug.o
In file included from ./include/asm-generic/bug.h:5,
                 from ./arch/x86/include/asm/bug.h:83,
                 from ./include/linux/bug.h:5,
                 from ./include/linux/average.h:5,
                 from /home/user/Downloads/rtw89/core.h:8,
                 from /home/user/Downloads/rtw89/coex.h:8,
                 from /home/user/Downloads/rtw89/debug.c:6:
In function ‘rtw89_write8_mask’,
    inlined from ‘rtw89_debug_mac_dbg_port_dump’ at /home/user/Downloads/rtw89/debug.c:2021:4,
    inlined from ‘rtw89_debug_priv_mac_dbg_port_dump_get’ at /home/user/Downloads/rtw89/debug.c:2072:9:
./include/linux/compiler.h:417:38: error: call to ‘__compiletime_assert_393’ declared with attribute error: FIELD_PREP: mask is not constant
  417 |  _compiletime_assert(condition, msg, __compiletime_assert_, __COUNTER__)
      |                                      ^
./include/linux/compiler.h:398:4: note: in definition of macro ‘__compiletime_assert’
  398 |    prefix ## suffix();    \
      |    ^~~~~~
./include/linux/compiler.h:417:2: note: in expansion of macro ‘_compiletime_assert’
  417 |  _compiletime_assert(condition, msg, __compiletime_assert_, __COUNTER__)
      |  ^~~~~~~~~~~~~~~~~~~
./include/linux/build_bug.h:39:37: note: in expansion of macro ‘compiletime_assert’
   39 | #define BUILD_BUG_ON_MSG(cond, msg) compiletime_assert(!(cond), msg)
      |                                     ^~~~~~~~~~~~~~~~~~
./include/linux/bitfield.h:46:3: note: in expansion of macro ‘BUILD_BUG_ON_MSG’
   46 |   BUILD_BUG_ON_MSG(!__builtin_constant_p(_mask),  \
      |   ^~~~~~~~~~~~~~~~
./include/linux/bitfield.h:81:3: note: in expansion of macro ‘__BF_FIELD_CHECK’
   81 |   __BF_FIELD_CHECK(_mask, 0ULL, _val, "FIELD_PREP: "); \
      |   ^~~~~~~~~~~~~~~~
/home/user/Downloads/rtw89/core.h:3047:25: note: in expansion of macro ‘FIELD_PREP’
 3047 |  set = (orig & ~mask) | FIELD_PREP(mask, data);
      |                         ^~~~~~~~~~
./include/linux/compiler.h:417:38: error: call to ‘__compiletime_assert_394’ declared with attribute error: FIELD_PREP: mask is zero
  417 |  _compiletime_assert(condition, msg, __compiletime_assert_, __COUNTER__)
      |                                      ^
./include/linux/compiler.h:398:4: note: in definition of macro ‘__compiletime_assert’
  398 |    prefix ## suffix();    \
      |    ^~~~~~
./include/linux/compiler.h:417:2: note: in expansion of macro ‘_compiletime_assert’
  417 |  _compiletime_assert(condition, msg, __compiletime_assert_, __COUNTER__)
      |  ^~~~~~~~~~~~~~~~~~~
./include/linux/build_bug.h:39:37: note: in expansion of macro ‘compiletime_assert’
   39 | #define BUILD_BUG_ON_MSG(cond, msg) compiletime_assert(!(cond), msg)
      |                                     ^~~~~~~~~~~~~~~~~~
./include/linux/bitfield.h:48:3: note: in expansion of macro ‘BUILD_BUG_ON_MSG’
   48 |   BUILD_BUG_ON_MSG((_mask) == 0, _pfx "mask is zero"); \
      |   ^~~~~~~~~~~~~~~~
./include/linux/bitfield.h:81:3: note: in expansion of macro ‘__BF_FIELD_CHECK’
   81 |   __BF_FIELD_CHECK(_mask, 0ULL, _val, "FIELD_PREP: "); \
      |   ^~~~~~~~~~~~~~~~
/home/user/Downloads/rtw89/core.h:3047:25: note: in expansion of macro ‘FIELD_PREP’
 3047 |  set = (orig & ~mask) | FIELD_PREP(mask, data);
      |                         ^~~~~~~~~~
./include/linux/compiler.h:417:38: error: call to ‘__compiletime_assert_397’ declared with attribute error: BUILD_BUG_ON failed: (((mask) + (1ULL << (__builtin_ffsll(mask) - 1))) & (((mask) + (1ULL << (__builtin_ffsll(mask) - 1))) - 1)) != 0
  417 |  _compiletime_assert(condition, msg, __compiletime_assert_, __COUNTER__)
      |                                      ^
./include/linux/compiler.h:398:4: note: in definition of macro ‘__compiletime_assert’
  398 |    prefix ## suffix();    \
      |    ^~~~~~
./include/linux/compiler.h:417:2: note: in expansion of macro ‘_compiletime_assert’
  417 |  _compiletime_assert(condition, msg, __compiletime_assert_, __COUNTER__)
      |  ^~~~~~~~~~~~~~~~~~~
./include/linux/build_bug.h:39:37: note: in expansion of macro ‘compiletime_assert’
[... 399 lines cut ...]
  417 |  _compiletime_assert(condition, msg, __compiletime_assert_, __COUNTER__)
      |  ^~~~~~~~~~~~~~~~~~~
./include/linux/build_bug.h:39:37: note: in expansion of macro ‘compiletime_assert’
   39 | #define BUILD_BUG_ON_MSG(cond, msg) compiletime_assert(!(cond), msg)
      |                                     ^~~~~~~~~~~~~~~~~~
./include/linux/build_bug.h:50:2: note: in expansion of macro ‘BUILD_BUG_ON_MSG’
[... cut ...]
   95 |   __BF_FIELD_CHECK(_mask, _reg, 0U, "FIELD_GET: "); \
      |   ^~~~~~~~~~~~~~~~
/home/user/Downloads/rtw89/core.h:2999:9: note: in expansion of macro ‘FIELD_GET’
 2999 |  return FIELD_GET(mask, rtw89_read32(rtwdev, addr));
      |         ^~~~~~~~~
make[2]: *** [scripts/Makefile.build:270: /home/user/Downloads/rtw89/debug.o] Error 1
make[1]: *** [Makefile:1762: /home/user/Downloads/rtw89] Error 2
make[1]: Leaving directory '/usr/src/linux-headers-5.4.0-90-generic'
make: *** [Makefile:54: all] Error 2
```

tayoagagu, 2021-11-30:
same issue here

```
OS: Ubuntu 20.04.3 LTS x86_64
Kernel: 5.11.0-41-generic

```

snieguu, 2021-11-30:
I have tried also:
```
snieguu@snieguu-Legion-5-15ACH6H:~/rtw89$ uname -r
5.15.5-051505-generic
snieguu@snieguu-Legion-5-15ACH6H:~/rtw89$ lsb_release -a
No LSB modules are available.
Distributor ID:	Ubuntu
Description:	Ubuntu 21.10
Release:	21.10
Codename:	impish
snieguu@snieguu-Legion-5-15ACH6H:~/rtw89$ make
make -C /lib/modules/5.15.5-051505-generic/build M=/home/snieguu/rtw89 modules
make[1]: Entering directory '/usr/src/linux-headers-5.15.5-051505-generic'
warning: the compiler differs from the one used to build the kernel
  The kernel was built by: gcc (Ubuntu 11.2.0-12ubuntu1) 11.2.0
  You are using:           gcc (Ubuntu 11.2.0-7ubuntu2) 11.2.0
  CC [M]  /home/snieguu/rtw89/core.o
In file included from /home/snieguu/rtw89/coex.h:8,
                 from /home/snieguu/rtw89/core.c:6:
/home/snieguu/rtw89/core.h: In function ‘rtw89_read32_mask’:
/home/snieguu/rtw89/core.h:2999:16: error: implicit declaration of function ‘field_get’; did you mean ‘of_led_get’? [-Werror=implicit-function-declaration]
 2999 |         return field_get(mask, rtw89_read32(rtwdev, addr));
      |                ^~~~~~~~~
      |                of_led_get
/home/snieguu/rtw89/core.h: In function ‘rtw89_write32_mask’:
/home/snieguu/rtw89/core.h:3023:32: error: implicit declaration of function ‘field_prep’ [-Werror=implicit-function-declaration]
 3023 |         set = (orig & ~mask) | field_prep(mask, data);
      |                                ^~~~~~~~~~
cc1: some warnings being treated as errors
make[2]: *** [scripts/Makefile.build:277: /home/snieguu/rtw89/core.o] Error 1
make[1]: *** [Makefile:1874: /home/snieguu/rtw89] Error 2
make[1]: Leaving directory '/usr/src/linux-headers-5.15.5-051505-generic'
make: *** [Makefile:54: all] Error 2
```

lwfinger, 2021-11-30:
These problems are my fault. I forgot that the bitfield commands would only build with kernel 5.16. That patch has been reversed. Remove any changes that you have made ('git checkout -f), pull with 'git pull', and rebuild.

closed 2021-11-30 by lwfinger