lwfinger/rtw89 #85: latency issues on t14 amd gen2

jeffclay, 2021-11-30:
Anyone else having latency problems? Pinging hosts on my LAN returns some very wild variations up to and sometimes exceeding 200ms. Testing with a T14 AMD Gen1 sitting right next to the Gen2 and pinging the same LAN host returns a consistent 1-5ms latency. This makes me think the problem is with the driver or the card on the Gen2. I am using v7 from this repo.

capfredf, 2021-11-30:
I run opensuse TW on a T14 AMD gen 2. I used the driver since v5 from this repo till recently the driver was backported to the TW kernel. I have never run into any issue with my WIFI.

To rule out the possibility of a hardware problem, I'd start Windows and see if the performance issue are still around.

lwfinger, 2021-11-30:
Yes, this driver has some latency problems with ping because of the packet aggregation it uses to get the high wifi5 throughput. Even with these "high ping" latencies, I still get roughly 600 Mbps up and down with the 5G band while connected to an AC1900 router on the 5G band. These rates are using a local Librespeed server. My external line maxes out at 200 Mbps down and 12 Mbps up. Driver rtw89 can easily saturate my broadband connection.

The small packet size of a ping packet is good for testing connectivity, but it is a lousy way to estimate real throughput.

closed 2021-11-30 by lwfinger

jeffclay, 2021-11-30:
Throughput is not my primary concern. Throughput is worthless when your latency causes constant lag when just trying to type in a ssh session. I don't even want to imagine how this detrimental this latency would be to voip or video communications.

Is this driver line for line the same as what was submitted for inclusion in the 5.16 kernel?

lwfinger, 2021-11-30:
Yes, other than the modifications needed for it to build on kernels older than 5.16.