lwfinger/rtw89 #86: make error ‘struct ieee80211_bss_conf’ has no member named ‘he_bss_color’

cordx, 2021-11-30:
```
user@computer ~/D/rtw89 (main)> lsb_release -d
Description:	Linux Mint 20.2
user@computer ~/D/rtw89 (main)> uname -r
5.4.0-90-generic
```
i can see bss_color in /usr/src/linux-headers-5.4.0-90-generic/include/net/mac80211.h but no he_bss_color. removing he_ and .color from /rtw89/phy.c line 2903 (so that it reads bss_color = vif->bss_conf.bss_color;) allows make to finish, but i am not sure if that is a proper solution.
```
user@computer ~/D/rtw89 (main)> make
make -C /lib/modules/5.4.0-90-generic/build M=/home/user/Downloads/rtw89 modules
make[1]: Entering directory '/usr/src/linux-headers-5.4.0-90-generic'
  CC [M]  /home/user/Downloads/rtw89/core.o
  CC [M]  /home/user/Downloads/rtw89/debug.o
  CC [M]  /home/user/Downloads/rtw89/mac80211.o
  CC [M]  /home/user/Downloads/rtw89/mac.o
  CC [M]  /home/user/Downloads/rtw89/phy.o
/home/user/Downloads/rtw89/phy.c: In function ‘rtw89_phy_set_bss_color’:
/home/user/Downloads/rtw89/phy.c:2903:28: error: ‘struct ieee80211_bss_conf’ has no member named ‘he_bss_color’; did you mean ‘bss_color’?
 2903 |  bss_color = vif->bss_conf.he_bss_color.color;
      |                            ^~~~~~~~~~~~
      |                            bss_color
make[2]: *** [scripts/Makefile.build:270: /home/user/Downloads/rtw89/phy.o] Error 1
make[1]: *** [Makefile:1762: /home/user/Downloads/rtw89] Error 2
make[1]: Leaving directory '/usr/src/linux-headers-5.4.0-90-generic'
make: *** [Makefile:54: all] Error 2
```

lwfinger, 2021-12-01:
I think I got that fixed - finally.

cordx, 2021-12-01:
i removed the old directory, cloned and ran make a couple of times. each time i started new. unfortunately i am still getting the same error. i see he_bss_color in cam.c and mac.c also, but the error is still (all other lines identical as well):
```
/rtw89/phy.c:2903:28: error: ‘struct ieee80211_bss_conf’ has no member named ‘he_bss_color’; did you mean ‘bss_color’?
 2903 |  bss_color = vif->bss_conf.he_bss_color.color;
```

cordx, 2021-12-01:
is CONFIG_SUSE_VERSION (or any of the other SUSE configs) defined in version.h on SUSE systems? if so, (according to core.h) should line 2900 of phy.c be #if LINUX_VERSION_CODE < KERNEL_VERSION(5, 7, 0)  && SUSE == 0 for any system not running SUSE? make finishes if i use that line, but i am not entirely certain i have the logic right.

lwfinger, 2021-12-01:
You do. Mine was wrong, but I think I have it right now. At least the code compiles cleanly on my 5.16 kernel, and on openSUSE Leap 15.3. Do a 'git pull' and check that I have it right now.

cordx, 2021-12-01:
unfortunately i get a new error now:
```
user@computer ~/P/rtw89 (main)> make
make -C /lib/modules/5.4.0-90-generic/build M=/home/user/Public/rtw89 modules
make[1]: Entering directory '/usr/src/linux-headers-5.4.0-90-generic'
  CC [M]  /home/user/Public/rtw89/core.o
In file included from /home/user/Public/rtw89/coex.h:8,
                 from /home/user/Public/rtw89/core.c:6:
/home/user/Public/rtw89/core.h:3448:31: error: redefinition of ‘ieee80211_tx_dequeue_ni’
 3448 | static inline struct sk_buff *ieee80211_tx_dequeue_ni(struct ieee80211_hw *hw,
      |                               ^~~~~~~~~~~~~~~~~~~~~~~
In file included from /home/user/Public/rtw89/core.h:13,
                 from /home/user/Public/rtw89/coex.h:8,
                 from /home/user/Public/rtw89/core.c:6:
./include/net/mac80211.h:6276:31: note: previous definition of ‘ieee80211_tx_dequeue_ni’ was here
 6276 | static inline struct sk_buff *ieee80211_tx_dequeue_ni(struct ieee80211_hw *hw,
      |                               ^~~~~~~~~~~~~~~~~~~~~~~
make[2]: *** [scripts/Makefile.build:270: /home/user/Public/rtw89/core.o] Error 1
make[1]: *** [Makefile:1762: /home/user/Public/rtw89] Error 2
make[1]: Leaving directory '/usr/src/linux-headers-5.4.0-90-generic'
make: *** [Makefile:54: all] Error 2
```

lwfinger, 2021-12-01:
That routine was added in 5.4, not 5.7. Please try again.

cordx, 2021-12-01:
that is the one. make was successful. thank you for your work and help.

cordx, 2021-12-01:
that worked. make was successful. thank you for your work and help.

cordx, 2021-12-01:
sorry for the close and reopen. i clicked the wrong button a couple of times.

closed 2021-12-01 by lwfinger